isaac-sim/IsaacLab #2786: ModuleNotFoundError: No module named 'isaacsim'

quinn-kzk, 2025-06-26:
Isaac Sim Version

4.5.0 x
4.2.0
4.1.0
4.0.0
4.5.0
2023.1.1
2023.1.0-hotfix.1
Other (please specify):
Operating System

Ubuntu 22.04 x
Ubuntu 20.04
Windows 11
Windows 10
Other (please specify):
GPU Information

    Model: 5060ti
    Driver Version: 570.153.02

Topic Description
Detailed Description

I’m trying to install isaaclab in order to run robot dog training simulations but it won’t work even though I believe I installed it correctly (though I definitely did not considering I’m here). I downloaded isaac sim and have it working, downloaded the dependencies after cloning Isaac Lab repo and then installed using the .sh file but it is still not recognizing as a module. When running the verification file, I get the following error:

Traceback (most recent call last):
File “/home/XXXXX/Desktop/IsaacLab/scripts/tutorials/00_sim/create_empty.py”, line 20, in
from isaaclab.app import AppLauncher
File “/home/XXXXX/Desktop/IsaacLab/source/isaaclab/isaaclab/app/init.py”, line 15, in
from .app_launcher import AppLauncher # noqa: F401, F403
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File “/home/XXXXX/Desktop/IsaacLab/source/isaaclab/isaaclab/app/app_launcher.py”, line 30, in
from isaacsim import SimulationApp
ModuleNotFoundError: No module named ‘isaacsim’
Steps to Reproduce

    Virtual Environment with Python 3.10
    Install IsaacSim and have it working
    Clone IsaacLab and connect to IsaacSIm
    Install dependencies with no problem
    Attempt to install through isaaclab.sh and have everything work except rl_games because “requires a different Python”
    Run the verification Isaac Lab command and get a no module found error.

Error Messages

(If applicable, copy and paste any error messages you received)
Traceback (most recent call last):
File “/home/XXXXX/Desktop/IsaacLab/scripts/tutorials/00_sim/create_empty.py”, line 20, in
from isaaclab.app import AppLauncher
File “/home/XXXXX/Desktop/IsaacLab/source/isaaclab/isaaclab/app/init.py”, line 15, in
from .app_launcher import AppLauncher # noqa: F401, F403
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File “/home/XXXXX/Desktop/IsaacLab/source/isaaclab/isaaclab/app/app_launcher.py”, line 30, in
from isaacsim import SimulationApp
ModuleNotFoundError: No module named ‘isaacsim’
Screenshots or Videos

(If applicable, add screenshots or links to videos that demonstrate the issue)
Additional Information
What I’ve Tried

(Describe any troubleshooting steps you’ve already taken)
Trying different environments, reinstalling isaaclab entirely, reconnecting the link to isaacsim
Related Issues

(If you’re aware of any related issues or forum posts, please link them here)
Not sure if it’s connected but when installing isaaclab from ./isaaclab.sh -i, I get an error saying rl_games requires a different python that doesn’t match up with the version that is running in my environment. However, I did go and make sure I finished all the installs of the other frameworks without a problem.

quinn-kzk, 2025-06-26:
Hey @Toni-SM , of course, sorry about that!
Made a Python 3.10 environment with
python3.10 -m venv isaac_env

IsaacSim install - 
pip install 'isaacsim[all,extscache]==4.5.0' --extra-index-url https://pypi.nvidia.com
and ran isaacsim to make sure it worked

Cloned Isaac Lab to my desktop with
git clone https://github.com/isaac-sim/IsaacLab.git

Installed Dependencies with
sudo apt install cmake build-essential

Connected to my isaacsim with
ln -s isaacsim _isaac_sim

and then installed Isaac Lab with
cd IsaacLab
./isaaclab.sh -i

That installs fine except for an error saying
ERROR: Package 'rl-games' requires a different Python: 3.12.1 not in '<3.11,>=3.7.1'

but I installed the rest of the modules individually

Toni-SM, 2025-06-26:
Hi @Toni-SM 

You are merging two different installations ways:
* [Installation using Isaac Sim pip](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html)
* [Installation using Isaac Sim Binaries](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/binaries_installation.html): that where `ln -s isaacsim _isaac_sim` is required

Please follows only one of those 2 ways

quinn-kzk, 2025-06-30:
Hey @Toni-SM , I hope you're still able to get this but I tried again and followed both sets of instructions separately and neither way worked. No matter what I do, I get a ModuleNotFoundError with no module named isaacsim even though the link is there. Is it maybe cause the isaaclab install doesn't work correctly because I'm getting a "Package rl-games requires a different Python" but that doesn't make sense either because my environment is in 3.10.12

Toni-SM, 2025-06-30:
Hi @quinn-kzk 

Please provide in details, and steps by steps, the commands you are running