isaac-sim/IsaacLab #2791: [Bug Report] Bug title

tom2002888, 2025-06-27:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

Isaac sim 4.5

I am now trying the Isaac sim Isaac lab tutorial about Modifying an existing Direct RL Environment. I followed the step, but when executing in the program using:

python scripts/reinforcement_learning/skrl/train.py --task Isaac-H1-Direct-v0

python scripts/reinforcement_learning/rl_games/train.py --task Isaac-H1-Direct-v0

./isaaclab.sh -p scripts/reinforcement_learning/rl_games/train.py --task Isaac-H1-Direct-v0 --headless

three of them will show the window of isaac lab, but after that the window closed and the error message is that:

env_isaaclab/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 376, in _check_name_exists
    raise error.NameNotFound(
gymnasium.error.NameNotFound: Environment `Isaac-H1-Direct` doesn't exist. Did you mean: `Isaac-Ant-Direct`?

however,  when i run python scripts/reinforcement_learning/skrl/train.py --task IIsaac-Humanoid-Direct-v0, it work very well. Please help.

The code files are under 

Desktop/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/direct/humanoid/__init__.py
source code below:

__init__.py:
`# Copyright (c) 2022-2025, The Isaac Lab Project Developers (https://github.com/isaac-sim/IsaacLab/blob/main/CONTRIBUTORS.md).
# All rights reserved.
#
# SPDX-License-Identifier: BSD-3-Clause

"""
Humanoid locomotion environment.
"""
from .h1_env import H1Env, H1EnvCfg
import gymnasium as gym

from . import agents

##
# Register Gym environments.
##

from .h1_env import H1Env, H1EnvCfg

gym.register(
    id="Isaac-H1-Direct-v0",
    entry_point=f"{__name__}.h1_env:H1Env",
    disable_env_checker=True,
    kwargs={
        "env_cfg_entry_point": f"{__name__}.h1_env:H1EnvCfg",
        "rl_games_cfg_entry_point": f"{agents.__name__}:rl_games_ppo_cfg.yaml",
        "rsl_rl_cfg_entry_point": f"{agents.__name__}.rsl_rl_ppo_cfg:HumanoidPPORunnerCfg",
        "skrl_cfg_entry_point": f"{agents.__name__}:skrl_ppo_cfg.yaml",
    },
)

gym.register(
    id="Isaac-Humanoid-Direct-v0",
    entry_point=f"{__name__}.humanoid_env:HumanoidEnv",
    disable_env_checker=True,
    kwargs={
        "env_cfg_entry_point": f"{__name__}.humanoid_env:HumanoidEnvCfg",
        "rl_games_cfg_entry_point": f"{agents.__name__}:rl_games_ppo_cfg.yaml",
        "rsl_rl_cfg_entry_point": f"{agents.__name__}.rsl_rl_ppo_cfg:HumanoidPPORunnerCfg",
        "skrl_cfg_entry_point": f"{agents.__name__}:skrl_ppo_cfg.yaml",
    },
)
`

The h1_env.py:

`from __future__ import annotations

from isaaclab_assets import H1_CFG

import isaaclab.sim as sim_utils
from isaaclab.assets import ArticulationCfg
from isaaclab.envs import DirectRLEnvCfg
from isaaclab.scene import InteractiveSceneCfg
from isaaclab.sim import SimulationCfg
from isaaclab.terrains import TerrainImporterCfg
from isaaclab.utils import configclass
from isaaclab_tasks.direct.locomotion.locomotion_env import LocomotionEnv

@configclass
class H1EnvCfg(DirectRLEnvCfg):
    # env
    episode_length_s = 15.0
    decimation = 2
    action_scale = 1.0
    action_space = 19
    observation_space = 69
    state_space = 0

    # simulation
    sim: SimulationCfg = SimulationCfg(dt=1 / 120, render_interval=decimation)
    terrain = TerrainImporterCfg(
        prim_path="/World/ground",
        terrain_type="plane",
        collision_group=-1,
        physics_material=sim_utils.RigidBodyMaterialCfg(
            friction_combine_mode="average",
            restitution_combine_mode="average",
            static_friction=1.0,
            dynamic_friction=1.0,
            restitution=0.0,
        ),
        debug_vis=False,
    )

    # scene
    scene: InteractiveSceneCfg = InteractiveSceneCfg(num_envs=4096, env_spacing=4.0, replicate_physics=True)

    # robot
    robot: ArticulationCfg = H1_CFG.replace(prim_path="/World/envs/env_.*/Robot")
    joint_gears: list = [
        50.0,  # left_hip_yaw
        50.0,  # right_hip_yaw
        50.0,  # torso
        50.0,  # left_hip_roll
        50.0,  # right_hip_roll
        50.0,  # left_shoulder_pitch
        50.0,  # right_shoulder_pitch
        50.0,  # left_hip_pitch
        50.0,  # right_hip_pitch
        50.0,  # left_shoulder_roll
        50.0,  # right_shoulder_roll
        50.0,  # left_knee
        50.0,  # right_knee
        50.0,  # left_shoulder_yaw
        50.0,  # right_shoulder_yaw
        50.0,  # left_ankle
        50.0,  # right_ankle
        50.0,  # left_elbow
        50.0,  # right_elbow
    ]

    heading_weight: float = 0.5
    up_weight: float = 0.1

    energy_cost_scale: float = 0.05
    actions_cost_scale: float = 0.01
    alive_reward_scale: float = 2.0
    dof_vel_scale: float = 0.1

    death_cost: float = -1.0
    termination_height: float = 0.8

    angular_velocity_scale: float = 0.25
    contact_force_scale: float = 0.01

class H1Env(LocomotionEnv):
    cfg: H1EnvCfg

    def __init__(self, cfg: H1EnvCfg, render_mode: str | None = None, **kwargs):
        super().__init__(cfg, render_mode, **kwargs)
        from .h1_env import H1Env, H1EnvCfg`

[image]

[image]

Thank you so much for your time and helping!

RandomOakForest, 2025-06-28:
Thank you for posting this. Are the solutions in #1522 or #2253 of any help?

tom2002888, 2025-06-30:
Can anybody help? I stuck in this issue for 1 week

closed 2025-08-01 by RandomOakForest