isaac-sim/IsaacLab #2792: [Bug Report] RuntimeError: Failed to open project URL https://pypi.nvidia.com/isaacsim/

DexHand, 2025-06-27:
### Describe the bug

I could install and run isaaclab+isaacsim4.5.0 before.
But when I reinstalled the conda env today, the error comes:
RuntimeError: Failed to open project URL https://pypi.nvidia.com/isaacsim/

EntangledQuantum, 2025-06-27:
Got Same Error

```console
Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [72 lines of output]
        File "/tmp/pip-build-env-u79y00px/overlay/lib/python3.10/site-packages/wheel_stub/wheel.py", line 249, in download_wheel
          return download_manual(wheel_directory, distribution, version, config)
        File "/tmp/pip-build-env-u79y00px/overlay/lib/python3.10/site-packages/wheel_stub/wheel.py", line 176, in download_manual
          raise RuntimeError(f"Failed to open project URL {project_url}") from e
      Traceback (most recent call last):
        File "/tmp/pip-build-env-u79y00px/overlay/lib/python3.10/site-packages/wheel_stub/wheel.py", line 174, in download_manual
          index_response = urlopen_with_retry(project_url)
        File "/tmp/pip-build-env-u79y00px/overlay/lib/python3.10/site-packages/wheel_stub/wheel.py", line 92, in urlopen_with_retry
          return urlopen(url, **kwargs)
        File "/home/azureuser/miniconda3/envs/sim/lib/python3.10/urllib/request.py", line 216, in urlopen
          return opener.open(url, data, timeout)
        File "/home/azureuser/miniconda3/envs/sim/lib/python3.10/urllib/request.py", line 525, in open
          response = meth(req, response)
        File "/home/azureuser/miniconda3/envs/sim/lib/python3.10/urllib/request.py", line 634, in http_response
          response = self.parent.error(
        File "/home/azureuser/miniconda3/envs/sim/lib/python3.10/urllib/request.py", line 563, in error
          return self._call_chain(*args)
        File "/home/azureuser/miniconda3/envs/sim/lib/python3.10/urllib/request.py", line 496, in _call_chain
          result = func(*args)
        File "/home/azureuser/miniconda3/envs/sim/lib/python3.10/urllib/request.py", line 643, in http_error_default
          raise HTTPError(req.full_url, code, msg, hdrs, fp)
      urllib.error.HTTPError: HTTP Error 404: Not Found
```

Darth-S1d1ous, 2025-06-27:
Same here, earlier this day `ping https//pypi.nvidia.com` returns `404`. Now the site is working but isaacsim is not in the package list

Toni-SM, 2025-06-27:
Hi @DexHand, @LouKordos, @EntangledQuantum, @Darth-S1d1ous 

Thank you for the report.

We are experiencing an issue with the Isaac Sim and Isaac Lab packages in the pypi.nvidia.com Python index.
We are working on restoring them.

songhat, 2025-06-29:
@Toni-SM Same . Is this issue going to fix?

isFcs, 2025-06-29:
how long

songhat, 2025-06-30:
@Toni-SM thank you for your time. I will try this way.

Toni-SM, 2025-07-01:
The issues with the Isaac Sim and Isaac Lab Python packages have been resolved.
These packages are available again.

Thank you for your patience.