isaac-sim/IsaacLab #2795: [Question] Action Cfg Discrete Space

shiven2001, 2025-06-27:
So it seems that there are no examples (at least I didn't see it, if there are please let me know) that use discrete action space in IsaacLab. My task is following:

I have a floating camera and I want the action space to be pose and orientation change in each dt meaning the camera can move in x,y,z by 1 unit or decide to rotate along x,y,z by 10 degrees. I made the code according to what I thought should be ok but it seems to have a couple of issues:

      from enum import Enum, auto
      from typing import Sequence, Tuple, Type
      import os
      import sys
      import torch
      from pydantic import BaseModel
      import math
      
      from isaaclab.utils import configclass
      from isaaclab.managers import ActionTerm, ActionTermCfg
      from isaaclab.assets import RigidObject
      from isaaclab.envs import ManagerBasedEnv
      
      script_dir = os.path.dirname(os.path.abspath(__file__))
      sys.path.append(script_dir)
      
      from xutils.xmath import deg_to_rad
      from xutils.xlogging import logging
      
      
      class FloatingCameraActionDiscrete(ActionTerm):
          cfg: "FloatingCameraActionCfg"
          _env: ManagerBasedEnv
          _asset: RigidObject
      
          def __init__(self, cfg: "FloatingCameraActionCfg", env: ManagerBasedEnv):
              super().__init__(cfg, env)
      
              # Initialize asset
              self._asset = env.scene[cfg.asset_name]
              self._initial_state = self._asset.data.root_state_w  # Shape: (num_envs, 13)
      
              # Setup action space
              self._setup_action_space()
      
              # Initialize action tensors
              self._raw_actions = torch.zeros((self.num_envs, 1), device=self.device)
              self._processed_actions = torch.zeros(
                  (self.num_envs, 6), device=self.device
              )  # pos (3) + rot (3)
      
              # Initialize current state (position + quaternion)
              self._current_state = self._initial_state.clone()  # Shape: (num_envs, 13)
              # Set initial position to (0, 0, 1.5) relative to env origin, keep initial quaternion
              self._current_state[:, :3] = torch.tensor([0.0, 0.0, 1.5], device=self.device)
      
              # Define position bounds (relative to env origin)
              self._pos_bounds_min = torch.tensor(
                  [
                      self.cfg.pos_bounds[0][0],
                      self.cfg.pos_bounds[1][0],
                      self.cfg.pos_bounds[2][0],
                  ],
                  device=self.device,
              )
              self._pos_bounds_max = torch.tensor(
                  [
                      self.cfg.pos_bounds[0][1],
                      self.cfg.pos_bounds[1][1],
                      self.cfg.pos_bounds[2][1],
                  ],
                  device=self.device,
              )
      
              # Define orientation bounds (radians, from cfg.rot_bounds)
              self._rot_bounds_min = torch.tensor(
[... 218 lines cut ...]
      
      
      @configclass
      class FloatingCameraActionDiscreteCfg(ActionTermCfg):
          class_type: Type = FloatingCameraActionDiscrete
          asset_name: str = "float_cam"
          scale: float = 1.0
          control_mode: CameraControlMode = CameraControlMode.DISCRETE
          pos_bounds: Tuple[Tuple[float, float], Tuple[float, float], Tuple[float, float]] = (
              (-10.0, 10.0),  # x
              (-10.0, 10.0),  # y
              (0.0, 10.0),  # z
          )
          rot_bounds: Tuple[Tuple[float, float], Tuple[float, float], Tuple[float, float]] = (
              (-90.0, 90.0),  # roll
              (-90.0, 90.0),  # pitch
              (-90.0, 90.0),  # yaw
          )
          spacing: float = 1.0
          rot_angle: float = 10.0
      

ISSUE 1: I always test my stuff is working using the random_agent.py file but when I tried this discrete space, the camera didn't move at all. 

ISSUE 2: I tried with the rsl_rl train.py with PPO default config and then camera moves it seems it is not working because most of the time, camera stays at a place or maybe moves left or right a bit along x axis but or rotation and movement along other axis happens. I am quite sure it is my bad implementation of the discrete space. Can sometime tell me where I went wrong and how to fix this? I think an example of discrete space would really benefit the community.

Toni-SM, 2025-06-27:
Hi @shiven2001 

Refer to the Isaac Lab's [Spaces showcase](https://isaac-sim.github.io/IsaacLab/main/source/overview/environments.html#spaces-showcase) documentation for examples on defining/using the various Gymnasium observation and action spaces supported in Isaac Lab.

Also, see the https://github.com/isaac-sim/IsaacLab/tree/main/source/isaaclab_tasks/isaaclab_tasks/direct/cartpole_showcase code

Toni-SM, 2025-06-30:
Hi @shiven2001 

> Does that mean I cannot use discrete space in Manager Based Env?

Only the Isaac Lab’s Direct workflow supports the different spaces

closed 2025-08-01 by RandomOakForest