isaac-sim/IsaacLab #2799: [Question] Issue adding new terrain to Manager-BasedRL

rushabhdhoke, 2025-06-27:
### Question

I am trying to add a new terrain to the Manager Based RL for training a GO2 robot. I generated a new terrain using terrain generator and converted it to USD file which is saved in a folder outside of IsaacLab directory. 

~/terrains/steps_0.usd

I have updated the rough_env_cfg.py file as well.

"self.scene.robot = UNITREE_GO2_CFG.replace(prim_path="{ENV_REGEX_NS}/Robot")
        self.scene.height_scanner.prim_path = "{ENV_REGEX_NS}/Robot/base"
        self.scene.terrain=TerrainImporterCfg(
            prim_path="/World/gorund",
            terrain_type = "usd",
            usd_path = "/home/rushabh/terrains/steps_0_a.usd",
            physics_material = dict(dynamic_friction = 0.7,restitution=0.0),)"

After running the script "./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py     --task Isaac-Velocity-Ramp-Unitree-GO2-v0     --headless"

I am getting
"raise ValueError(f"Prim at path '{prim_path}' is not valid.")
ValueError: Prim at path '/World/ground' is not valid."

I dont get what's wrong.. Can someone please help

RandomOakForest, 2025-06-28:
Thank you for posting this. Here's a summary to consider. The error occurs because:

1. Your `prim_path` is set to `"/World/gorund"` (typo in "ground")
2. Isaac Lab expects terrain at `/World/ground` by default in some workflows
3. The USD file path might not be resolved correctly

### Step-by-Step Fix

1. **Correct the `prim_path` typo**:

```python
self.scene.terrain = TerrainImporterCfg(
    prim_path="/World/ground",  # Corrected spelling
    terrain_type="usd",
    usd_path="/home/rushabh/terrains/steps_0_a.usd",
    physics_material=dict(dynamic_friction=0.7, restitution=0.0)
)
```

2. **Verify USD file existence**:
    - Confirm the file exists at `/home/rushabh/terrains/steps_0_a.usd`
    - Ensure Isaac Lab has read permissions for this path
3. **Use absolute path with `pathlib` (recommended)**:

```python
from pathlib import Path
usd_path = str(Path.home() / "terrains" / "steps_0_a.usd")
```

4. **Check terrain initialization order**:
Ensure terrain is imported before robots:

```python
self.scene.terrain = ...  # Should be configured BEFORE robots
self.scene.robot = ... 
```

### Important Checks

1. **USD File Validation**:
    - Open `/home/rushabh/terrains/steps_0_a.usd` in Isaac Sim to confirm:
        - Mesh loads correctly
        - No errors in Omniverse Log
    - Use `Usd.Stage.Open(usd_path)` to validate programmatically
2. **Collision Settings**:
Ensure your terrain has collision meshes:

```python
physics_material=dict(
    dynamic_friction=0.7,
    restitution=0.0,
    collision_enabled=True  # Explicitly enable
)
```

rushabhdhoke, 2025-06-29:
Ok alright, I have made the necessary changes as suggested.. and it worked although, I am getting another error now. 

`
"Traceback (most recent call last):

  File "/home/rushabh/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/hydra.py", line 101, in hydra_main

    func(env_cfg, agent_cfg, *args, **kwargs)

  File "/home/rushabh/IsaacLab/scripts/reinforcement_learning/rsl_rl/train.py", line 164, in main

    env = RslRlVecEnvWrapper(env, clip_actions=agent_cfg.clip_actions)

  File "/home/rushabh/IsaacLab/source/isaaclab_rl/isaaclab_rl/rsl_rl/vecenv_wrapper.py", line 85, in __init__

    self.env.reset()

  File "/home/rushabh/IsaacLab/_isaac_sim/kit/python/lib/python3.10/site-packages/gymnasium/wrappers/common.py", line 400, in reset

    return super().reset(seed=seed, options=options)

  File "/home/rushabh/IsaacLab/_isaac_sim/kit/python/lib/python3.10/site-packages/gymnasium/core.py", line 333, in reset

    return self.env.reset(seed=seed, options=options)

  File "/home/rushabh/IsaacLab/_isaac_sim/kit/python/lib/python3.10/site-packages/gymnasium/wrappers/common.py", line 293, in reset

    return env_reset_passive_checker(self.env, seed=seed, options=options)

  File "/home/rushabh/IsaacLab/_isaac_sim/kit/python/lib/python3.10/site-packages/gymnasium/utils/passive_env_checker.py", line 185, in env_reset_passive_checker

    result = env.reset(**kwargs)

  File "/home/rushabh/IsaacLab/source/isaaclab/isaaclab/envs/manager_based_env.py", line 295, in reset

    self._reset_idx(env_ids)

  File "/home/rushabh/IsaacLab/source/isaaclab/isaaclab/envs/manager_based_rl_env.py", line 355, in _reset_idx

    self.curriculum_manager.compute(env_ids=env_ids)

  File "/home/rushabh/IsaacLab/source/isaaclab/isaaclab/managers/curriculum_manager.py", line 138, in compute

    state = term_cfg.func(self._env, env_ids, **term_cfg.params)

  File "/home/rushabh/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/manager_based/locomotion/velocity/mdp/curriculums.py", line 48, in terrain_levels_vel
`

`move_up = distance > terrain.cfg.terrain_generator.size[0] / 2

AttributeError: 'NoneType' object has no attribute 'size'
`

To get rid of this I commented  out Curriculum in the velocity_env_cfg.py file

`    curriculum: CurriculumCfg = CurriculumCfg()
`
Although commenting out this line works for me... but as I observed, the rewards obtained for my training were terrible and kept on decreasing to around "Mean reward: -166083.08" 

Is there something I could do to completely fix it and get my robot to train on this terrain.

Thanks again!