isaac-sim/IsaacLab #2815: [Question] When play the task Isaac-Reach-UR10-v0, the robotic arm always shakes near the target point

YanmingChang666, 2025-06-30:
**Problem Summary**
I trained for 1000 epochs and 20000 epochs respectively, and during play the modle, the robotic arm always shakes near the target point.

Here are the results of my two training sessions and a demonstration video. Is it because the total reward is not high?

**Environment Details**
    Isaac Lab version: [2.0.0] 
    Task:Isaac-Reach-UR10-v0
    Device: CUDA

Could you please investigate this issue? Thank you for your time and consideration!

YanmingChang666, 2025-06-30:
[image]
[image]

https://github.com/user-attachments/assets/ba6d923c-577c-4b63-a5f7-a3773e3e7813

YuHoChau, 2025-06-30:
Try disable the noise.

RandomOakForest, 2025-07-01:
Thank you for posting this. I'll move the post to our _Discussions_ section for follow up. Here are a few notes you may consider trying.

### Joint Drive Tuning Procedure

Follow this iterative process for each joint to stabilize movement:

1. **Set damping to zero** and tune stiffness first[^5].
2. **Increase stiffness** incrementally until the joint converges near the target position without overshoot[^5].
3. **Reduce stiffness by one order of magnitude** from the initial stable value[^5].
4. **Add damping** at one order of magnitude lower than the final stiffness value[^5].
5. **Fine-tune both gains** to balance response speed and stability, avoiding excessive values that cause oscillations[^5][^2].

### Configuration Adjustments

- **Disable gravity** for all robot rigid bodies in the properties panel to prevent gravity-induced drift[^5][^3].
- **Set joint friction to zero** to avoid unintended resistance that amplifies oscillations[^7].
- **Avoid zero stiffness**: This prevents movement entirely; use minimal non-zero values instead[^2][^3].

### Training Considerations

- **Policy refinement**: If oscillations persist after tuning, adjust the RL policy’s action constraints to reduce aggressive velocity commands near targets[^6].
- **Actuator model**: Use implicit actuators for better compliance control, as they handle joint dynamics more effectively than direct effort commands[^7].

### Troubleshooting Notes

- **Persistent oscillations** often indicate insufficient damping relative to stiffness or excessive policy aggressiveness[^6].
- **Kinematic simulation** (without physics) isn’t natively supported; tuning gains remains essential even with gravity disabled[^2][^3].

Also, consider to explicitly override damping/friction parameters in asset configuration files to ensure values align with the tuning process above[^7].

[^1]: https://github.com/isaac-sim/IsaacLab/issues/2815

[^2]: https://forums.developer.nvidia.com/t/how-can-i-control-or-eliminate-oscillation-when-the-robot-arm-moves-to-the-target-position/329175

[^3]: https://forums.developer.nvidia.com/t/how-to-simulate-a-robot-arm-in-isaac-sim-without-physics-no-gravity-or-inertia/328982

[^4]: https://forums.developer.nvidia.com/t/shaking-manipulator-links-in-isaac-sim/232896

[^5]: https://docs.omniverse.nvidia.com/isaacsim/latest/advanced_tutorials/tutorial_advanced_joint_tuning.html

[^6]: https://forums.developer.nvidia.com/t/no-movement-has-been-assigned-to-the-head-of-the-robot-but-it-is-still-moving-in-simulation-after-training/321266

[^7]: https://github.com/isaac-sim/IsaacLab/discussions/2470

[^8]: https://forums.developer.nvidia.com/t/robot-oscilation-in-isaac-sim/278885

[^9]: https://docs.omniverse.nvidia.com/isaacsim/latest/how_to_guides/robots_simulation.html

[^10]: https://forums.developer.nvidia.com/t/shaking-object-issue-in-isaac-sim-of-the-omniverse/217355

[^11]: https://github.com/isaac-sim/IsaacLab/issues

[^12]: https://www.hackster.io/Elephant-Robotics-Official/mycobot-gripping-task-reinforcement-learning-with-isaac-gym-5621db

[^13]: https://www.youtube.com/watch?v=pGje2slp6-s

[^14]: https://www.youtube.com/watch?v=-jqepxgzs_I

[^15]: https://www.youtube.com/watch?v=MbjkapbKy_w

[^16]: https://forums.developer.nvidia.com/t/how-to-control-custom-agv-and-robot-arm/330291

[^17]: https://www.youtube.com/watch?v=Eb2zuQxOBlY

[^18]: https://wiki.seeedstudio.com/lerobot_so100m_isaacsim/