isaac-sim/IsaacLab #2827: [Bug Report] Physics related tutorials fail on isaacsim 5. branch

ewirbel, 2025-07-01:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

On the IsaacSim 5.0 branch of IsaacLab, all tutorial scripts involving physics (so anything except the 00_sim folder) fails at startup with an error on non existing physics scene `Exception: physics scene specified /physicsScene doesn't exist `(full error stack below)

### Steps to reproduce

- Compile IsaacSim 5.0
- Checkout the corresponding IsaacLab branch
- run any tutorial script eg `./_isaac_sim/python.sh scripts/tutorials/02_scene/create_scene.py`

```
Traceback (most recent call last):
  File "/home/emilie.wirbel/workspace/IsaacLab/scripts/tutorials/05_controllers/run_diff_ik.py", line 216, in <module>
    main()
  File "/home/emilie.wirbel/workspace/IsaacLab/scripts/tutorials/05_controllers/run_diff_ik.py", line 197, in main
    sim = sim_utils.SimulationContext(sim_cfg)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/emilie.wirbel/workspace/IsaacLab/source/isaaclab/isaaclab/sim/simulation_context.py", line 251, in __init__
    super().__init__(
  File "/home/emilie.wirbel/workspace/IsaacSim/_build/linux-x86_64/release/exts/isaacsim.core.api/isaacsim/core/api/simulation_context/simulation_context.py", line 150, in __init__
    self._init_stage(
  File "/home/emilie.wirbel/workspace/IsaacLab/source/isaaclab/isaaclab/sim/simulation_context.py", line 670, in _init_stage
    _ = super()._init_stage(*args, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/emilie.wirbel/workspace/IsaacSim/_build/linux-x86_64/release/exts/isaacsim.core.api/isaacsim/core/api/simulation_context/simulation_context.py", line 1351, in _init_stage
    self._physics_context = PhysicsContext(
                            ^^^^^^^^^^^^^^^
  File "/home/emilie.wirbel/workspace/IsaacSim/_build/linux-x86_64/release/exts/isaacsim.core.api/isaacsim/core/api/physics_context/physics_context.py", line 83, in __init__
    SimulationManager.set_default_physics_scene(self._prim_path)
  File "/home/emilie.wirbel/workspace/IsaacSim/_build/linux-x86_64/release/exts/isaacsim.core.simulation_manager/isaacsim/core/simulation_manager/impl/simulation_manager.py", line 250, in set_default_physics_scene
    raise Exception("physics scene specified {} doesn't exist".format(physics_scene_prim_path))
Exception: physics scene specified /physicsScene doesn't exist
```

For more information on this, check: https://www.markdownguide.org/extended-syntax/#fenced-code-blocks

-->

### System Info

Describe the characteristic of your environment:

- Commit: 7830f887b07d3eb794c3032f382c4916bf5f6e4e
- Isaac Sim Version: 5.0.0-rc.23+main.0.eeb2e862.local
- OS: Ubuntu 24.04
- GPU:  NVIDIA GeForce RTX 3090
- CUDA: 12.9
- GPU Driver: 575.57.08

### Additional context

The demo scripts (for example scripts/demos/quadcopter.py) work fine so I know it's not an issue with physx extension or IsaacSim

### Checklist

- [ x] I have checked that there is no similar issue in the repo (**required**)
- [ x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

jneeven, 2025-07-08:
I am encountering the same issue. I simply create an empty stage, create a `World` object with default parameters, and get this error. It seems that a physics scene exists, but is not present in `SimulationManager._physics_scene_apis` because it already existed before the register hooks were defined.

I've "fixed" this by monkey-patching `SimulationManager.set_default_physics_scene` as follows:
```python
    @classmethod
    def set_default_physics_scene(cls, physics_scene_prim_path: str):
        if SimulationManager._warm_start_callback is None:
            carb.log_warn(
                "Calling set_default_physics_scene while SimulationManager is not tracking physics scenes"
            )
            return
        if physics_scene_prim_path in SimulationManager._physics_scene_apis:
            SimulationManager._default_physics_scene_idx = list(
                SimulationManager._physics_scene_apis.keys()
            ).index(physics_scene_prim_path)
        else:
            prim = get_prim_at_path(physics_scene_prim_path)
            if prim.GetTypeName() == "PhysicsScene":
                SimulationManager._physics_scene_apis[physics_scene_prim_path] = (
                    PhysxSchema.PhysxSceneAPI.Apply(prim)
                )
            else:
                raise Exception(
                    "physics scene specified {} doesn't exist".format(
                        physics_scene_prim_path
                    )
                )
```

Whether that's correct I don't know; I'm not currently using any physics. It at least succeeds in not crashing everything.

AntoineRichard, 2026-08-27:
Thanks for catching this on a pre-release branch, @ewirbel — testing `feature/isaacsim_5_0` early is genuinely helpful. That crash path no longer exists: `isaaclab.sim.SimulationContext` is now a standalone class that never constructs a `PhysicsContext` or calls `SimulationManager.set_default_physics_scene`, so the `physics scene specified /physicsScene doesn't exist` error can't be raised. The feature survives as `SimulationCfg.create_stage_in_memory` (renamed from `create_scene_in_memory`), covered by `source/isaaclab/test/sim/test_simulation_stage_in_memory.py`.

Isaac Lab now ships against released Isaac Sim 6.0. Please reopen with a repro if you see anything similar.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard