isaac-sim/IsaacLab #2837: [Question] error when using UsdToUrdf

ifmorning, 2025-07-02:
### Question

I was trying to use teleoperation in IsaacLab to control a different robot which does not exist in IsaacLab file. For the pick_place task, **ControllerUtils.convert_usd_to_urdf()** function will be used in order to obtain the urdf_output_path and the meshes_output_path. For the robot I used, I encountered a error as the following:

**ValueError: The 'idx61_ee_l_joint''s joint transforms are not consistent.
global___joint_via_parent:
[[-0.00000009 -0.00000009  1.          0.13100001]
 [-1.         -0.00000071 -0.00000009  0.89390004]
 [ 0.00000071 -1.         -0.00000009  0.95349991]
 [ 0.          0.          0.          1.        ]]
global___joint_via_child:
[[-0.00000009 -0.00000009  1.          0.13099197]
 [-1.         -0.00000071 -0.00000009  0.89650011]
 [ 0.00000071 -1.         -0.00000009  0.95350003]
 [ 0.          0.          0.          1.        ]]**

For details, see the following.  These two matrix are so close, how can I bypass this error? Can I change the error threshold?

**Details:

[10.008s] Simulation App Startup Complete
[10.051s] [ext: isaacsim.asset.exporter.urdf-1.1.3] startup
2025-07-01 11:01:54 [11,284ms] [Error] [omni.kit.app._impl] [py stderr]: Traceback (most recent call last):
  File "/workspace/isaaclab/scripts/environments/teleoperation/teleop_se3_agent.py", line 333, in <module>
    main()
  File "/workspace/isaaclab/scripts/environments/teleoperation/teleop_se3_agent.py", line 137, in main
    env_cfg = parse_env_cfg(args_cli.task, device=args_cli.device, num_envs=args_cli.num_envs)
  File "/workspace/isaaclab/source/isaaclab_tasks/isaaclab_tasks/utils/parse_cfg.py", line 120, in parse_env_cfg
    cfg = load_cfg_from_registry(task_name, "env_cfg_entry_point")
  File "/workspace/isaaclab/source/isaaclab_tasks/isaaclab_tasks/utils/parse_cfg.py", line 82, in load_cfg_from_registry
    cfg_cls = cfg_entry_point()
  File "<string>", line 24, in __init__
  File "/workspace/isaaclab/source/isaaclab/isaaclab/utils/configclass.py", line 408, in _combined
    f1(*args, **kwargs)
  File "/workspace/isaaclab/source/isaaclab_tasks/isaaclab_tasks/manager_based/manipulation/pick_place/pickplace_g1_env_cfg.py", line 361, in __post_init__
    temp_urdf_output_path, temp_urdf_meshes_output_path = ControllerUtils.convert_usd_to_urdf(
  File "/workspace/isaaclab/source/isaaclab/isaaclab/controllers/utils.py", line 46, in convert_usd_to_urdf
    usd_to_urdf = UsdToUrdf.init_from_file(usd_path, **usd_to_urdf_kwargs)
  File "/isaac-sim/exts/isaacsim.asset.exporter.urdf/pip_prebundle/nvidia/srl/from_usd/to_urdf.py", line 129, in init_from_file
    usd_to_urdf = cls(stage, **kwargs)
  File "/isaac-sim/exts/isaacsim.asset.exporter.urdf/pip_prebundle/nvidia/srl/from_usd/to_urdf.py", line 74, in __init__
    self._graph = TransformGraph.init_from_stage(
  File "/isaac-sim/exts/isaacsim.asset.exporter.urdf/pip_prebundle/nvidia/srl/from_usd/transform_graph.py", line 340, in init_from_stage
    transform_graph._build_transform_graph()
  File "/isaac-sim/exts/isaacsim.asset.exporter.urdf/pip_prebundle/nvidia/srl/from_usd/transform_graph.py", line 686, in _build_transform_graph
    self._update_joint_node(prim)
  File "/isaac-sim/exts/isaacsim.asset.exporter.urdf/pip_prebundle/nvidia/srl/from_usd/transform_graph.py", line 816, in _update_joint_node
    raise ValueError(msg)
ValueError: The 'idx61_ee_l_joint''s joint transforms are not consistent.
global___joint_via_parent:
[[-0.00000009 -0.00000009  1.          0.13100001]
 [-1.         -0.00000071 -0.00000009  0.89390004]
 [ 0.00000071 -1.         -0.00000009  0.95349991]
 [ 0.          0.          0.          1.        ]]
global___joint_via_child:
[[-0.00000009 -0.00000009  1.          0.13099197]
 [-1.         -0.00000071 -0.00000009  0.89650011]
 [ 0.00000071 -1.         -0.00000009  0.95350003]
 [ 0.          0.          0.          1.        ]]**

RandomOakForest, 2025-07-02:
Thank you for posting this. The error `ValueError: The 'idx61_ee_l_joint''s joint transforms are not consistent` occurs when URDF conversion detects a mismatch between joint transforms calculated via parent and child links. Your matrices show minimal numerical differences (e.g., 0.13100001 vs. 0.13099197), suggesting a floating-point precision issue rather than a structural problem. Here are a few things to try:

1. **Disable joint merging via URDF attribute**
Add `dont_collapse="true"` to problematic joints in your URDF file:

```xml
<joint name="idx61_ee_l_joint" type="fixed" dont_collapse="true">
  <!-- joint parameters -->
</joint>
```
This prevents the converter from merging joints during USD conversion[^3][^8].

2. **Adjust conversion tolerance**
Modify the tolerance threshold in the URDF exporter's:

* Locate `transform_graph.py` in the URDF exporter module 
* Increase the `atol` parameter in the transform consistency check:

```python
# Original (typically 1e-5)
if not np.allclose(matrix_a, matrix_b, atol=1e-5): 

# Modified (increase tolerance)
if not np.allclose(matrix_a, matrix_b, atol=1e-3): 
```
   This bypasses floating-point precision issues[^3][^7].

3. **Use GUI-based conversion**
Convert URDF via Isaac Sim's GUI instead of scripts:
    - Enable `isaacsim.asset.importer.urdf` in **Window > Extensions**
    - Use **File > Import > URDF** with these settings:
        - Uncheck "Merge Fixed Joints"
        - Enable "Visualize Collisions"
GUI conversion handles numerical discrepancies more leniently[^1][^3][^4].
 

[^1]: https://docs.isaacsim.omniverse.nvidia.com/4.5.0/robot_setup/import_urdf.html

[^2]: https://docs.omniverse.nvidia.com/isaacsim/latest/features/external_communication/ext_omni_exporter_urdf.html

[^3]: https://github.com/isaac-sim/IsaacLab/issues/1800

[^4]: https://docs.isaacsim.omniverse.nvidia.com/4.5.0/robot_setup/ext_omni_exporter_urdf.html

[^5]: https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.sim.converters.html

[^6]: https://docs.robotsfan.com/isaaclab_official/main/_modules/isaaclab/sim/converters/urdf_converter.html

[^7]: https://isaac-sim.github.io/IsaacLab/main/_modules/isaaclab/sim/converters/urdf_converter.html

[^8]: https://isaac-sim.github.io/IsaacLab/main/source/how-to/import_new_asset.html

[^9]: https://github.com/isaac-sim/IsaacLab/blob/main/source/isaaclab_tasks/isaaclab_tasks/manager_based/manipulation/pick_place/pickplace_gr1t2_env_cfg.py

[^10]: https://nvidia-isaac-ros.github.io/concepts/manipulation/xrdf.html

ifmorning, 2025-07-02:
Thanks! The second method works!

closed 2025-07-02 by ifmorning

croustimsh, 2025-08-29:
Hello! I am having the same type of error as @hhaofan , Trying to export USD to URDF
my values seems to be very similar, probably a floating point error as well. 

the script @RandomOakForest is giving is the correct script to edit for our purpose, 
for me, setting the accuracy very low did not fix the issue, for testing purposes I ended up forcing the child values to the parents one 
at line 805:
`   global___joint_via_child_rot = Transform.get_rotation(global___joint_via_parent)`

obvioulsy its not a good idea to remove this checkup, but it can allow you to finish the conversion