isaac-sim/IsaacLab #2844: [Question] Question title

alaanagyi, 2025-07-02:
[image]
trying to teleoperate using so100, why this error occur while am sure articulation root is applied to robot?

RandomOakForest, 2025-07-03:
Thank you for posting this. The error "RuntimeError: Failed to find an articulation when resolving..." in Isaac Sim usually indicates that the USD file being used to define a robot (or other articulated object) doesn't have the USD ArticulationRootAPI applied to the correct prim. Could you share your USD?

alaanagyi, 2025-07-03:
I tried to open the usd file with usd composer and the articulationroot is correctly set to Robot prim. However, you can find the USD file attached

alaanagyi, 2025-07-03:
[so_100.zip](https://github.com/user-attachments/files/21033232/so_100.zip)

RandomOakForest, 2025-08-01:
Thanks for following up. Have you tried Isaac Sim 5.0? If you still face this issue with 5.0 please open a _bug report_. Follow #3021 to install. Thanks.

closed 2025-08-01 by RandomOakForest