isaac-sim/IsaacLab #2858: [Bug Report] h1_locomotion.py: No module named 'omni.kit.viewport'

sssn-tech, 2025-07-05:
Hi team,

I've encountered a bug when running the following script:

```
(is) ecs-user@iZuf6g6vii5ci3ikxssy9lZ:~/IsaacLab$ ./isaaclab.sh -p scripts/demos/h1_locomotion.py --headless
[...much info...]
Traceback (most recent call last):
  File "/home/ecs-user/IsaacLab/scripts/demos/h1_locomotion.py", line 50, in <module>
    from omni.kit.viewport.utility import get_viewport_from_window_name
ModuleNotFoundError: No module named 'omni.kit.viewport'
2025-07-05 16:25:35 [5,959ms] [Warning] [omni.fabric.plugin] gFabricState->gUsdStageToSimStageWithHistoryMap had 1 outstanding SimStageWithHistory(s) at shutdown
2025-07-05 16:25:35 [6,045ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
```

I followed the official instructions to install isaacsim and isaaclab, which is, to pip install isaacsim and git clone isaaclab.
Isaaclab seems good since I have already run some demo tasks, such as ants and Unitree robot, etc. But it broke this time.

I'll paste my issac versions bellow:
```
(is) ecs-user@iZuf6g6vii5ci3ikxssy9lZ:~/IsaacLab$ cat VERSION
2.1.0
(is) ecs-user@iZuf6g6vii5ci3ikxssy9lZ:~/IsaacLab$ pip show isaacsim
Name: isaacsim
Version: 4.5.0.0
Summary: A metapackage to manage the Isaac Sim installation
Home-page: https://developer.nvidia.com/isaac-sim
Author: Isaac Sim team
Author-email:
License: NVIDIA Proprietary Software
Location: /home/ecs-user/miniconda3/envs/is/lib/python3.10/site-packages
Requires: isaacsim-kernel
Required-by:
(is) ecs-user@iZuf6g6vii5ci3ikxssy9lZ:~/IsaacLab$
```

And here's all terminal output when crash happened:
```
(is) ecs-user@iZuf6g6vii5ci3ikxssy9lZ:~/IsaacLab$ ./isaaclab.sh -p scripts/demos/h1_locomotion.py --headless
[INFO] Using python from: /home/ecs-user/miniconda3/envs/is/bin/python
[INFO][AppLauncher]: Using device: cuda:0
[INFO][AppLauncher]: Loading experience file: /home/ecs-user/IsaacLab/apps/isaaclab.python.headless.kit
[Warning] [simulation_app.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [simulation_app.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Loading user config located at: '/home/ecs-user/miniconda3/envs/is/lib/python3.10/site-packages/omni/data/Kit/Isaac-Sim/4.5/user.config.json'
[Info] [carb] Logging to file: /home/ecs-user/miniconda3/envs/is/lib/python3.10/site-packages/omni/logs/Kit/Isaac-Sim/4.5/kit_20250706_002529.log
2025-07-05 16:25:29 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
2025-07-05 16:25:29 [2ms] [Warning] [omni.ext.plugin] [ext: rendering_modes] Extensions config 'extension.toml' doesn't exist '/home/ecs-user/IsaacLab/apps/rendering_modes' or '/home/ecs-user/IsaacLab/apps/rendering_modes/config'
2025-07-05 16:25:29 [130ms] [Warning] [omni.usd_config.extension] Enable omni.materialx.libs extension to use MaterialX
2025-07-05 16:25:30 [204ms] [Warning] [omni.platforminfo.plugin] failed to open the default display.  Can't verify X Server version.
2025-07-05 16:25:30 [445ms] [Warning] [omni.datastore] OmniHub is inaccessible
2025-07-05 16:25:30 [545ms] [Warning] [omni.isaac.dynamic_control] omni.isaac.dynamic_control is deprecated as of Isaac Sim 4.5. No action is needed from end-users.

|---------------------------------------------------------------------------------------------|
| Driver Version: 550.90.07     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA A10                       | Yes: 0 |     | 24564   MB | 10de      | 0          |
|     |                                  |        |     |            | 2236      | de967012.. |
|     |                                  |        |     |            | 0         |            |
|=============================================================================================|
| OS: 22.04.5 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.5, Kernel: 5.15.0-124-generic
| Processor: Intel(R) Xeon(R) Platinum 8369B CPU @ 2.90GHz
| Cores: 16 | Logical Cores: 32
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 188678 | Free Memory: 181261
| Total Page/Swap (MB): 0 | Free Page/Swap: 0
|---------------------------------------------------------------------------------------------|
Traceback (most recent call last):
  File "/home/ecs-user/IsaacLab/scripts/demos/h1_locomotion.py", line 50, in <module>
    from omni.kit.viewport.utility import get_viewport_from_window_name
ModuleNotFoundError: No module named 'omni.kit.viewport'
2025-07-05 16:25:35 [5,959ms] [Warning] [omni.fabric.plugin] gFabricState->gUsdStageToSimStageWithHistoryMap had 1 outstanding SimStageWithHistory(s) at shutdown
2025-07-05 16:25:35 [6,045ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
(is) ecs-user@iZuf6g6vii5ci3ikxssy9lZ:~/IsaacLab$ 
```

I hope informations above are helpful. Any suggestions or workarounds would be much appreciated. Thanks in advance!

kellyguo11, 2025-07-05:
Hi there, since this is an interactive demo, it was not designed to be run with `--headless`. Could you please try running without the flag? We can update the script to error out if running in headless.

RandomOakForest, 2025-07-05:
[like]  Pablo Hennings reacted to your message:

sssn-tech, 2025-07-05:
> Hi there, since this is an interactive demo, it was not designed to be run with `--headless`. Could you please try running without the flag? We can update the script to error out if running in headless.

It runs normally after removing `--headless`. Thank you.