isaac-sim/IsaacLab #2875: [Question] Combining different state machines

Ozzey, 2025-07-08:
### Question

Are there any documentations/pointers on how to combine different tasks (and state machines) which have the same task space.

For example combining Drawer-Open with Cube-Lift task to make a new long horizon task where the robot is supposed to open the cabinet and then pick up the cube.

RandomOakForest, 2025-07-11:
Thank you for posting this. This is something we are looking into and will be able to provide some examples later this year.

Ozzey, 2025-07-14:
> Thank you for posting this. This is something we are looking into and will be able to provide some examples later this year.

Thanks for the response! I wil try to work on this if I get the chance. 

Closing the issue.

closed 2025-07-14 by Ozzey