isaac-sim/IsaacLab #2886: [Question] How to do gravity compensation for mobile manipulators?

ShangqunYu, 2025-07-10:
### Question

Hey team,

We’re working with mobile manipulators and training a diffusion policy or fine-tuning a VLA model. When using a PD controller, we notice the arms sag under gravity unless we use very high stiffness (e.g., 25,000). I tried a stiffness of 2,500, but it didn’t suffice. In Isaac Sim, by contrast, stiffness values below 1,000 work fine.

We can’t disable gravity for the entire robot since it isn’t a fixed-base manipulator, and I’d like to avoid unrealistically high stiffness because it can introduce stability issues.

Questions:
1. Is there a way to disable gravity on specific parts of the robot (for example, the two arms)?
2. Why do Isaac Sim and Isaac Lab behave differently? Could it be that one uses degrees for joint angles and the other uses radians, causing the same angle error to produce different torque values?

Thanks!

RandomOakForest, 2025-07-14:
Thank you for posting this. It is possible to disable gravity for specific links or parts of a robot in both Isaac Sim and Isaac Lab:

- **Isaac Sim (USD/PhysX):**
    - You can disable gravity for individual rigid bodies or links using the Python API. For example, with the PhysX schema, you can apply:

```python
from pxr import PhysxSchema
physxAPI = PhysxSchema.PhysxRigidBodyAPI.Apply(prim)
physxAPI.CreateDisableGravityAttr(True)
```

This disables gravity for the specified prim (link or body) without affecting the rest of the robot[^1].
- **Isaac Lab:**
    - Isaac Lab exposes methods to set gravity flags for specific links. You can identify the indices of the links you want to affect (e.g., the arms) and set their gravity flags:

```python
gravity_flags = env.scene.articulations['robot'].root_physx_view.get_disable_gravities().clone()
for link_index in arm_bodies:
    gravity_flags[articulation_index, link_index] = 1  # 1 = disable gravity
env.scene.articulations['robot'].root_physx_view.set_disable_gravities(gravity_flags, indices)
```

This approach allows you to selectively disable gravity for the arms while keeping it enabled for the mobile base[^2].

**Note:** If your robot is defined as a single asset (e.g., a mobile base with arms in one URDF), you must programmatically identify and set the gravity flag for each relevant link.

Currently, the difference in required PD controller stiffness between Isaac Sim and Isaac Lab is primarily due to unit conventions for angles and controller parameters. In Isaac Sim, parameters such as joint angles, velocity limits, damping, and stiffness, are specified in **degrees** (deg, deg/s, 1/deg). Isaac Lab uses **radians** (rad, rad/s, 1/rad) for all internal computations and controller parameters. When Isaac Lab loads values from USD (which are in degrees), it converts them to radians. If you set PD gains in Isaac Lab, ensure they are in radian-based units. If you leave them as `None`, Isaac Lab will use the USD defaults and convert them internally. This difference in input units may change in future releases as the team is actively looking into it.

mi-mran, 2025-07-14:
@RandomOakForest thanks for the suggested fix! I'm new to Isaac Lab and I've encountered the same issue while loading an FR3 arm. I'd just like to clarify the solution you provided:

> gravity_flags = env.scene.articulations['robot'].root_physx_view.get_disable_gravities().clone()
for link_index in arm_bodies:
    gravity_flags[articulation_index, link_index] = 1  # 1 = disable gravity
env.scene.articulations['robot'].root_physx_view.set_disable_gravities(gravity_flags, indices)

How can I identify the 'articulation_index' and 'indices' variables?