isaac-sim/IsaacLab #2889: [Proposal] Add Deep Robotics robots to Isaaclab official github repo.

bopeng5221, 2025-07-10:
### Proposal

Add Deep Robotics robots to Isaaclab official github repo.

### Motivation

Deeprobotics is a well known robotics company and we have different types of robots that many people are curious about. We hope to add our robots to the Isaaclab github repo. Specifically, we want to add our new m20 robot which gained great attention around the world. Here's the link to the [video](https://www.youtube.com/watch?v=iL833P0Vino) We have the model and code ready but we want to confirm with you first cause this requires uploading our model to the ISAAC_NUCLEUS_DIR and adding a task in alongside classic examples like: source/isaaclab_tasks/isaaclab_tasks/direct/ant. We hope to add this with the release of Isaaclab v2.2 which is build with isaacsim 5.0.0. 

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)

ooctipus, 2025-07-12:
Hi @bopeng5221 

This is some very impressive work you guys have done!! Kudos! 
Thanks for your interest in contributing to IsaacLab, we are happy collaborate on this but we need to make sure to follow our legal criterion

For the assets uploading, what is your license file? We will need these license so we get proper approval before we uploading assets to our Bucket.

For the environment contribution, once we get the approval we can do a in depth review of the code. Yes it is very possible to add the task like classic examples if your environment provides some extinctive feature that differs  from the rest of available manager based environment . Though if your environment resembles our manager based environments, we might ask you and collaborate to put it there because they can scale much better than direct.

bopeng5221, 2025-07-14:
> Hi [@bopeng5221](https://github.com/bopeng5221)
> 
> This is some very impressive work you guys have done!! Kudos! Thanks for your interest in contributing to IsaacLab, we are happy collaborate on this but we need to make sure to follow our legal criterion
> 
> For the assets uploading, what is your license file? We will need these license so we get proper approval before we uploading assets to our Bucket.
> 
> For the environment contribution, once we get the approval we can do a in depth review of the code through PR. Yes it is very possible to add the task like classic examples if your environment provides some extinctive feature that differs from the rest of available manager based environment . Though if your environment resembles our manager based environments, we might ask you and collaborate to put it there because they can scale much better than direct.

Great! Let me discuss within the company and get back to you in a few days.

AntoineRichard, 2026-08-27:
@bopeng5221 thanks for offering to bring the Deep Robotics M20 assets into Isaac Lab — we'd still like to have them.

The one blocker is the asset license file, which legal review needs before we can host third-party assets. That detail was still open and the issue slipped through the cracks since. We're closing to keep the backlog tidy, not because we lost interest — please reopen or ping us here with the license terms and we'll restart the review right away.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard