isaac-sim/IsaacLab #2904: [Bug Report] Bug title

zh1224, 2025-07-11:
Thread 0x00007f32e642a740 (most recent call first):
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 1176 in create_module
  File "<frozen importlib._bootstrap>", line 571 in module_from_spec
  File "<frozen importlib._bootstrap>", line 674 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "/home/lzh/isaacsim/extscache/omni.kit.telemetry-0.5.1+d02c707b.lx64.r.cp310/omni/kit/telemetry/_itelemetry.py", line 7 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 883 in exec_module
  File "<frozen importlib._bootstrap>", line 688 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "/home/lzh/isaacsim/extscache/omni.kit.telemetry-0.5.1+d02c707b.lx64.r.cp310/omni/kit/telemetry/__init__.py", line 8 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 883 in exec_module
  File "<frozen importlib._bootstrap>", line 688 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "<frozen importlib._bootstrap>", line 1050 in _gcd_import
  File "/home/lzh/miniconda3/envs/env_isaaclab/lib/python3.10/importlib/__init__.py", line 126 in import_module
  File "/home/lzh/isaacsim/kit/kernel/py/omni/ext/_impl/custom_importer.py", line 76 in import_module
  File "/home/lzh/isaacsim/kit/kernel/py/omni/ext/_impl/_internal.py", line 94 in _import_public
  File "/home/lzh/isaacsim/kit/kernel/py/omni/ext/_impl/_internal.py", line 187 in _custom_importer
  File "/home/lzh/isaacsim/kit/kernel/py/omni/ext/_impl/_internal.py", line 211 in startup
  File "/home/lzh/isaacsim/kit/kernel/py/omni/ext/_impl/_internal.py", line 328 in startup_extension
  File "PythonExtension.cpp::startup()", line 2 in <module>
  File "/home/lzh/isaacsim/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py", line 427 in _start_app
  File "/home/lzh/isaacsim/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py", line 212 in __init__
  File "/home/lzh/IsaacLab/source/isaaclab/isaaclab/app/app_launcher.py", line 781 in _create_app
  File "/home/lzh/IsaacLab/source/isaaclab/isaaclab/app/app_launcher.py", line 131 in __init__
  File "/home/lzh/IsaacLab/scripts/tutorials/00_sim/create_empty.py", line 29 in <module>

Extension modules: psutil._psutil_linux, psutil._psutil_posix (total: 2)
./isaaclab.sh：行 274: 19056 段错误               （核心已转储） ${python_exe} "$@"

zh1224, 2025-07-11:
我一直在遇到这个问题，不知道该怎么解决

RandomOakForest, 2025-07-11:
Thank you for posting this. Could you provide the error messages you got and what you where running when this happened? Also please include the versions of Isaac Sim and Isaac Lab you are using and the link to the steps you used to install them.

zh1224, 2025-07-12:
I was running ./isaaclab.sh -p scripts/tutorials/00_sim/create_empty.py and there was a problem. My Issac sim version was 4.5.0, pytorch was 2.5.1, graphics card was 5070ti, Nvidia driver was 570., cuda was 11.8, and I installed them completely according to the following document: https://docs.robotsfan.com/isaaclab/source/setup/installation/pip_installation.html

RandomOakForest, 2025-08-16:
Following up, this is likely a setup issue. Please follow #3021 to reinstall and submit a new report if you still have any issues. Thank you.

closed 2025-08-16 by RandomOakForest