isaac-sim/IsaacLab #2919: [Bug Report] Terrain reference in Direct RL environments

Ashutosh781, 2025-07-13:
### Describe the bug

In my direct environment referenced from Anymal C direct environment here - [[Link to the line](https://github.com/isaac-sim/IsaacLab/blob/main/source/isaaclab_tasks/isaaclab_tasks/direct/anymal_c/anymal_c_env.py#L66)], the terrain is created from the config and set as reference to `self._terrain`. But when calling the `reset_root_state_uniform` event it polls `env.scene.env_origins` and gets default origins instead of terrain origins. This bug is caused because `env.scene._terrain` is None when created through the Direct environment. This can be fixed by adding a line `self.scene._terrain = self._terrain` so that the terrain variable inside the scene object also references the generated terrain.

This bug is unique to the Direct environment, and this line needs to be added manually. This bug doesn't occur in a manager-based environment because the interactive scene does [this](https://github.com/isaac-sim/IsaacLab/blob/main/source/isaaclab/isaaclab/scene/interactive_scene.py#L656).

### Steps to reproduce

Direct environment with Rough terrain similar to Anymal C environment, and added the `reset_root_state_uniform` event. Where the terrain was higher than 0.0 in the z-coordinate, the base of the robot was always being initialized inside the terrain and causing resets.

### System Info

Describe the characteristic of your environment:

- IsaacLab Commit: 25f7a5d
- Isaac Sim Version: 4.5
- OS: Ubuntu 22.04
- GPU: RTX 3060
- CUDA: 12.2
- GPU Driver: 535.230.02

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

RandomOakForest, 2025-07-16:
Thank you for catching this and posting the report. We will review.

StafaH, 2026-09-24:
Thanks @Ashutosh781 for reporting that `reset_root_state_uniform` used the wrong origins with rough terrain in the Anymal C direct task.

This is addressed on `develop`: the direct task now declares its terrain in `AnymalCRoughSceneCfg`, so `InteractiveScene` owns it and `env.scene.env_origins` comes from that terrain. The change landed in https://github.com/isaac-sim/IsaacLab/pull/7608. Please reopen if a direct environment using the current scene configuration still places the robot inside rough terrain after reset.

closed 2026-09-24 by StafaH