isaac-sim/IsaacLab #2922: [Question] USD Terrain Collision and Import issue

rushabhdhoke, 2025-07-14:
### Question

Can anyone share how to import a USD terrain in isaacLab environment. I am trying to run a manager based Locomotion Velocity task for the GO2 quadruped. I have a custom terrain on which I want to train the quadruped to walk. 

When I visualize the scene, I can see the robots and the terrain both present on the screen, but the robots fall through the surface of the terrain without colliding. I have manually enabled collider, both in the USD file as well as the terrain_importer in the config file.

If you have previously imported a USD file in to your environment please guide me. Thanks

rushabhdhoke, 2025-07-15:
Hi, I tried this but it did not work.. 

Can you please take a look, if you are able to do it?
[USD_File](https://drive.google.com/file/d/1ZEmE6JZyIH8JRANSuzjjOClU0Y8Z0-td/view?usp=sharing)

I can share my code as well, if that's helpful.

> Can you try something like this
> 
> ```
> robot_cfg: ArticulationCfg = ArticulationCfg(
>         prim_path="/World/envs/env_.*/Robot",
>         spawn=sim_utils.UsdFileCfg(
>             usd_path=ROBOT_CONFIGS["jackal"]["usd_path"],
>         ),
>         actuators={
>             "wheel_acts": ImplicitActuatorCfg(
>                 joint_names_expr=ROBOT_CONFIGS["jackal"]["wheel_joint_expr"],
>                 damping=None,
>                 stiffness=None
>             )
>         },
>         debug_vis=False
>     
>     )
> ```
> 
> And Terrain
> 
> ```
>  terrain_cfg: TerrainImporterCfg = TerrainImporterCfg(
>         prim_path="/World/ground",
>         terrain_type="usd",
>         usd_path=f"{ISAAC_NUCLEUS_DIR}/Environments/Simple_Warehouse/warehouse_with_forklifts.usd",
>         collision_group=-1,
>         physics_material=sim_utils.RigidBodyMaterialCfg(),
>         debug_vis=False,
>     )
> ```

tohsin, 2025-07-15:
I am currently training on my GPU and will take a while for me to test your USD file.

tohsin, 2025-07-15:
Just ran my env and Robot in your USD Not exactly sure what the world the USD is supposed to be, but yeah, my robot is also just kept on falling