isaac-sim/IsaacLab #2931: Interpenetration problem

kaijieshi7, 2025-07-14:
Hello, Isaac team.

I’m using a dexterous robotic hand to grasp an object, but I keep running into interpenetration issues. The tip part of the thumb is inside the object. I’ve tweaked a lot of parameters in an attempt to resolve it, yet the problem persists. Below you’ll find a video demonstrating the issue, along with the relevant code and asset files.

Could you help me solve it?

https://github.com/user-attachments/assets/bdb1d5a1-3ae4-40c0-9743-64d303134cf9

[github_issue.zip](https://github.com/user-attachments/files/21217057/github_issue.zip)

kaijieshi7, 2025-07-14:
Hi @Viru97. I tried sdf, and it slightly alleviates the penetration issue. Does that mean we can’t guarantee 100% that there will be no penetration?

RandomOakForest, 2025-08-04:
Thank you for posting this bug report. This is a known issue and the team is working on it. In the meantime, you may try moving to CPU-only physics.

closed 2025-08-04 by kaijieshi7