isaac-sim/IsaacLab #2939: [Question] Something wrong with my ManagedRL training

AndreaRossetto, 2025-07-15:
### Something wrong with my ManagedRL

I'm trying to make working the ManagedRL with a custom model, but during the training seems the network is not moving the joints. I expect the joint to be moved more than this:

https://github.com/user-attachments/assets/4c33cc5c-55b4-4ded-bf62-271504b12c1b

If I print the actions I can see the values change but not so much.
I made following the isaaclab examples.

AndreaRossetto, 2025-07-16:
If I reduce the number of enviroments from (100 to 10) the model seems moving the joint... so now the question is why if I use more envs I get the previous result? maybe is only   my impression or there's something wrong?

https://github.com/user-attachments/assets/83bb7ba6-444d-40da-9011-1ce4e88487f6

AndreaRossetto, 2025-07-16:
Problem solved!

closed 2025-07-16 by AndreaRossetto

RandomOakForest, 2025-07-16:
Thanks for following up. It would be great if you could share your solution. Thanks.