isaac-sim/IsaacLab #2972: [Question] Differences in camera image in Isaac Lab and Isaac Sim for the same settings

cowcnovo, 2025-07-17:
I have a certain project where I am creating a camera in Isaac Lab. I have another project where I am creating the same camera in an Isaac Sim Replicator script. The camera settings are identical as shown in the code below:

Isaac Sim:
```
cam = rep.create.camera(focal_length=1.93, focus_distance=0.8, horizontal_aperture=3.896, clipping_range=(0.1, 1000000))
with cam:
  rep.modify.pose(
      position=(1, 0, 1.85)
  )
```

Isaac Lab:
```
camera = TiledCameraCfg(
    prim_path="{ENV_REGEX_NS}/camera",
    data_types=["rgb"],
    spawn=sim_utils.PinholeCameraCfg(
        focal_length=1.93, focus_distance=0.8, horizontal_aperture=3.896,
    ),
    width=1280,
    height=960,
    update_period=1/20,
    offset=CameraCfg.OffsetCfg(
        pos=(1.0, 0.0, 1.85), 
        rot=(-0.24184, 0.66446, 0.66446, -0.24184),
    ),
)
```

The orientations and positions of the cameras are the same. And, I have created scenes in both the Isaac Sim and Isaac Lab projects which are identical too (apart from the configuration of the robot and the number of objects on the tray of course). The comparison between these can be seen in the photos below.

**My question**: Why are these two cameras showing different views in terms of zoom, viewport etc? All camera properties seem to be the same, and the image dimensions too.

**Note**: The vertical apertures are not the same, but I saw and changing them makes no difference to the view.

[image]
[image]

RandomOakForest, 2025-07-17:
Thank you for posting this. How are you setting the `render_product` call for Isaac Sim. Is it the following?
```python
render_product = rep.create.render_product(camera, (1280, 960))
```
Could you try the following? this may be all is necessary for Isaac Sim to match the `TiledCamera`:
```python
import omni.replicator.core as rep

# Camera creation with matching intrinsics and pose
camera = rep.create.camera(
    focal_length=1.93,
    focus_distance=0.8,
    horizontal_aperture=3.896,
    clipping_range=(0.1, 1000000)  # Set to a wide range; adjust if needed
)

# Set camera pose using quaternion rotation (x, y, z, w)
with camera:
    rep.modify.pose(
        position=(1.0, 0.0, 1.85),
        rotation=(-0.24184, 0.66446, 0.66446, -0.24184)  # Quaternion (x, y, z, w)
    )

# Create the render product with desired resolution (width, height)
render_product = rep.create.render_product(camera, (1280, 960))
```

RandomOakForest, 2025-08-22:
Thank you for following up. I'm going to move this post to our _Discussions_ section to continue and for others to contribute.