isaac-sim/IsaacLab #2973: [Where is URLift] Isaac Lab tutorials Train Your Second Robot in Isaac Lab

AAndroid96, 2025-07-17:
I am working on the, “Train Your Second Robot in Isaac Lab” and I am trying to locate the source/URLift/URLift/tasks/manager_based/urlift/init.py file but this does not exist when I created the project. I am running into issues running the, “python scripts/zero_agent.py --task Template-Reach-v0 --num_envs=10” command because of this error:

[14.155s] Simulation App Startup Complete
Traceback (most recent call last):
File “C:\IsaacLab\Reach\Reach\scripts\zero_agent.py”, line 70, in
main()
File “C:\IsaacLab\Reach\Reach\scripts\zero_agent.py”, line 44, in main
env_cfg = parse_env_cfg(
File “c:/isaaclab/isaaclab/source/isaaclab_tasks/isaaclab_tasks/utils/parse_cfg.py”, line 138, in parse_env_cfg
cfg = load_cfg_from_registry(task_name.split(“:”)[-1], “env_cfg_entry_point”)
File “c:/isaaclab/isaaclab/source/isaaclab_tasks/isaaclab_tasks/utils/parse_cfg.py”, line 55, in load_cfg_from_registry
cfg_entry_point = gym.spec(task_name.split(“:”)[-1]).kwargs.get(entry_point_key)
File “C:\IsaacLab\env_isaaclab\lib\site-packages\gymnasium\envs\registration.py”, line 1008, in spec
_check_version_exists(ns, name, version)
File “C:\IsaacLab\env_isaaclab\lib\site-packages\gymnasium\envs\registration.py”, line 399, in _check_version_exists
_check_name_exists(ns, name)
File “C:\IsaacLab\env_isaaclab\lib\site-packages\gymnasium\envs\registration.py”, line 376, in _check_name_exists
raise error.NameNotFound(
gymnasium.error.NameNotFound: Environment Template-Reach doesn’t exist. Did you mean: Template-Reach-Play?
2025-07-14 04:18:47 [14,114ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Annotators’ for removal
2025-07-14 04:18:47 [14,114ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Annotators’ for removal
2025-07-14 04:18:47 [14,114ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Annotators’ for removal
2025-07-14 04:18:47 [14,114ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Annotators’ for removal
2025-07-14 04:18:47 [14,115ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Annotators’ for removal
2025-07-14 04:18:47 [14,115ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Annotators’ for removal
2025-07-14 04:18:47 [14,115ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Annotators’ for removal
2025-07-14 04:18:47 [14,115ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Annotators’ for removal
2025-07-14 04:18:47 [14,115ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Annotators’ for removal
2025-07-14 04:18:47 [14,115ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Annotators’ for removal
2025-07-14 04:18:47 [14,115ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Annotators’ for removal
2025-07-14 04:18:47 [14,115ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Annotators’ for removal
2025-07-14 04:18:47 [14,115ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Core’ for removal
2025-07-14 04:18:47 [14,116ms] [Warning] [omni.graph.core.plugin] Could not find category ‘animation’ for removal
2025-07-14 04:18:47 [14,128ms] [Warning] [omni.fabric.plugin] gFabricState->gUsdStageToSimStageWithHistoryMap had 1 outstanding SimStageWithHistory(s) at shutdown
2025-07-14 04:18:47 [14,129ms] [Warning] [carb.audio.context] 1 contexts were leaked
2025-07-14 04:18:47 [14,254ms] [Warning] [carb] Recursive unloadAllPlugins() detected!

RandomOakForest, 2025-07-19:
Thank you for posting this. Could you share the steps you followed before this? You are probably forgetting 
```
python -m pip install -e source/<given-project-name>
```
See [this tutorial for reference](https://isaac-sim.github.io/IsaacLab/main/source/overview/developer-guide/template.html#template-generator).

AAndroid96, 2025-07-23:
Hello RandomOakForest, 

When I created the project following the instructions on "Creating an Isaac Lab template project" I did as follows: 

Open a new terminal.

Navigate to the Isaac Lab folder.

For example, cd ~/IsaacLab
Activate your environment. If you already have it activated from installing Isaac Lab, or you’re using our Brev Launchable, you can skip this step.

If using conda:
conda activate env_isaaclab

If using venv:
Linux: source env_isaaclab/bin/activate
Windows: python3.10 -m venv env_isaaclab

Run the Isaac Lab script with the --new argument to create the template project:
Linux: ./isaaclab.sh --new
Windows: .\isaaclab.bat --new

The template generator will ask you a few questions. In this command-line menu, use arrows to move and the spacebar to select an option, then press enter to choose the option.

Task type: select External.
Project path: set this where you’d like, just make note of the folder.
Project name: give the project a name such as Reach
Isaac Lab Workflow: choose the Manager-based | single-agent workflow.
RL library: choose skrl for the reinforcement learning library.
RL algorithms: choose PPO, an abbreviation for [Proximal Policy Optimization](https://spinningup.openai.com/en/latest/algorithms/ppo.html).

**Install the project**
Change directory into the new project folder: Linux: cd ~/Reach
Windows: dir {directory of project}

Install your external project by running the following command. This installs the package in “editable” mode, which is useful during development. Notice our project name being used at the end of the command.

python -m pip install -e source/Reach
To confirm the project was installed, run this command to list installed environments.
python scripts/list_envs.py
Confirm that our project is listed:
Great! At this point we have both a robot USD file ready, and a project for our training code. Now we begin bringing the two together, by implementing configuraiton and logic for our training in the Isaac Lab framework.

RandomOakForest, 2025-08-16:
Thank you for following up. You may want to try the latest versions of Isaac Lab. Follow #3021 to update. I'll move this post to our _Discussions_ for follow up by the team and others.