isaac-sim/IsaacLab #2987: [Question] How to directly modify the transform of a sub-prim within a RigidBody?

SevenFo, 2025-07-20:
Based on the documentation, a `RigidBody` is defined as "A rigid body is a single body that can be simulated by PhysX." To achieve this, the `UsdPhysics.RigidBodyAPI` is applied to the root prim of the asset. The `RigidBody` class then provides methods to modify the `root_state` (position, orientation, velocity) of the entire body.

My question arises when a `RigidBody` prim has child prims (sub-prims). Since the parent prim already possesses the `RigidBodyAPI`, it's not possible to apply the same API to its children. This is expected, as the entire hierarchy is treated as a single rigid entity by the physics engine.

However, I would like to know if there is a way to directly and kinematically modify the transform (i.e., position and orientation) of these sub-prims, independent of the parent's rigid body simulation, and without having to convert the asset into an Articulation. I am looking for a purely kinematic update for these sub-prims, as I don't require the dynamics (e.g., forces, joints) that an Articulation would introduce.

Is there a recommended method in IsaacLab to set the local transform of a sub-prim that is part of a larger `RigidBody`? Or is the transform of all sub-prims strictly determined by the rigid-body state of the root prim once simulation starts?

Thank you for your guidance.

RandomOakForest, 2025-07-26:
Thank you for posting this. The short answer is it seems this is not currently supported, so there's not a recommended way to do so in Isaac Lab. Having said that, here's a summary that may be helpful, as a reference.

When a USD prim has the `UsdPhysics.RigidBodyAPI` applied to its root (making it a PhysX RigidBody in Isaac Lab or Isaac Sim), all sub-prims become part of a single, rigid entity from the perspective of the physics simulation. This means that, during simulation, the entire hierarchy moves as one: sub-prims' transforms are rigidly determined by their initial local offset relative to the root prim and will maintain this relationship while the parent is simulated as a physics object[^1][^2][^3].

**Key Points:**

- **Physics Determinism:** Once simulation begins, the transforms of sub-prims (children) are determined by the parent's rigid-body world transform and their own initial local transform. You cannot independently move or animate these children at runtime via direct transform updates—the simulation overwrites such changes since the entire hierarchy is considered one rigid body[^1][^4][^3].
- **No Nested Rigid Bodies:** The USD and Omniverse physics system does not support nested rigid bodies, so it's not possible (nor allowed) to apply `RigidBodyAPI` to child prims of a rigid body parent—the system will ignore such attempts[^5][^3].
- **Kinematic Updates:** Applying purely kinematic updates (i.e., setting position or orientation at runtime without involving forces or joints) is only possible at the root prim level when it is marked as kinematic. Sub-prims cannot be kinematically updated independently; their motion is inherently tied to the root prim's state in the physics simulation[^1][^5][^3].
- **Possible Workarounds:**
    - If you require independently movable components, the canonical approach is to break the rigid object into multiple rigid bodies and (if needed) connect them with joints. However, this does introduce articulation dynamics (forces, constraints) which you explicitly do not want.
    - *ResetXformStack Exception:* There is a technical exception in the USD specification—if a sub-prim introduces a `resetXformStack` operation, it can break transform inheritance, allowing that child to be treated independently (and potentially even become its own rigid body). But, in practice (and as of the latest Isaac Lab and Isaac Sim releases), this pattern is not commonly used for direct per-child kinematic updates and may not be robust or recommended for most applications[^3][^1].
    - For visual-only effects (not affecting physics), you could animate visuals (e.g., meshes), but the collision and physical state will still be determined by the rigid body hierarchy.

**Isaac Lab/Isaac Sim Best Practices:**

- **RigidBody Hierarchy:** All sub-prims are rigidly attached once the root is a rigid body. The "single rigid body" rule is strictly enforced—children cannot be independently moved at runtime via transform APIs[^1][^2][^3].
- **Articulations for Flexibility:** If you need independent kinematic control, use an Articulation and joints—even if you do not use dynamics, this gives you the per-link kinematic control you desire.
- **No Supported Low-Level API** for directly updating a sub-prim’s transform independently of a rigid-body parent in the physics simulation for purely kinematic, local updates in Isaac Lab.

**Summary Table:**

| Feature | Supported in RigidBody Hierarchy? |
| :-- | :--: |
| Move root prim kinematically in simulation | Yes (set kinematic flag) |
| Move child (sub-prim) kinematically in simulation | **No** |
| Animate visuals only (not physics) | Possible, but not physically valid |
| Independent per-child transform at runtime | Not supported (requires articulation) |

**References to NVIDIA/Omniverse Documentation:**

- UsdPhysics Rigid Body API Description: "all children of the marked prim are assumed to be part of this rigid body, and move rigidly along with the body... it is not possible to have nested bodies..."[^1][^3].
- Isaac Lab asset documentation and rigid object rules reinforce that the parent’s state determines the child transforms during simulation[^6][^5].
- Isaac Sim physics pages and PhysX schemas state that schema properties cannot be applied to children under a rigid body, and sub-prims cannot be promoted during play to separate kinematic or dynamic entities[^2][^4][^5][^7].

**Conclusion:**
Once a USD hierarchy is made into a RigidBody (by applying `UsdPhysics.RigidBodyAPI` at the root in Isaac Lab or Isaac Sim), you cannot directly and independently update the transform of its child prims in a kinematic (physics-valid) way without using articulations or fundamentally breaking up the object. All sub-prims will strictly follow the parent’s rigid-body transform during simulation[^1][^5][^3].

[^1]: https://openusd.org/dev/api/usd_physics_page_front.html

[^2]: https://docs.omniverse.nvidia.com/kit/docs/omni_physics/107.0/dev_guide/rigid_bodies_articulations/rigid_bodies.html

[^3]: https://docs.omniverse.nvidia.com/kit/docs/omni_usd_schema_physics/106.1/index.html

[^4]: https://docs.omniverse.nvidia.com/kit/docs/omni_physics/107.3/dev_guide/deformables_beta/omniphysics_deformable_schema.html

[^5]: https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.sim.html

[^6]: https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.assets.html

[... cut ...]
[^30]: https://behavior.stanford.edu/omnigibson/reference/systems/macro_particle_system.html

[^31]: https://forums.developer.nvidia.com/t/i-dont-understand-the-prim-cannot-be-added-as-a-child-to-another-prim/230787

[^32]: https://discussions.unity.com/t/moveposition-in-parent-child-relations-of-rigidbodies/393869

[^33]: https://github.com/isaac-sim/IsaacLab/discussions/683

[^34]: https://discussions.unity.com/t/parented-kinematic-rigidbodies-moving-independently-when-parent-is-nudged/712428

[^35]: https://www.sidefx.com/docs/hdk/prim_8h_source.html

SevenFo, 2025-08-04:
Thank you for your reply. It's much clearer now!

closed 2025-08-04 by SevenFo