isaac-sim/IsaacLab #3041: WebRTC Visualization Error in Isaac Lab 2.1 Docker Container

yk785496878, 2025-07-29:
docker pull [nvcr.io/nvidia/isaac-lab:2.1.0](http://nvcr.io/nvidia/isaac-lab:2.1.0)

in docker :

KirdenKe, 2025-07-29:
I encountered a similar issue, but it occurred when I executed the following command: ```LIVESTREAM=2 ./isaaclab.sh -p scripts/tutorials/00_sim/launch_app.py --size 0.5```.

Env setting:
* NVIDIA RTX A6000
* Driver Version: 535.247.01

Then I noticed that if I ran ```runheadless.sh```, it connected successfully. It seems that the disconnect issue only occurs when the livestream is opened via the python file.

Currently, the only way to run the program without issues is to add the ```--headless``` flag.

yk785496878, 2025-07-30:
I added --headless, but the running result remains the same in docker

[58.819s] Simulation App Startup Complete
[59.031s] [ext: omni.physx.fabric-106.5.7] startup
[INFO]: Setup complete...
main: thread_init: already added for thread
2025-07-30 08:53:20 [63,024ms] [Warning] [carb.livestream-rtc.plugin] onClientResizeRequested: Stream can't be resized, please make sure that /app/livestream/allowResize is set to true.
2025-07-30 08:53:36 [79,044ms] [Warning] [carb.livestream-rtc.plugin] onClientEventRaised: NVST_CCE_DISCONNECTED when m_connectionCount 0 != 1
main: thread_init: already added for thread
main: thread_init: already added for thread
2025-07-30 09:11:45 [1,167,375ms] [Warning] [carb.livestream-rtc.plugin] onClientResizeRequested: Stream can't be resized, please make sure that /app/livestream/allowResize is set to true.
2025-07-30 09:11:48 [1,170,987ms] [Warning] [carb.livestream-rtc.plugin] onClientEventRaised: NVST_CCE_DISCONNECTED when m_connectionCount 4294967295 != 1
2025-07-30 09:11:48 [1,170,992ms] [Warning] [carb.livestream-rtc.plugin] onClientEventRaised: NVST_CCE_DISCONNECTED when m_connectionCount 4294967294 != 1
main: thread_init: already added for thread
main: thread_init: already added for thread
2025-07-30 09:12:59 [1,241,963ms] [Warning] [carb.livestream-rtc.plugin] onClientResizeRequested: Stream can't be resized, please make sure that /app/livestream/allowResize is set to true.
2025-07-30 09:13:15 [1,257,974ms] [Warning] [carb.livestream-rtc.plugin] onClientEventRaised: NVST_CCE_DISCONNECTED when m_connectionCount 4294967293 != 1

RandomOakForest, 2025-08-16:
You may want to try the latest version of Isaac Lab. Follow #3021 to update. I'll move this issue to our _Discussions_ for follow up.