isaac-sim/IsaacLab #3043: [Question] Low GPU Utilization vs High Single-Core CPU Usage During RL Training

LuyiLi, 2025-07-29:
## Low GPU Utilization  vs High Single-Core CPU Usage During RL Training

### Description
When testing the reinforcement learning training tasks in Isaac Lab, I consistently observe two phenomena:

1. GPU usage rarely reaches 100% (typically fluctuates between 80%), and Power consumption stays relatively low (Less than 50% for a 5090).

2. One logical core seems to be consistently at 100% load (observed via htop and system monitors) and other cores remain underutilized (idle or at low usage).

### Key Questions
Which component of the training pipeline is causing this single-core CPU bottleneck?

## Environment Details
### System:

**GPU:** RTX 5090, driver 575.57.08

**CPU:** AMD Ryzen 9800x3D

**Operating System:** Ubuntu 24.04 LTS

### Version:

**Isaac Lab version:** main branch

**Isaac Sim version:** 4.5.0

## Testing results

`python3 scripts/reinforcement_learning/rsl_rl/train.py --task=Isaac-Velocity-Flat-Anymal-C-v0 --headless`

[image]

[image]

`python3 scripts/reinforcement_learning/rsl_rl/train.py --headless --task=Isaac-Velocity-Rough-Unitree-Go2-v0`

[image]

[image]

`python3 scripts/reinforcement_learning/rsl_rl/train.py --headless --task=Isaac-Velocity-Rough-G1-v0`

[image]

[image]

`python3 scripts/reinforcement_learning/rsl_rl/train.py --headless --task=Isaac-Velocity-Flat-G1-v0`

[image]

[image]

`python3 scripts/reinforcement_learning/rsl_rl/train.py --headless --task=Isaac-Velocity-Flat-Spot-v0`

[image]

[image]

Currently, I don't have any other device I can test on. So I am not certain if this is a problem from the training pipline or it just happens on my setup. 

Any insights into what might be causing this CPU bottleneck? Additionally, it would be helpful to know if others observe similar behavior on different hardware configurations.

Thanks!

RandomOakForest, 2025-08-01:
Thank you for posting this. The team is reviewing it.

Josh00-Lu, 2025-08-02:
This is also observed by our team. The power consumption stays relatively low. It seems that the performance bottleneck is on CPU? That is quite strange.

Hope IssacLab Team could solve that. Thank!

K1wi-pcy, 2025-08-14:
I am using U7 265K+5070Ti, and there will be an error related to the core issue at startup.

[image]

RandomOakForest, 2025-08-19:
Following up, regarding the CPU utilization, during training when there is no rendering, PhysX simulation is at task utilizing GPU compute ops. Because the training doesn't use CPU multi-threading currently, you will see high utilization on a single CPU core. To improve GPU utilization, the team is working on adding a few features to address this.