isaac-sim/IsaacLab #3054: [Question] Hands/objects penetrating through each other during G1 teleoperation

edisimo, 2025-07-30:
### Problem

I'm working on a custom environment using a G1 humanoid robot (with Inspire hands) in IsaacLab for teleoperation and dataset recording. During live teleoperation using teleop_se3_agent.py, I noticed that the robot’s hands often penetrate into objects (e.g., tools, boxes) or the objects penetrate each other. 

I have experimented with various combinations of solver position / velocity iteration count, different sim.dt
I have sdf collision meshes on my objects since i need precise manipulation

### Question

Is there any way to prevent object penetration? 
Do you have any idea what I could try to tamper with in order to prevent this unwanted behaviour?

### Info

IsaacLab: v2.2 (feature/isaacsim_5_0)
Isaac Sim: 5.0 (prerelease)
Task: custom Pick&Place task
Control device: Apple Vision Pro (via handtracking)

edisimo, 2025-08-04:
@RandomOakForest  @rwiltz 

I was able to replicate this bug using the original pickplace_gr1t2_env_cfg.py configuration script with an object that was spawned this way: 

```
object = RigidObjectCfg(
        prim_path="{ENV_REGEX_NS}/Object",
        init_state=RigidObjectCfg.InitialStateCfg(
            pos=[0, 0.45, 1.145],
            rot=[1, 0, 0, 0]
        ),
        spawn=sim_utils.CylinderCfg(
            radius=0.1,
            height=0.2,
            rigid_props=sim_utils.RigidBodyPropertiesCfg(),
            mass_props=sim_utils.MassPropertiesCfg(mass=2.0),
            collision_props=sim_utils.CollisionPropertiesCfg(),
            physics_material=sim_utils.RigidBodyMaterialCfg(
                static_friction=0.5,
                dynamic_friction=0.5,
                restitution=0.1,
            ),
        ),
    )
```
This object has primitive cylindrical collisions, yet I was able to replicate it. It becomes worse with more complex objects when using the SDF collision type.

This was the command I had used for the data collection: 

```
/isaaclab.sh -p scripts/tools/record_demos.py     --task Isaac-PickPlace-GR1T2-Abs-v0   --teleop_device handtracking     --enable_pinocchio  --device cpu --num_demos 1 --dataset_file ./datasets/gr1/test/test_penetration.hdf5 --enable_cameras
```
I have also added a head camera and a wrist camera just for the videos I have uploaded that are attached to this comment, however I was also able to replicate the bug exactly the same way without those cameras, so I don't think the issue was caused by them.  The cameras only slowed down the simulation frame rate, but FYI, I added them this way:

```
head_camera = CameraCfg(
        prim_path="/World/envs/env_.*/Robot/head_yaw_link/HeadCamera",
        height=1080,
        width=1920,
        data_types=["rgb"],
        spawn=PinholeCameraCfg(focal_length = 6.0),
        offset=CameraCfg.OffsetCfg(
                pos=(0.11, 0.0, 0.05),  
                rot=(0.65328, 0.2706, -0.2706, -0.65328),  # Forward, looking on object
                convention="opengl", 
        ),
    )

right_wrist_camera = CameraCfg(
        prim_path="/World/envs/env_.*/Robot/right_hand_roll_link/RightWristCamera",
        height=720,
        width=1280,
        data_types=["rgb"],
        spawn=PinholeCameraCfg(focal_length = 6.0),
        offset=CameraCfg.OffsetCfg(
                pos=(0.0, -0.1, -0.1),
                rot=(0, 0.0, 0.6427, -0.7660),  
                convention="opengl", 
        ),
    )

```

And their respective observations: 

```
head_camera_rgb = ObsTerm(
            func=base_mdp.image,
            params={"sensor_cfg": SceneEntityCfg("head_camera"), "data_type": "rgb", "normalize": False},
        )

        right_wrist_camera_rgb = ObsTerm(
            func=base_mdp.image,
            params={
                "sensor_cfg": SceneEntityCfg("right_wrist_camera"),
                "data_type": "rgb",
                "normalize": False,
            },
        )
```

Hope this additional information helps :)

https://github.com/user-attachments/assets/a5d42452-d43d-46fa-bb71-9821738f58ab
https://github.com/user-attachments/assets/a7ec288f-840f-4923-aa1e-dee69984894d

sashansamarajeewa, 2025-11-23:
Hi @edisimo ,

Were you able to find a solution for this problem?

I am also facing a similar issue https://github.com/isaac-sim/IsaacLab/issues/3993

AntoineRichard, 2026-08-27:
@edisimo thanks for the clear report on hands and objects interpenetrating during G1 teleoperation.

Worth flagging that the shipped pick-place configs already run `sim.dt = 1/120` with `decimation = 6` — including the `pickplace_gr1t2_env_cfg.py` you reproduced this on — so step size alone won't explain it, and your switch to convex decomposition is the real lever. Isaac Sim 6.0's OpenUSD tuning series is built on this exact hand: https://docs.isaacsim.omniverse.nvidia.com/6.0.0/openusd_tuning_tutorials/index.html — tutorial 3 covers hull-vs-decomposition at the fingertips, tutorial 4 covers collision-pair filtering. Closing as answered — please reopen if you find a reproducible Isaac Lab-side defect.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard