isaac-sim/IsaacLab #3063: [Bug Report] "Adding sensor on a robot tutorials" related file path is wrong

Rango96, 2025-07-31:
### Describe the bug

When I follow the tutorial of "[Adding sensor on a robot ](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/04_sensors/add_sensors_on_robot.html)". I run the script as described in the sectiion of  "The Code Execution" by command `./isaaclab.sh -p scripts/tutorials/04_sensors/add_sensors_on_robot.py --num_envs 2 --enable_cameras`.
But there is something wrong with this step, and the output in the terminal saying that:
```
`(env_lab) workstation@workstation-MS-7E06:~/IsaacLab$ ./isaaclab.sh -p scripts/tutorials/04_sensors/add_sensors_on_robot.py --num_envs 2 --enable_cameras
[INFO] Using python from: /home/workstation/anaconda3/envs/env_lab/bin/python                                                                                    
[INFO][AppLauncher]: Using device: cuda:0
[INFO][AppLauncher]: Loading experience file: /home/workstation/IsaacLab/apps/isaacsim_4_5/isaaclab.python.rendering.kit
[Warning] [simulation_app.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [simulation_app.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Loading user config located at: '/home/workstation/anaconda3/envs/env_lab/lib/python3.10/site-packages/omni/data/Kit/Isaac-Sim/4.5/user.config.json'
[Info] [carb] Logging to file: /home/workstation/anaconda3/envs/env_lab/lib/python3.10/site-packages/omni/logs/Kit/Isaac-Sim/4.5/kit_20250731_223346.log
2025-07-31 14:33:46 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
2025-07-31 14:33:46 [1ms] [Warning] [omni.ext.plugin] [ext: rendering_modes] Extensions config 'extension.toml' doesn't exist '/home/workstation/IsaacLab/apps/isaacsim_4_5/rendering_modes' or '/home/workstation/IsaacLab/apps/isaacsim_4_5/rendering_modes/config'

|---------------------------------------------------------------------------------------------|
| Driver Version: 575.64.03     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 5080          | Yes: 0 |     | 16303   MB | 10de      | 0          |
|     |                                  |        |     |            | 2c02      | 8279cf94.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: 24.04.2 LTS (Noble Numbat) ubuntu, Version: 24.04.2, Kernel: 6.11.0-29-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101011 (1.21.1.11)
| Processor: Intel(R) Core(TM) i9-14900KF
| Bare Metal Cores: 24 | Bare Metal Logical Cores: 48
| Available Cores:  32 
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 64074 | Free Memory: 57358
| Total Page/Swap (MB): 8191 | Free Page/Swap: 8191
|---------------------------------------------------------------------------------------------|
2025-07-31 14:33:46 [676ms] [Warning] [gpu.foundation.plugin] IOMMU is enabled.
2025-07-31 14:33:48 [2,414ms] [Warning] [omni.log] Source: omni.hydra was already registered.
2025-07-31 14:33:49 [2,742ms] [Warning] [omni.isaac.dynamic_control] omni.isaac.dynamic_control is deprecated as of Isaac Sim 4.5. No action is needed from end-users.
2025-07-31 14:33:50 [3,751ms] [Warning] [omni.replicator.core.scripts.extension] No material configuration file, adding configuration to material settings directly.
2025-07-31 14:33:50 [4,471ms] [Warning] [omni.kit.menu.utils.app_menu] add_menu_items: menu [<MenuItemDescription name:'New'>, <MenuItemDescription name:'Open'>, <MenuItemDescription name:'Re-open with New Edit Layer'>, <MenuItemDescription name:'Save'>, <MenuItemDescription name:'Save With Options'>, <MenuItemDescription name:'Save As...'>, <MenuItemDescription name:'Save Flattened As...'>, <MenuItemDescription name:'Add Reference'>, <MenuItemDescription name:'Add Payload'>, <MenuItemDescription name:'Exit'>] cannot change delegate
2025-07-31 14:33:51 [5,624ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/ambientOcclusion/enabled'
2025-07-31 14:33:51 [5,624ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/directLighting/sampledLighting/enabled'
2025-07-31 14:33:51 [5,624ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/indirectDiffuse/enabled'
2025-07-31 14:33:51 [5,624ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/raytracing/cached/enabled'
2025-07-31 14:33:51 [5,625ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/reflections/enabled'
2025-07-31 14:33:51 [5,625ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/sceneDb/ambientLightIntensity'
2025-07-31 14:33:51 [5,625ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/translucency/enabled'
2025-07-31 14:33:51 [5,625ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/viewTile/limit'
2025-07-31 14:33:52 [5,710ms] [Warning] [rtx.neuraylib.plugin] [IRAY:RENDER]   1.1   IRAY   rend warn : GPU 1 (NVIDIA GeForce RTX 5080) with CUDA compute capability 12.0 is unsupported by this version of iray photoreal. Please look for an update with your software provider.
2025-07-31 14:33:52 [5,713ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit buffer size 7508933632
2025-07-31 14:33:52 [5,713ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit : valid false, within: false
2025-07-31 14:33:52 [5,713ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit : decrement: 167690, decrement size: 7433845248
2025-07-31 14:33:52 [5,713ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : New limit 9574251 (slope: 447, intercept: 13179904)
2025-07-31 14:33:52 [5,713ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit buffer size 4287216384
2025-07-31 14:33:52 [5,713ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit : valid true, within: true
2025-07-31 14:33:52 [5,993ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/directLighting/sampledLighting/samplesPerPixel'
2025-07-31 14:33:52 [5,994ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/pathtracing/maxSamplesPerLaunch'
2025-07-31 14:33:52 [5,994ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults-transient/meshlights/forceDisable'
2025-07-31 14:33:52 [6,024ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/post/dlss/execMode'
imDefLkup.c,419: The application disposed a key event with 336 serial.
Traceback (most recent call last):
  File "/home/workstation/IsaacLab/scripts/tutorials/04_sensors/add_sensors_on_robot.py", line 177, in <module>
    main()
  File "/home/workstation/IsaacLab/scripts/tutorials/04_sensors/add_sensors_on_robot.py", line 161, in main
    sim = sim_utils.SimulationContext(sim_cfg)
  File "/home/workstation/IsaacLab/source/isaaclab/isaaclab/sim/simulation_context.py", line 167, in __init__
    self._apply_render_settings_from_cfg()
  File "/home/workstation/IsaacLab/source/isaaclab/isaaclab/sim/simulation_context.py", line 349, in _apply_render_settings_from_cfg
    with open(preset_filename) as file:
[... 32 lines cut ...]

### System Info

- Commit: [feature/isaacsim_5_0]
- Isaac Sim Version: [5.0.0-rc.23]
- OS: [e.g. Ubuntu 24.04]
- GPU: [e.g. RTX 5080]
- CUDA: [e.g. 12.9]
- GPU Driver: [575.64.03 ]

### Additional context

Add any other context about the problem here.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

RandomOakForest, 2025-08-01:
Thank you for posting this. The team will review it.

AntoineRichard, 2026-08-27:
Thanks for the report, @Rango96 — sorry for the delay. The `FileNotFoundError` came from `SimulationContext` resolving rendering presets under `apps/rendering_modes/<mode>.kit`. That preset-loading block was removed from `source/isaaclab/isaaclab/sim/simulation_context.py` in https://github.com/isaac-sim/IsaacLab/pull/6279, and the preset files themselves were deleted in https://github.com/isaac-sim/IsaacLab/pull/6405, so on 3.0 there's no `rendering_mode` lookup left to fail.

Closing as obsolete — please reopen with the log if the sensors tutorial still misbehaves on a current release.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard

Rango96, 2026-08-27:
这是来自QQ邮箱的自动回复邮件：
你好，邮件已收到。