isaac-sim/IsaacLab #3070: [Proposal] Enhanced Asset Selection Strategy for Multi-USD Multi-Agent Environments

Kyu3224, 2025-08-01:
### Proposal

Enhanced Asset Selection Strategy for Multi-USD Multi-Agent Environments

### Motivation

I am using multi-agent robots that require multiple robot assets, and I utilize the multi-USD spawning feature. Currently, only "random" or "sequential" (e.g., 0, 1, 2, 0, 1, 2) selection methods are supported, which limits the pool of robot combinations that can be spawned. For instance, if I want to spawn 2000 environments using two multi-USD assets, each consisting of 4 different robots, using the sequential method would result in repeated patterns like (0, 1, 2, 0, 1, 2), which leads to environment compositions such as (0, 0), (1, 1), (2, 2), and so on.

Moreover, with the random method, I cannot control the distribution ratio of the robot types being spawned. To address this, I added a new "split" option in addition to "sequential", which allows assets to be spawned in grouped segments, such as (0, 0, 1, 1, 2, 2). This way, by combining "sequential" and "split" methods, we can deterministically structure the environment setup—for example: (0, 1, 2, 0, 1, 2) + (0, 0, 1, 1, 2, 2) results in compositions like (0, 0), (1, 0), (2, 1), (0, 1), (1, 2), (2, 2), which allows for more structured and controlled spawning across environments.

### Alternatives

I will submit this as a pull request.

### Checklist

- [X] I have checked that there is no similar issue in the repo (**required**)

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

Kyu3224, 2025-08-01:
Hi @Mayankm96, thank you for your comment!

In my project, I initially implemented a simple method to switch between two selection modes. 

[image]

But after considering your suggestion, I agree that allowing a function in the config would be a more flexible and extensible approach.

For example, the current "sequential" mode could be expressed as a simple function like:

```
def choice_fn(x, num_assets):
    return x % num_assets
```

And more advanced strategies — such as controlling the distribution of assets — could also be supported via user-defined functions.

I'll explore this direction further, and if I come up with a concrete implementation idea, I'll share it here. Thanks again for the great suggestion!

AntoineRichard, 2026-08-27:
Thanks @Kyu3224 — and good news, this is basically implemented. The spawner-level flags you built on (`MultiAssetSpawnerCfg.random_choice`, `MultiUsdFileCfg.random_choice`) are deprecated and ignored, but selection moved to `InteractiveSceneCfg.clone_cfg` (`isaaclab.cloner.CloneCfg`): `clone_strategy` is a callable (`sequential` by default, `random` also provided) and `clone_combinations` lets you declare which asset sets are legal per environment — the config-level selector @Mayankm96 suggested in-thread.

That should express your grouped/split distributions directly. If it doesn't cover your case, please reopen or file a PR against `CloneCfg`.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard