isaac-sim/IsaacLab #3095: [Proposal] All RL environment should use `gymnasium.VectorEnv` rather than `gymnasium.Env`

pseudo-rnd-thoughts, 2025-08-05:
### Proposal

The RL environments (e.g., `DirectRLEnv`) should all inherit from `gymnasium.VectorEnv` rather than `gymnasium.Env`. 

### Motivation

The `DirectRLEnv` docstring say 

> This class implements the core functionality for reinforcement learning (RL)
> environments. It is designed to be used with any RL library. The class is designed
> to be used with vectorized environments, i.e., the environment is expected to be run
> in parallel with multiple sub-environments.
> 
> While the environment itself is implemented as a vectorized environment, we do not
> inherit from :class:`gym.vector.VectorEnv`. This is mainly because the class adds
> various methods (for wait and asynchronous updates) which are not required.
> Additionally, each RL library typically has its own definition for a vectorized
> environment. Thus, to reduce complexity, we directly use the :class:`gym.Env` over
> here and leave it up to library-defined wrappers to take care of wrapping this
> environment for their agents.

This description is true for Gym and Gymnasium < 1.0, however, in v1.0, the problematic functions described (wait and asynchronous updates) were removed from the base class. 

Updating to VectorEnv should make it clearer to users that the environments are vectorised and remove the necessity for libraries to implement their own converter functions. 
Further, users will be able to use vector wrappers ()with the environments, which isn't possible with `gymnasium.Env` implementations.

Looking at the code, I believe that minimal code changes will be necessary, as the library already requires Gymnasium v1.2 and we should be able to simplify the implementation to remove components already included in `VectorEnv` but not in `Env`

### Alternatives

Keep the current solution

### Additional context

Add any other context or screenshots about the feature request here.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)

StafaH, 2026-09-20:
After investigating this, the migration to VectorEnv has more limitations than benefits. It would require substantial breaking changes and there are no clear benefits. Closing this as not planned.

closed 2026-09-20 by StafaH

pseudo-rnd-thoughts, 2026-09-20:
@StafaH Could you post what the limitations and benefits are plus the current required breaking changes such that future readers can assess if the change is more feasible.

StafaH, 2026-09-20:
Hi @pseudo-rnd-thoughts m

The main benefits would be correct `VectorEnv` type identification and access to compatible Gymnasium vector wrappers. The problem is that changing the base class alone would advertise a contract that Isaac Lab does not currently meet:

- Since Gymnasium 1.0, `VectorEnv` is separate from `Env`. `gym.make()` rejects a pure `VectorEnv`, so our registrations, scripts, and downstream user code would need to move to `vector_entry_point` and `gym.make_vec()`. The earlier attempt in #3136 already showed this migration scope.
- Isaac Lab returns Torch tensors, commonly on CUDA, while several Gymnasium vector wrappers assume NumPy data. Some wrappers also reject our Same-Step autoreset mode.
- `DirectRLEnv` exposes policy and critic observations in a layout that does not match Gymnasium's `single_observation_space`/`observation_space` invariant, and our `extras` format is not Gymnasium's masked vector-info format.
- The existing SB3, RSL-RL, RL-Games, and skrl adapters would still be needed because those libraries have their own vector APIs and data conventions.
- `DirectMARLEnv` has per-agent semantics and is not a direct fit for Gymnasium's single-agent `VectorEnv` contract.

Given the compatibility cost, I don't think directly changing the environment base classes is the right migration. A better direction would be an opt-in `VectorEnv` adapter and `vector_entry_point` alongside the existing `gym.make()` API. That would preserve compatibility while providing a proper Gymnasium vector interface.

Please feel free to create a prototype and open a PR so it's easier to reason about the migration. We would need 1 deprecation cycle of the old gym.Env with opt-in for gym.VectorEnv.