isaac-sim/IsaacLab #3115: [Bug Report] No module named 'pink'

RandomOakForest, 2025-08-07:
### Discussed in https://github.com/isaac-sim/IsaacLab/discussions/3108

<div type='discussions-op-text'>

<sup>Originally posted by **LinhTranUlsan** August  7, 2025</sup>
Hi everyone,
I'm using  "--task Isaac-PickPlace-GR1T2-Abs-v0"  for training and got an error about a missing 'pink' module as follows. I've searched for that library but couldn't find it. Do you know where to download this library?
Thank you!
[image]

</div>

RandomOakForest, 2025-08-10:
See workarounds in [this comment](https://github.com/isaac-sim/IsaacLab/discussions/3108#discussioncomment-14035853).

AntoineRichard, 2026-08-27:
Thanks for reporting this, @RandomOakForest. `pin`, `pin-pink==3.3.0` and `daqp==0.8.5` are now core (non-optional) dependencies in the root `pyproject.toml`, gated to Linux x86_64/aarch64, so the uv-based install pulls the Pink IK stack in automatically. Note that gating: on other platforms the Pink IK controller — and the GR1T2 tasks that use it — isn't available.

Closing — please reopen with your install command and platform if a fresh install still leaves `pink` missing.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard