isaac-sim/IsaacLab #3117: [Bug Report] Deformable Object Tutorial Crashes on Multi-GPU with "illegal memory access" in headless mode

Rahuboy, 2025-08-07:
### Describe the bug

Running the official deformable object tutorial (`scripts/tutorials/01_assets/run_deformable_object.py`) in headless mode on a multi-GPU system crashes (illegal memory access) during initialization of the physics scene. The bug appears to be specific to multiple GPUs as it does not occur when the simulation is constrained to a single GPU.

### Steps to reproduce

1. On a multi-GPU system with a fresh installation of Isaac Lab, run the unmodified deformable object tutorial in headless mode.
2. `python run_deformable_object.py --headless` will reproduce the crash

### System Info

Describe the characteristic of your environment:

- Commit: 24a78b688e6edd1a883c8d66ffb0f189f97e458c
- Isaac Sim Version: 4.5.0
- OS: Ubuntu 22.04.5 LTS
- GPU: 8x NVIDIA L40S
- GPU Driver: 550.90.12

### Additional context

The simulation succeeds under the following conditions:
1. The environment is constrained to a single GPU (e.g., using `CUDA_VISIBLE_DEVICES=0`).
2. No problems encountered with livestreaming.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] The unmodified run_deformable_object.py script runs to completion on a multi-GPU system without crashing.

maxkra15, 2026-09-22:
Thanks @Rahuboy and @sardanian. Current 3.0 experiences pin the renderer to one GPU by default via #5933, which may remove the multi-GPU interaction seen here. That change did not modify the PhysX deformable path, however, and PhysX–Fabric GPU interop remains enabled by default. Could someone with two or more GPUs retest the current deformable tutorial on Isaac Lab 3.0 / Isaac Sim 6.1? If it still reproduces, please also try `ISAACLAB_FABRIC_USE_GPU_INTEROP=0` and share the driver/GPU details plus the first CUDA error. Keeping this open until we have a current-stack result.

hujc7, 2026-11-02:
Thanks @Rahuboy and @sardanian for the reproduction and logs.

We retested the unmodified `scripts/tutorials/01_assets/run_deformable_object.py` on a 4× L40S node with the current `main` image (Isaac Lab 3.0, Isaac Sim 6.3), all GPUs visible, both with the default settings and with `ISAACLAB_FABRIC_USE_GPU_INTEROP=0`. Every run initialized and ran ~240 reset cycles over 5 minutes without the illegal memory access. Please reopen with your GPU/driver details if you still hit it on Isaac Lab 3.0.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-11-02 by hujc7