isaac-sim/IsaacLab #3129: [Bug Report] No module named 'rclpy._rclpy_pybind11'

agrotrushant, 2025-08-08:
### Describe the bug

I am using the container workflow in IsaacLab 2.2 (main branch) and am using ros2 profile but seems like rcpy with pybind11 has some issue.

### Steps to reproduce

Clone the IsaacLab repo
```
git clone git@github.com:isaac-sim/IsaacLab.git
cd IsaacLab
```

Build ros2 profile container
```
./docker/container.py start ros2
```

Launch IsaacSim 5.0
```
runapp # alias mentioned in ~/.bashrc
```

I am getting following error:
```
[29.612s] [ext: isaacsim.ros2.bridge-4.9.3] startup
[29.681s] Attempting to load system rclpy
[29.756s] Could not import system rclpy: No module named 'rclpy._rclpy_pybind11'
The C extension '/opt/ros/humble/lib/python3.10/site-packages/_rclpy_pybind11.cpython-311-x86_64-linux-gnu.so' isn't present on the system. Please refer to 'https://docs.ros.org/en/humble/Guides/Installation-Troubleshooting.html#import-failing-without-library-present-on-the-system' for possible solutions
[29.756s] Attempting to load internal rclpy for ROS Distro: humble
[29.757s] Could not import internal rclpy: No module named 'rclpy._rclpy_pybind11'
The C extension '/opt/ros/humble/lib/python3.10/site-packages/_rclpy_pybind11.cpython-311-x86_64-linux-gnu.so' isn't present on the system. Please refer to 'https://docs.ros.org/en/humble/Guides/Installation-Troubleshooting.html#import-failing-without-library-present-on-the-system' for possible solutions
[29.757s] To use the internal libraries included with the extension please set the following environment variables to use with FastDDS (default) or CycloneDDS before starting Isaac Sim:

FastDDS (default):
export ROS_DISTRO=humble
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/humble/lib

OR

CycloneDDS:
export ROS_DISTRO=humble
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/humble/lib

2025-08-08T21:02:31Z [29,727ms] [Warning] [humble.rclpy] Could not import rclpy

```

For more information on this, check: https://www.markdownguide.org/extended-syntax/#fenced-code-blocks

-->

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: 5.0.0
- OS: Ubuntu 22.04
- GPU: RTX Pro 6000 Blackwell
- CUDA: 12.8
- GPU Driver: 570.169

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] It should use built ros-base

camopel, 2025-08-22:
you can try the ROS i build for python 3.11, it works with sim 5.0, this build only support rclpy, not ros cli. which i will release soon https://github.com/camopel/isaacsim-ros2-python3.11

chugunovyar, 2025-10-06:
So, there is no chance to install at ubuntu software from deb ? How you made release ? I can`t run software at any of versions. There always some problems which not allow to do it. 

Have any one versions list of software and OS at which with garanty it running and  works fine ?  Documentation not helps also not have any buttons to riase bug or error.

Накипело ! Извините.

Benzales, 2025-10-07:
Are there any updates with this bug? I'm experiencing the exact same issue. I'm seeing the Isaac Sim doc that specifies the need to run custom ROS 2 packages with build based on Python 3.11. However, I don't see a trivial linkage of that to the Isaac Lab deployment.
Doc: [Enabling rclpy, Custom ROS 2 Packages, and Workspaces with Python 3.11](https://docs.isaacsim.omniverse.nvidia.com/latest/installation/install_ros.html#enabling-rclpy-custom-ros-2-packages-and-workspaces-with-python-3-11)

HJGrant, 2025-12-04:
I have the same issue. I am following [this](https://isaac-sim.github.io/IsaacLab/main/source/deployment/docker.html) Docker guide in the IsaacLab docs. I was not able to build the ROS2 image before downgrading my base image to IsaacSim version 5.0.0 in the `.env.base ` file. 
Now it's built, but I am getting the same error as above.

StafaH, 2026-09-30:
We have deprecated our ros2 docker. Please use resources from Isaac ROS for ros based development

closed 2026-09-30 by StafaH