isaac-sim/IsaacLab #3135: [Question] RayCasterCamera returns incorrect hit points on Plane when rotation offset is applied

2540971687, 2025-08-09:
**Description**  
In the official tutorial `scripts/tutorials/04_sensors/run_ray_caster.py`, applying a rotation offset in `RayCasterCfg` causes RayCasterCamera to return incorrect hit points when targeting a Plane primitive.

With the default identity offset, all hit points are correct (Z ≈ 0 for a horizontal plane at Z=0).  
After adding a valid normalized quaternion in `offset.rot`, some returned hit points have Z values significantly different from 0, even though the plane is perfectly horizontal and located at Z=0.

**Steps to Reproduce**  
1. Use the unmodified `scripts/tutorials/04_sensors/run_ray_caster.py`.
2. Change the offset configuration:
```python
from omni.isaac.lab.sensors import RayCasterCfg
ray_cfg = RayCasterCfg(
    # ...
    offset=RayCasterCfg.OffsetCfg(
        pos=(0.0, 0.0, 0.0),
        rot=(0.924, 0.0, -0.383, 0.0),  # normalized quaternion
    ),
)
3.Run the script and inspect the hit points along the Z axis.

Observed Behavior
For a plane at Z=0, some hit points have Z ≠ 0 (both positive and negative deviations).
Many rays still return correct hits at Z≈0, but a subset shows noticeable Z-axis errors.
Reverting to identity rotation fixes the problem.

Expected Behavior
Applying a rotation offset should only change the ray direction pattern.
All valid hit points on a flat horizontal plane at Z=0 should have Z coordinates equal to 0 (within numerical tolerance).

Environment

Isaac Lab version: 2.1
Isaac Sim version: 4.5
System: ubuntu 22.04
GPU: 5090, Driver/CUDA: [570.153.02]
Python: 3.12

code:
def define_sensor() -> RayCaster:
    """Defines the ray-caster sensor to add to the scene."""
    # Create a ray-caster sensor
    ray_caster_cfg = RayCasterCfg(
        prim_path="/World/Origin.*/ball",
        mesh_prim_paths=["/World/ground"],
        offset = RayCasterCfg.OffsetCfg(pos=(0., 0., 0.), rot=(  0.924, 0.,  -0.383, 0.)),
        pattern_cfg=patterns.GridPatternCfg(resolution=0.1, size=(2.0, 2.0)),
        attach_yaw_only=True,
        debug_vis=True,
    )
    ray_caster = RayCaster(cfg=ray_caster_cfg)

    return ray_caster

2540971687, 2025-08-09:
[image]

RandomOakForest, 2025-08-10:
Thank you for posting this. Have you tried changing
```
attach_yaw_only=True,
```
to
```
ray_alignment="yaw"
```
[The ray caster is deprecating the former](https://isaac-sim.github.io/IsaacLab/main/source/overview/core-concepts/sensors/ray_caster.html#ray-caster).

XiaoYi-Wei, 2025-08-11:
> > 谢谢你发帖。你试过改变
> > ```
> > attach_yaw_only=True,
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 到
> > ```
> > ray_alignment="yaw"
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > [射线投射器正在贬低前者](https://isaac-sim.github.io/IsaacLab/main/source/overview/core-concepts/sensors/ray_caster.html#ray-caster)。
> 
> 感谢你的回复！
> 
> 我需要使用 RayCaster 传感器的相机模式，因此它应该跟踪底座的位置和完整方向，而不仅仅是偏航角。

> > Thank you for posting this. Have you tried changing
> > ```
> > attach_yaw_only=True,
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > to
> > ```
> > ray_alignment="yaw"
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > [The ray caster is deprecating the former](https://isaac-sim.github.io/IsaacLab/main/source/overview/core-concepts/sensors/ray_caster.html#ray-caster).
> 
> Thank you for your reply!
> 
> I need to use the RayCaster sensor's camera pattern, so it should track both the position and full orientation of the base, not just the yaw angle.

[image]
After modifying it with your help, there are still errors

[image]