isaac-sim/IsaacLab #3142: [Bug Report] Occupancy map in Isaac Lab not work

HaoLiuHust, 2025-08-11:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

I want to generate an occupancy map for every env, I have tried omap in Isaac sim, it works, but if I do it in Isaac Lab, the occupancy map not work, all element in occupancy map is same value(free zone)

### Steps to reproduce

generate omap in Isaac Lab or just open Isaac Sim GUI with Lab, and then generate omap in the GUI

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: [e.g. 5.0, this can be obtained by `cat ${ISAACSIM_PATH}/VERSION`]
- OS: [e.g. Ubuntu 22.04]
- GPU: [e.g. RTX 5090]
- CUDA: [e.g. 12.8]
- GPU Driver: [e.g. 553.05, this can be seen by using `nvidia-smi` command.]

### Additional context

Add any other context about the problem here.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

tohsin, 2025-08-12:
Yeah, I pretty much had to write my occupancy map, as I had access to the camera's pose and used ray tracing from the point cloud in world coordinates to mark free or occupied cells.

HaoLiuHust, 2025-08-12:
I have tired https://github.com/isaac-sim/IsaacLab/issues/659， also not work

RandomOakForest, 2025-08-12:
Thank you for posting this. Could you verify you have tried the following?

- Launch with headless or livestream as needed and enable offscreen rendering: set ENABLE_CAMERAS=1 (via env or AppLauncher flag) for parity with UI initialization in headless.
- After SimulationApp is created, enable the omap extension “isaacsim.asset.gen.omap”.[^3_4]
- Spawn envs, then acquire physx and stage_id and construct one Generator for the active stage.[^3_7][^3_3]
- For each env, set origin above free ground and bounds that tightly enclose that env; use cell size around 0.05 m; then call generate2d() and read the buffer.[^3_1][^3_3]
- Verify collisions exist on all obstacles; the UI documentation emphasizes collisions must be enabled for objects to appear in the map.[^3_1]

These steps reproduce the UI omap results in Isaac Lab’s standalone mode and allow generating a correct occupancy map per environment in vectorized stages.[^3_2][^3_3][^3_1]

[^3_1]: https://docs.isaacsim.omniverse.nvidia.com/latest/digital_twin/ext_isaacsim_asset_generator_occupancy_map.html

[^3_2]: https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.app.html

[^3_3]: https://docs.isaacsim.omniverse.nvidia.com/4.5.0/py/source/extensions/isaacsim.asset.gen.omap/docs/index.html

[^3_4]: https://github.com/isaac-sim/IsaacSim/issues/63

[^3_5]: https://docs.robotsfan.com/isaaclab_official/main/_modules/isaaclab/app/app_launcher.html

[^3_6]: https://isaac-sim.github.io/IsaacLab/main/source/tutorials/00_sim/launch_app.html

[^3_7]: https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.envs.html

[^3_8]: https://github.com/isaac-sim/IsaacLab/issues/3142

[^3_9]: https://docs.isaacsim.omniverse.nvidia.com/latest/overview/release_notes.html

[^3_10]: https://github.com/isaac-sim/IsaacSim/issues/73

[^3_11]: https://docs.robotsfan.com/isaacsim/4.2.0/ros_tutorials/tutorial_ros_navigation.html

[^3_12]: https://docs.isaacsim.omniverse.nvidia.com/latest/replicator_tutorials/tutorial_replicator_getting_started.html

[^3_13]: https://forums.developer.nvidia.com/t/get-camera-image-from-gym-in-headless-mode/245681

[^3_14]: https://www.youtube.com/watch?v=khFeo9DukUY

[^3_15]: https://blog.marvik.ai/2024/11/26/exploring-nvidia-omniverse-and-isaac-sim/

[^3_16]: https://git.ias.informatik.tu-darmstadt.de/cai/IsaacLab/-/blob/874b7b628d501640399a241854c83262c5794a4b/docs/source/api/lab/omni.isaac.lab.app.rst

[^3_17]: https://isaac-sim.github.io/IsaacLab/main/source/refs/troubleshooting.html

[^3_18]: https://www.youtube.com/watch?v=HHzNIh72B_Y

[^3_19]: https://isaac-sim.github.io/IsaacLab/main/source/overview/core-concepts/sensors/camera.html

[^3_20]: https://www.youtube.com/watch?v=8h0w8Sbx4rM

HaoLiuHust, 2025-08-13:
> Thank you for posting this. Could you verify you have tried the following?
> 
> * Launch with headless or livestream as needed and enable offscreen rendering: set ENABLE_CAMERAS=1 (via env or AppLauncher flag) for parity with UI initialization in headless.
> * After SimulationApp is created, enable the omap extension “isaacsim.asset.gen.omap”.[1](#user-content-fn-3_4-9a22bd43c8bb8c0a14a9c2ec884b304e)
> * Spawn envs, then acquire physx and stage_id and construct one Generator for the active stage.[2](#user-content-fn-3_7-9a22bd43c8bb8c0a14a9c2ec884b304e)[3](#user-content-fn-3_3-9a22bd43c8bb8c0a14a9c2ec884b304e)
> * For each env, set origin above free ground and bounds that tightly enclose that env; use cell size around 0.05 m; then call generate2d() and read the buffer.[4](#user-content-fn-3_1-9a22bd43c8bb8c0a14a9c2ec884b304e)[3](#user-content-fn-3_3-9a22bd43c8bb8c0a14a9c2ec884b304e)
> * Verify collisions exist on all obstacles; the UI documentation emphasizes collisions must be enabled for objects to appear in the map.[4](#user-content-fn-3_1-9a22bd43c8bb8c0a14a9c2ec884b304e)
> 
> These steps reproduce the UI omap results in Isaac Lab’s standalone mode and allow generating a correct occupancy map per environment in vectorized stages.[5](#user-content-fn-3_2-9a22bd43c8bb8c0a14a9c2ec884b304e)[3](#user-content-fn-3_3-9a22bd43c8bb8c0a14a9c2ec884b304e)[4](#user-content-fn-3_1-9a22bd43c8bb8c0a14a9c2ec884b304e)
> 
> ## Footnotes
> 1. https://github.com/isaac-sim/IsaacSim/issues/63 [↩](#user-content-fnref-3_4-9a22bd43c8bb8c0a14a9c2ec884b304e)
> 2. https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.envs.html [↩](#user-content-fnref-3_7-9a22bd43c8bb8c0a14a9c2ec884b304e)
> 3. https://docs.isaacsim.omniverse.nvidia.com/4.5.0/py/source/extensions/isaacsim.asset.gen.omap/docs/index.html [↩](#user-content-fnref-3_3-9a22bd43c8bb8c0a14a9c2ec884b304e) [↩2](#user-content-fnref-3_3-2-9a22bd43c8bb8c0a14a9c2ec884b304e) [↩3](#user-content-fnref-3_3-3-9a22bd43c8bb8c0a14a9c2ec884b304e)
> 4. https://docs.isaacsim.omniverse.nvidia.com/latest/digital_twin/ext_isaacsim_asset_generator_occupancy_map.html [↩](#user-content-fnref-3_1-9a22bd43c8bb8c0a14a9c2ec884b304e) [↩2](#user-content-fnref-3_1-2-9a22bd43c8bb8c0a14a9c2ec884b304e) [↩3](#user-content-fnref-3_1-3-9a22bd43c8bb8c0a14a9c2ec884b304e)
> 5. https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.app.html [↩](#user-content-fnref-3_2-9a22bd43c8bb8c0a14a9c2ec884b304e)

yes, I have tried the steps you have mentioned above, the strange thing is, even I am start Isaac Lab with GUI mode, I still can not get the right omap with omap extension in the code, but I can get the omap with the omap ui extension in the GUI

RandomOakForest, 2025-08-22:
Thank you for trying this. The team will review and update.

HaoLiuHust, 2025-08-29:
> Thank you for trying this. The team will review and update.

Thanks, have you reproduce the problem?

matthewtrepte, 2026-10-19:
Closing for now since we haven't had a repro or system details, and this concerns Isaac Sim's occupancy map extension rather than Isaac Lab. Please reopen with your Isaac Sim / Isaac Lab versions and a minimal repro if it's still a problem.

closed 2026-10-19 by matthewtrepte