isaac-sim/IsaacLab #3152: [Bug Report] `joint_pos_out_of_limit` indexing error when `asset_cfg.joint_ids` is set

moribots, 2025-08-12:
### Describe the bug

`isaaclab/envs/mdp/terminations.py::joint_pos_out_of_limit` reduces over joints before indexing into `asset_cfg.joint_ids`. When `asset_cfg.joint_ids` is not `None`, the code tries to index a 1-D tensor with 2 dims and raises:

```
IndexError: too many indices for tensor of dimension 1
```

Root Cause:

```
out_of_upper_limits = torch.any(asset.data.joint_pos > asset.data.soft_joint_pos_limits[..., 1], dim=1)  # [N]
out_of_lower_limits = torch.any(asset.data.joint_pos < asset.data.soft_joint_pos_limits[..., 0], dim=1)  # [N]
return torch.logical_or(out_of_upper_limits[:, asset_cfg.joint_ids], out_of_lower_limits[:, asset_cfg.joint_ids])
#                                                 ^ indexing a 1-D tensor with [:, ids]

```

### Steps to reproduce

Minimal example using Franka with 7 selected arm joints:

```
from isaaclab.envs import ManagerBasedRLEnv
from isaaclab.managers import SceneEntityCfg
from isaaclab.envs.mdp.terminations import joint_pos_out_of_limit
from isaac_learning.isaac_learning.tasks.manager_based.isaac_learning.config.franka.franka_cfg import FRANKA_PANDA_CFG

# Build env with FRANKA_PANDA_CFG (any ManagerBasedRLEnv using this robot is fine)
# Assume env.num_envs = 1, total joints J = 9
env: ManagerBasedRLEnv = ...  # construct per standard template and reset()

robot_cfg = SceneEntityCfg(name="robot", joint_ids=[0,1,2,3,4,5,6])  # 7 arm joints
_ = joint_pos_out_of_limit(env, robot_cfg)  # triggers IndexError

```

Error and Trace:

```
Traceback (most recent call last):
  File ".../manager_based_rl_env.py", line 204, in step
    self.reset_buf = self.termination_manager.compute()
  File ".../termination_manager.py", line 165, in compute
    value = term_cfg.func(self._env, **term_cfg.params)
  File ".../isaac_learning/.../mdp/terminations.py", line 147, in franka_joint_limits
    pos = joint_pos_out_of_limit(env, robot_cfg)
  File ".../isaaclab/envs/mdp/terminations.py", line 87, in joint_pos_out_of_limit
    return torch.logical_or(out_of_upper_limits[:, asset_cfg.joint_ids], out_of_lower_limits[:, asset_cfg.joint_ids])
IndexError: too many indices for tensor of dimension 1

```

Shapes during failure:

```
asset.data.joint_pos.shape              -> torch.Size([1, 9])       # [N, J]
asset.data.soft_joint_pos_limits.shape  -> torch.Size([1, 9, 2])    # [1(or N), J, 2]
asset_cfg.joint_ids                     -> [0, 1, 2, 3, 4, 5, 6]     # K=7
```

### System Info

Commit: b5d094e5de0d35d0f6f87a0709394cbe230159d9 (branch: feature/isaacsim_5_0)

Isaac Sim Version: 5.0

OS: Ubuntu 22.04

GPU: RTX 5080

CUDA: 12.8

GPU Driver: 570.144

### Additional context

This fcn is not used anywhere in the `IsaacLab` repo. The following proposed fix maintains the API and style and changes the indexing order to address the dimension reduction issue.

```
def joint_pos_out_of_limit(env: ManagerBasedRLEnv, asset_cfg: SceneEntityCfg = SceneEntityCfg("robot")) -> torch.Tensor:
    """Terminate when the asset's joint positions are outside of the soft joint limits."""
    # extract the used quantities (to enable type-hinting)
    asset: Articulation = env.scene[asset_cfg.name]
    # compute any per-joint violations (avoid reducing before indexing)
    out_of_upper_limits = asset.data.joint_pos > asset.data.soft_joint_pos_limits[..., 1]  # [N, J]
    out_of_lower_limits = asset.data.joint_pos < asset.data.soft_joint_pos_limits[..., 0]  # [N, J]

    # truncate above output to just the joints we care about
    out_of_upper_limits = out_of_upper_limits[:, asset_cfg.joint_ids]  # [N, K]
    out_of_lower_limits = out_of_lower_limits[:, asset_cfg.joint_ids]  # [N, K]

    # reduce over selected joints
    out_of_upper_limits = torch.any(out_of_upper_limits, dim=1)  # [N]
    out_of_lower_limits = torch.any(out_of_lower_limits, dim=1)  # [N]

    return torch.logical_or(out_of_upper_limits, out_of_lower_limits)  # [N]
```

This fixes the error for both `[1, J, 2]` and `[N, J, 2]` limit tensors.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

- [ ]  Function updates to index before reduction without changing its signature.

ooctipus, 2025-08-20:
somehow I am not able to reproduce this error
I added 
```
    out_of_limit = DoneTerm(
        func=mdp.joint_pos_out_of_limit, params={"asset_cfg": SceneEntityCfg("robot", joint_ids=[0, 1, 2, 3])}
    )
```
to `source/isaaclab_tasks/isaaclab_tasks/manager_based/manipulation/lift/lift_env_cfg.py`

```
asset.data.joint_pos.shape              -> torch.Size([16, 9])
asset.data.soft_joint_pos_limits.shape  -> torch.Size([16, 9, 2])
asset_cfg.joint_ids                     -> [0, 1, 2, 3]
```

```
    asset: Articulation = env.scene[asset_cfg.name]
    if asset_cfg.joint_ids is None:
        asset_cfg.joint_ids = slice(None)

    limits = asset.data.soft_joint_pos_limits[:, asset_cfg.joint_ids]
    #   asset.data.soft_joint_pos_limits[:, asset_cfg.joint_ids]  ----> torch.Size([16, 4, 2])
    out_of_upper_limits = torch.any(asset.data.joint_pos[:, asset_cfg.joint_ids] > limits[..., 1], dim=1)
    #   asset.data.joint_pos[:, asset_cfg.joint_ids].shape  ---> torch.Size([16, 4])
    #   (asset.data.joint_pos[:, asset_cfg.joint_ids] > limits[..., 1]).shape  ----> torch.Size([16, 4])
    out_of_lower_limits = torch.any(asset.data.joint_pos[:, asset_cfg.joint_ids] < limits[..., 0], dim=1)
    return torch.logical_or(out_of_upper_limits, out_of_lower_limits)
```

moribots, 2025-08-21:
Thank you for looking into this!

When I opened the ticket, I was operating on the `feature/isaacsim_5_0` branch. Looking at `main`.

There, the function looks like:

```python
def joint_pos_out_of_limit(env: ManagerBasedRLEnv, asset_cfg: SceneEntityCfg = SceneEntityCfg("robot")) -> torch.Tensor:
    """Terminate when the asset's joint positions are outside of the soft joint limits."""
    # extract the used quantities (to enable type-hinting)
    asset: Articulation = env.scene[asset_cfg.name]
    # compute any violations
    out_of_upper_limits = torch.any(asset.data.joint_pos > asset.data.soft_joint_pos_limits[..., 1], dim=1)
    out_of_lower_limits = torch.any(asset.data.joint_pos < asset.data.soft_joint_pos_limits[..., 0], dim=1)
    return torch.logical_or(out_of_upper_limits[:, asset_cfg.joint_ids], out_of_lower_limits[:, asset_cfg.joint_ids])
```

Comparing the lines

### `feature/isaacsim_5_0`

```python
out_of_upper_limits = torch.any(asset.data.joint_pos > asset.data.soft_joint_pos_limits[..., 1], dim=1)
```

### `main`

```python
out_of_upper_limits = torch.any(asset.data.joint_pos[:, asset_cfg.joint_ids] > limits[..., 1], dim=1)
```

`main` fixes the issue (and more cleanly than the solution I proposed) by indexing first, then reducing. It slices `joint_pos[:, asset_cfg.joint_ids]` and `limits[..., 1]` to shape `[N, J_sel]`, then does `any(dim=1) --> [N]`. The old code reduced to `[N]` across all joints and then tried to index `[:, joint_ids]`, which caused the error.

I should have checked again on `main` after you had altered me that `feature/isaacsim_5_0` was deprecated. Sorry about that!

I will close the ticket and the PR.

closed 2025-08-21 by moribots