isaac-sim/IsaacLab #3170: [Bug Report] Bug title

K1wi-pcy, 2025-08-14:
I am using U7 265K+5070Ti, and there will be an error related to the core issue at startup.
isaacsim 5.0，Install completely according to the official documentation, using miniconda3.

[image]

louislelay, 2025-08-14:
Hi @K1wi-pcy, you should try the Isaac Sim compability checker 

https://docs.isaacsim.omniverse.nvidia.com/latest/installation/install_workstation.html#isaac-sim-compatibility-checker

RandomOakForest, 2025-08-14:
Thank you for posting this. Let us know if you need further help.

K1wi-pcy, 2025-08-15:
> Hi [@K1wi-pcy](https://github.com/K1wi-pcy), you should try the Isaac Sim compability checker
> 
> https://docs.isaacsim.omniverse.nvidia.com/latest/installation/install_workstation.html#isaac-sim-compatibility-checker

Thanks for the suggestion, I just used the Isaac Sim compability checker. The output is as follows, it looks fine, but I still have the previous core issue, my CPU is 20 cores, it seems that only one can be recognized here

[image]

K1wi-pcy, 2025-08-15:
> Thank you for posting this. Let us know if you need further help.

I confirmed that the ubuntu system recognizes  20 cores, but this problem still occurs in isaacsim. so this issue affects my usage; is there a way to solve it?

louislelay, 2025-08-16:
Hi @K1wi-pcy, maybe you should open an issue on the Isaac Sim repository instead, as it does not seem to be related to Isaac Lab!

https://github.com/isaac-sim/IsaacSim