isaac-sim/IsaacLab #3199: Train tutorial failed!!!

SongJiang32, 2025-08-17:
Isaac Sim Version: 5.0.0

Operating System: Ubuntu 22.04

CPU Information Model: U7-265K

GPU Information Model: 
NVIDIA GeForce RTX 5070 Ti       Driver Version: 575.64.03

Topic Description
The official tutorial run failed

Detailed Description
I installed IsaacSim and  IsaacLab using the binary method. When I executed "./isaaclab.sh -p source/standalone/tutorials/00_sim/create_empty.py" it opreated Successfully. However I executed " ./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py --task=Isaac-Ant-v0 --headless" it failed.

Please help me!!! What is the problem and how to solve it.

Error Messages

2025-08-17T06:07:49Z [316,400ms] [Warning] [omni.physx.plugin] USD stage detach not called, holding a loose ptr to a stage!
2025-08-17T06:07:49Z [316,545ms] [Warning] [carb] Recursive unloadAllPlugins() detected!

Screenshots or Videos

[image]

[image]

[image]

[image]

[image]

[image]

SongJiang32, 2025-08-20:
> Hi [@SongJiang32](https://github.com/SongJiang32), do you also get those core errors on startup when running `./isaaclab.sh -p source/standalone/tutorials/00_sim/create_empty.py`? If so, it might be similar to [this issue](https://github.com/isaac-sim/IsaacLab/issues/3170).

Thanks, but it does not work not because of the new CPU series.

AntoineRichard, 2026-08-27:
Thanks for taking the time to report this, @SongJiang32. We weren't able to get to the bottom of it at the time, and with only screenshots and no text log or Isaac Lab commit there isn't enough here to reproduce it now. The invocation has also moved on: training runs through `scripts/reinforcement_learning/train.py` rather than the old `scripts/reinforcement_learning/rsl_rl/train.py` path.

Closing for now, but please reopen with the full text log if you still hit this on a current release.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard