isaac-sim/IsaacLab #3211: [Question] Is it possible to get the center of mass in the root link frame of an actor?

H-Hisamichi, 2025-08-19:
Hello everyone,  

I would like to obtain the center of mass (CoM) of the entire robot in the root link frame.  
However, according to the [documentation](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.assets.html#module-isaaclab.assets), it seems that such a function is not available.  

Is it possible to get the CoM of an actor in its root link frame?  
If not, I am planning to transform [root_com_pos_w](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.assets.html#isaaclab.assets.ArticulationData.root_com_pos_w) into the root link frame instead.  

Thanks!

RandomOakForest, 2025-08-19:
Thank you for posting this. Isaac Lab does not provide a direct function to get the center of mass (CoM) of the entire actor (robot) in the root link frame. You can obtain the CoM position of the robot in the world frame using properties like `root_com_pos_w` (or sometimes `root_com_pose_w`). The root link pose in the world frame can be accessed with `root_link_pose_w` (position and orientation).[^1][^2]

To get the CoM in the root link frame (i.e., transform from world to root link frame):

1. **Obtain both positions and orientation:**
    - `root_com_pos_w`: CoM of the robot in the world frame.
    - `root_link_pos_w`: Root link position in the world frame.
    - `root_link_quat_w`: Root link orientation as a quaternion in the world frame.[^2][^1]
2. **Transform to root link frame:**
Use the inverse of the root link pose to transform `root_com_pos_w` into the root link frame. The recommended approach is:
    - Subtract the root link position from the CoM position: $\Delta = root\_com\_pos\_w - root\_link\_pos\_w$
    - Rotate the delta vector by the inverse (conjugate) of the root link quaternion.

Python-like pseudocode using torch or numpy:

```python
import torch
from isaaclab.utils.math import quaternion_apply, quaternion_conjugate

# Assume x are torch.Tensors of shape (..., 3) or (..., 4)
com_pos_w = robot.data.root_com_pos_w
root_pos_w = robot.data.root_link_pos_w
root_quat_w = robot.data.root_link_quat_w

delta = com_pos_w - root_pos_w
root_quat_inv = quaternion_conjugate(root_quat_w)
com_in_root = quaternion_apply(root_quat_inv, delta)
```

This gives the CoM position relative to the root link frame.

3. **Summary:**
You must manually transform the CoM position using the above steps—there is no out-of-the-box property in Isaac Lab for the root link frame CoM, currently (v2.2.0).

[^1]: https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.assets.html

[^2]: https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.assets.html#articulation

[^3]: https://isaac-sim.github.io/IsaacLab/main/source/overview/core-concepts/sensors/frame_transformer.html

[^4]: https://isaac-sim.github.io/IsaacLab/ma

[^5]: https://github.com/isaac-sim/IsaacLab/issues/942

[^6]: https://forums.developer.nvidia.com/t/position-of-the-center-of-mass-of-a-robot-having-multiple-links/217651

[^7]: https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.envs.mdp.html

[^8]: https://isaac-sim.github.io/IsaacLab/main/source/api/index.html

[^9]: https://docs.robotsfan.com/isaaclab_official/main/_modules/isaaclab/assets/rigid_object/rigid_object_data.html

[^10]: https://isaac-sim.github.io/IsaacLab/main/source/refs/changelog.html

[^11]: https://docs.robotsfan.com/isaaclab_official/v2.0.0/_modules/isaaclab/assets/articulation/articulation.html

[^12]: https://isaac-sim.github.io/IsaacLab/main/source/setup/walkthrough/technical_env_design.html

H-Hisamichi, 2025-08-21:
Hello @RandomOakForest,

Thank you for the advice!
I was able to detect the CoM in the root link frame.

Note:
For those who want to implement this based on this discussion,
please replace
`quaternion_apply` -> `quat_apply`
`quaternion_conjugate` -> `quat_conjugate`

closed 2025-08-21 by H-Hisamichi