isaac-sim/IsaacLab #3239: Modules being loaded before SimulationApp when running unit tests with pytest

moribots, 2025-08-21:
can you confirm that if you hitting the same issue when running ./isaaclab.sh -p -m pytest source/isaaclab/test/assets/test_articulation.py::test_joint_pos_limits
on main branch?

_Originally posted by @ooctipus in https://github.com/isaac-sim/IsaacLab/issues/3153#issuecomment-3208517484_

on `main` `b81673267943e0faafa9de4312cc291543344c3b`

I get

```

```
[INFO] Using python from: /home/mrahme/miniconda3/envs/env_isaaclab/bin/python                                                                                                                                                                                                         
================================================================================================================================= test session starts ==================================================================================================================================
platform linux -- Python 3.11.13, pytest-8.4.0, pluggy-1.6.0
rootdir: /home/mrahme/Forks/IsaacLab/source/isaaclab
configfile: pyproject.toml
plugins: launch-testing-1.0.8, ament-flake8-0.12.12, ament-xmllint-0.12.12, ament-pep257-0.12.12, ament-lint-0.12.12, launch-testing-ros-0.19.9, ament-copyright-0.12.12, anyio-4.9.0, hydra-core-1.3.2, mock-3.14.1, flaky-3.8.1
collecting ... [INFO][AppLauncher]: Using device: cuda:0
[INFO][AppLauncher]: Loading experience file: /home/mrahme/Forks/IsaacLab/apps/isaaclab.python.headless.kit
Loading user config located at: '/home/mrahme/.cache/packman/chk/kit-kernel/107.3.1+isaac.206797.8131b85d.gl.manylinux_2_35_x86_64.release/data/Kit/Isaac-Sim/5.0/user.config.json'
[Info] [carb] Logging to file: /home/mrahme/.cache/packman/chk/kit-kernel/107.3.1+isaac.206797.8131b85d.gl.manylinux_2_35_x86_64.release/logs/Kit/Isaac-Sim/5.0/kit_20250820_200532.log
2025-08-21T00:05:33Z [871ms] [Warning] [omni.usd_config.extension] Enable omni.materialx.libs extension to use MaterialX
2025-08-21T00:05:34Z [1,681ms] [Warning] [omni.isaac.dynamic_control] omni.isaac.dynamic_control is deprecated as of Isaac Sim 4.5. No action is needed from end-users.

|---------------------------------------------------------------------------------------------|
| Driver Version: 570.144       | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 5080          | Yes: 0 |     | 16303   MB | 10de      | 0          |
|     |                                  |        |     |            | 2c02      | f80a7477.. |
|     |                                  |        |     |            | 2         |            |
|=============================================================================================|
| OS: 22.04.5 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.5, Kernel: 6.8.0-65-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101004 (1.21.1.4)
| Processor: Intel(R) Core(TM) Ultra 7 265KF
| Cores: 20 | Logical Cores: 20
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 63815 | Free Memory: 48319
| Total Page/Swap (MB): 2047 | Free Page/Swap: 2047
|---------------------------------------------------------------------------------------------|
[INFO][AppLauncher]: Using device: cuda:0
[INFO][AppLauncher]: Loading experience file: /home/mrahme/Forks/IsaacLab/apps/isaaclab.python.headless.kit
2025-08-21T00:05:40Z [8,217ms] [Warning] [simulation_app.simulation_app] Modules: ['omni.kit.main_event_loop_wrapper.main_event_loop_wrapper', 'omni.kit.main_event_loop_wrapper', 'omni.kit.async_engine.async_engine', 'omni.kit.async_engine', 'omni.client.utils.utils', 'omni.client.utils', 'omni.client.impl._omniclient', 'omni.client.impl', 'omni.client', 'omni.stats._stats', 'omni.stats', 'omni.hsscclient', 'omni.ujitso', 'omni.ujitso.default._ujitsodefault', 'omni.ujitso.default', 'omni.gpu_foundation', 'omni.gpu_foundation.shadercache', 'omni.gpu_foundation_factory._gpu_foundation_factory', 'omni.gpu_foundation_factory.impl', 'omni.gpu_foundation_factory.impl.foundation_extension', 'omni.gpu_foundation_factory', 'omni.gpu_foundation.shadercache.vulkan.shadercache_vulkan', 'omni.gpu_foundation.shadercache.vulkan', 'omni.kit.pipapi.pipapi', 'omni.kit.pipapi', 'omni.kit.pip_archive', 'omni.pip', 'omni.pip.compute', 'omni.pip.cloud', 'omni.isaac.core_archive', 'omni.core._core', 'omni.core', 'omni.kit.telemetry._telemetry', 'omni.kit.telemetry._itelemetry', 'omni.structuredlog._structuredlog', 'omni.structuredlog', 'omni.kit.telemetry.impl.sentry_extension', 'omni.kit.telemetry.impl', 'omni.kit.telemetry', 'omni.isaac.ml_archive', 'omni.usd_config.extension', 'omni.usd_config', 'omni.mdl', 'omni.mdl.pymdlsdk._pymdlsdk', 'omni.mdl.pymdlsdk', 'omni.mdl.extensions.material_graph_editor', 'omni.mdl.extensions', 'omni.mdl.pymdl', 'omni.mdl.neuraylib._neuraylib', 'omni.mdl.neuraylib.scripts.entrypoints', 'omni.mdl.neuraylib.scripts.async_helper', 'omni.mdl.neuraylib.scripts.extension_content', 'omni.mdl.neuraylib.scripts', 'omni.mdl.neuraylib.utils.utils', 'omni.mdl.neuraylib.utils', 'omni.mdl.neuraylib', 'pxr', 'pxr.Tf._tf', 'pxr.Tf', 'pxr.Trace._trace', 'pxr.Trace', 'pxr.Work._work', 'pxr.Work', 'pxr.Plug._plug', 'pxr.Plug', 'pxr.Gf._gf', 'pxr.Gf', 'pxr.Vt._vt', 'pxr.Vt', 'pxr.Ar._ar', 'pxr.Ar', 'pxr.Sdf._sdf', 'pxr.Sdf', 'pxr.Ndr._ndr', 'pxr.Ndr', 'pxr.Sdr._sdr', 'pxr.Sdr', 'pxr.Kind._kind', 'pxr.Kind', 'pxr.Pcp._pcp', 'pxr.Pcp', 'pxr.Usd._usd', 'pxr.Usd', 'pxr.UsdGeom._usdGeom', 'pxr.UsdGeom', 'pxr.UsdShade._usdShade', 'pxr.UsdShade', 'omni.UsdMdl._usdMdl', 'omni.UsdMdl', 'omni.kit.usd', 'omni.kit.usd.mdl.extension', 'omni.kit.usd.mdl', 'omni.MetricsAssembler._metricsAssembler', 'omni.MetricsAssembler', 'pxr.UsdPhysics._usdPhysics', 'pxr.UsdPhysics', 'pxr.PhysxSchema._physxSchema', 'pxr.PhysxSchema', 'pxr.PhysicsSchemaTools._physicsSchemaTools', 'pxr.PhysicsSchemaTools', 'pxr.UsdSkel._usdSkel', 'pxr.UsdSkel', 'omni.kit.actions', 'omni.kit.actions.core._kit_actions_core', 'omni.kit.actions.core.actions', 'omni.kit.actions.core', 'omni.kit.undo.history', 'omni.kit.commands.on_change', 'omni.kit.commands.command', 'omni.kit.commands.command_actions', 'omni.kit.commands._kit_commands', 'omni.kit.commands.command_bridge', 'omni.kit.commands', 'omni.kit.undo.undo', 'omni.kit.undo', 'omni.kit.commands.builtin.settings_commands', 'omni.kit.commands.builtin', 'omni.convexdecomposition.scripts', 'omni.convexdecomposition.bindings', 'omni.convexdecomposition.bindings._convexdecomposition', 'omni.convexdecomposition.scripts.extension', 'omni.convexdecomposition', 'omni.kvdb.scripts', 'omni.kvdb.bindings', 'omni.kvdb.bindings._kvdb', 'omni.kvdb.scripts.extension', 'omni.kvdb', 'omni.usd_resolver._omni_usd_resolver', 'omni.usd_resolver', 'omni.spectree', 'omni.physxfoundation.bindings', 'omni.physxfoundation.bindings._physxFoundation', 'omni.physxfoundation.scripts', 'omni.physxfoundation.scripts.extension', 'omni.physxfoundation', 'omni.localcache.scripts', 'omni.localcache.bindings', 'omni.localcache.bindings._localcache', 'omni.localcache.scripts.extension', 'omni.localcache', 'omni.usdphysics.scripts', 'omni.usdphysics.scripts.extension', 'omni.usdphysics', 'omni.resourcemonitor.scripts', 'omni.resourcemonitor._resourceMonitor', 'omni.resourcemonitor.scripts.extension', 'omni.resourcemonitor', 'omni.resourcemonitor.tests', 'omni.physxcooking.bindings', 'omni.physxcooking.bindings._physxCooking', 'omni.physxcooking.scripts', 'omni.physxcooking.scripts.extension', 'omni.physxcooking', 'omni.activity', 'omni.activity.core._activity', 'omni.activity.core', 'omni.kit.audiodeviceenum._audiodeviceenum', 'omni.kit.audiodeviceenum.extension', 'omni.kit.audiodeviceenum', 'omni.kit.loop', 'omni.kit.loop.tests', 'omni.physx.bindings', 'omni.physx.bindings._physx', 'omni.physx.scripts', 'omni.physx.scripts.extension', 'pxr.UsdUtils._usdUtils', 'pxr.UsdUtils.constantsGroup', 'pxr.UsdUtils.complianceChecker', 'pxr.UsdUtils.updateSchemaWithSdrNode', 'pxr.UsdUtils.fixBrokenPixarSchemas', 'pxr.UsdUtils.usdzUtils', 'pxr.UsdUtils', 'omni.ph [... cut ...]
[1.732s] [ext: omni.usd.metrics.assembler.ui-107.3.1] shutdown
2025-08-21T00:05:42Z [9,975ms] [Warning] [omni.ext._impl._internal] omni.usd.metrics.assembler.ui-107.3.1 -> <class 'omni.metrics.assembler.ui.extension.MetricsAssemblerUI'>: extension object is still alive, something holds a reference on it. References: ["[0]:type: <class 'frame'>, id: 138504443904640"]
[1.787s] [ext: omni.kit.usd.layers-2.2.10] shutdown
[1.818s] [ext: omni.physx.tensors-107.3.18] shutdown
[1.819s] [ext: omni.usd-1.13.10] shutdown
```
```

moribots, 2025-08-21:
Thanks for checking! I installed using the readme: https://github.com/isaac-sim/IsaacLab

but maybe I should install IsaacSim with pip now that it's been released. I'll mess around with this a bit and report back

ooctipus, 2025-08-21:
Hi @moribots 

Sorry for hearing you experience not able to run the test, I also tried binary install again, conda or non-conda, and it worked for me. And I guess if our CI also worked, it means the general installation step should work.
I will try to install in another machine and see if I can reproduce what you have. But currently I am not sure what issue is as I can't repro what you have there. Maybe do you want to try one different machine?

moribots, 2025-08-21:
Thanks for trying! When you say binary install do you mean these steps? https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/binaries_installation.html

moribots, 2025-08-21:
Interesting, so my attempt from source is equivalent to the binary attempt. In that case I'm not sure what's wrong, maybe it's my machine? I have an RTX 5080 running on ubuntu 22.04

Everything else seems to work for me except for that particular unit test. For example

```
./isaaclab.sh -p -m pytest source/isaaclab/test/performance/test_robot_load_performance.py
```

worked fine

closed 2025-08-22 by moribots