isaac-sim/IsaacLab #3250: [Bug Report] headless = True, enable_cameras = True, AppLauncher not render for cameras

camopel, 2025-08-22:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug
same code working under sim 4.5 + lab 2.1. I am migrating to sim 5.0 + lab 2.2. Pseudo code  as below:

from isaaclab.app import AppLauncher
app_launcher = AppLauncher(headless=True, enable_cameras=True)
...
# add `camera` sensor to scene
# baseEnv.reset()
# start baseEnv.step()

# problem
camera.data is always empty

### Error log:
File "/home/ubuntu/.local/lib/python3.11/site-packages/isaaclab/source/isaaclab/isaaclab/sensors/camera/camera.py", line 197, in data
    self._update_outdated_buffers()
  File "/home/ubuntu/.local/lib/python3.11/site-packages/isaaclab/source/isaaclab/isaaclab/sensors/sensor_base.py", line 353, in _update_outdated_buffers
    self._update_buffers_impl(outdated_env_ids)
  File "/home/ubuntu/.local/lib/python3.11/site-packages/isaaclab/source/isaaclab/isaaclab/sensors/camera/camera.py", line 504, in _update_buffers_impl
    self._create_annotator_data()
  File "/home/ubuntu/.local/lib/python3.11/site-packages/isaaclab/source/isaaclab/isaaclab/sensors/camera/camera.py", line 641, in _create_annotator_data
    data, info = self._process_annotator_output(name, output)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ubuntu/.local/lib/python3.11/site-packages/isaaclab/source/isaaclab/isaaclab/sensors/camera/camera.py", line 698, in _process_annotator_output
    data = data.view(height, width, 1)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: shape '[H, W, 1]' is invalid for input of size 0

My understanding: 
The camera try to _create_annotator_data for the first time when access camera.data, however rep.annotator.get_data only return empty output

I have tried to use sim.step() for 200 steps first, in case it is warm up issue, but got same failure
Also tried to call rep.orchestrator.run or async run, no help

### Cause from my understanding
under headless mode, even set enable_cameras, the camera dose not render at all

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: 5.0
- OS: Ubuntu 22.04
- GPU: L40S, AWS G6e.4x large
- CUDA: 12.8
- GPU Driver: 570.133.20

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

camopel, 2025-08-22:
tried:

inline AppLauncher(headless=True, enable_cameras=True),

set env, export ENABLE_CAMERAS=1

pass by cli like run.py --enable_cameras

all give same result

camopel, 2025-08-26:
> Thank you for posting this. Are you setting `ENABLE_CAMERAS` before running?
> 
> export ENABLE_CAMERAS=1
> [See this doc for ref.](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.app.html#environment-variables)

tried:

inline AppLauncher(headless=True, enable_cameras=True),

set env, export ENABLE_CAMERAS=1

pass by cli like run.py --enable_cameras

all give same result

tohsin, 2025-09-02:
Hi,

You can try hard coding the variable somethimes trying to pass it as an arg doesnt work

Example 
```
AppLauncher.add_app_launcher_args(parser)
args_cli = parser.parse_args()
args_cli.enable_cameras =  True
args_cli.headless = _headless
# launch omniverse app
app_launcher = AppLauncher(args_cli)
simulation_app = app_launcher.app
sys.argv.append("--enable_cameras")
```

This is what I did to fix that anyway, can't entirely remember if that last line is required though, but it's in my code.

camopel, 2025-09-02:
> Hi,
>
> You can try hard coding the variable somethimes trying to pass it as an arg doesnt work
>
> Example
>
> ```
> AppLauncher.add_app_launcher_args(parser)
> args_cli = parser.parse_args() [...]

Thanks Tohsin for your help. in my code I did same as your mentioned, directly assign args_cli.enable_cameras = True and args_cli.headless = True,  i can run in headless mode, but still not be able to access camera data. also tried to put `sys.argv.append("--enable_cameras")` also not helpful.

tohsin, 2025-09-02:
You have a file I can try running myself?