isaac-sim/IsaacLab #3252: [Question] Discrepancy in URDF `dont_collapse` / `--merge-joints` handling between Isaac Sim 4.5.0 and 5.0.0

GiulioRomualdi, 2025-08-22:
### Question

I'm currently trying to generate a USD from a URDF using **Isaac Sim 5.0.0** with the URDF-to-USD importer version **2.4.19**, and I compared the output with the one generated using **Isaac Sim 4.5.0** with importer version **2.3.10**.

In both cases, I’m using the script shown in the [Isaac Lab documentation](https://isaac-sim.github.io/IsaacLab/main/source/how-to/import_new_asset.html#using-urdf-importer).

I noticed a discrepancy in how the two versions handle the `--merge-joints` option and the URDF tag `dont_collapse="true"`.

---

#### Example URDF snippet

Let’s focus on one frame of the robot:

```xml
<link name="r_sole" />
<joint name="r_sole_fixed_joint" type="fixed" dont_collapse="true">
  <origin xyz="-9.854557236057519e-31 5.745617745498113e-20 -0.07160699999999998"
          rpy="5.158871806172491e-13 -3.6024474292099376e-16 8.99454122293897e-16" />
  <parent link="r_ankle_2" />
  <child link="r_sole" />
</joint>
```

In our case, `r_sole` is just a frame (a link without inertia). Historically, we added `dont_collapse="true"` because we did not want this frame to disappear.

---

#### Behavior in different versions

* **Isaac Sim 4.5.0 (plugin v2.3.10)**
  The `r_sole` frame appears under `r_ankle_2` as a child without mass (no mass property is assigned).

  [image]

* **Isaac Sim 5.0.0 (plugin v2.4.19)**
  The same `r_sole` frame is imported as a link with an automatically computed mass.

  [image]

---

### Questions

1. Is this difference in behavior between the two versions intended, or is it unexpected?
2. How should `--merge-joints` and `dont_collapse="true"` behave when the joint connects to a link with zero mass (like `r_sole` in our case)?
3. When mass and inertia are auto-computed, how does this affect the simulation? Could it influence the accuracy of sensors?
   For example, the [IMU sensor](https://isaac-sim.github.io/IsaacLab/main/source/overview/core-concepts/sensors/imu.html) states:

   > These sensors are traditionally designed to report linear accelerations and angular velocities, and function on similar principles to that of a digital scale: They report accelerations derived from the net force acting on the sensor

   If the mass is zero, are the sensor values reliable?
4. If a link has zero mass (like `r_sole`), does the simulator keep it at zero, or does it add a small delta mass to stabilize the simulation? If so, where can we find this value?

RandomOakForest, 2025-08-22:
Thank you for posting this. 

#### 1. Isaac Sim Versions Discrepancy

You are correct on the differences you notice. In Isaac Sim 4.5, the `--merge-joints` flag got removed, unfortunately, though it was there in the previous 4.2 version. It is now back in the 5.0 version.

#### 2. Expected Behavior: `--merge-joints` and `dont_collapse="true"`

- The `--merge-joints` flag is meant to reduce asset complexity by merging fixed joints during import. However, if a **fixed joint** in your URDF has the attribute `dont_collapse="true"`, the importer is instructed **not to merge that joint**, preserving the link for downstream tasks (e.g., frames needed for sensors or kinematic computations).[^1]
- For a link like `r_sole` that has zero mass, if `dont_collapse="true"` is specified, Isaac Sim 5.0.0’s URDF importer should keep the link explicitly, and not merge it, even when `--merge-joints` is set. This ensures such links remain in the scene graph for visualization, frame referencing, or sensor mounting.[^1]

#### 3. Auto-Computed Mass and Inertia: Effects on Simulation and Sensors

- If a link (e.g., `r_sole`) is preserved and its mass is zero—or not specified—the importer may **auto-compute** mass and inertia based on default density or other fallback values. Some links may end up with a small nonzero mass to prevent simulation instability.[^2][^3]
- IMU sensors in Isaac Sim compute values based on the physics simulation of the link to which they are attached. The IMU uses the link’s net force (including gravity and other forces) and the mass property to infer accelerations and angular velocities. If the link mass is zero:
    - **Physically**, accelerations could be undefined or unreliable (division by zero risks); simulators typically avoid zero mass for physics actors for stability reasons.
    - **Simulation**: Isaac Sim may substitute a small default mass to stabilize simulation, especially for links that need to interact physically (e.g., when sensors are attached).[^3][^2]
    - IMU sensor values may be **unreliable** or meaningless if mass is zero, as the computation assumes a nonzero mass.[^3]

#### 4. Small Delta Mass: Does Simulator Add It, and Where Is the Value?

- If no mass is assigned, Isaac Sim’s importer will use a default density or fallback value (see “Default Density” in the URDF importer docs), ensuring that every rigid body has a positive mass for simulation.[^2][^3]
- **Where to find the value:** In the GUI, you can inspect the auto-computed mass for each rigid body using the Physics Inspector Utility (“Simulation > Physics Authoring > Physics Inspector Toolbar”). This utility shows the total mass and inertia used for each link.[^4][^5]

[^1]: https://isaac-sim.github.io/IsaacLab/main/source/how-to/import_new_asset.html

[^2]: https://docs.isaacsim.omniverse.nvidia.com/latest/importer_exporter/ext_isaacsim_asset_importer_urdf.html

[^3]: https://docs.isaacsim.omniverse.nvidia.com/latest/importer_exporter/ext_isaacsim_asset_importer_mjcf.html

[^4]: https://www.youtube.com/watch?v=MbjkapbKy_w

[^5]: https://forums.developer.nvidia.com/t/autocomputed-mass-inertias/199875

[^6]: https://github.com/isaac-sim/IsaacLab/issues/3252

[^7]: https://forums.developer.nvidia.com/t/issue-with-urdf-file-import-to-isaac-sim-5-0-0/339784

[^8]: https://github.com/isaac-sim/IsaacLab/issues/1760

[^9]: https://github.com/isaac-sim/IsaacLab/issues/1800

[^10]: https://forums.developer.nvidia.com/t/computation-of-the-forces-between-rigid-and-soft-bodies-with-joint-data/320668

[^11]: https://docs.robotsfan.com/isaaclab_official/main/_modules/isaaclab/sim/converters/urdf_converter.html

GiulioRomualdi, 2025-08-25:
Hi @RandomOakForest thank you for your answer! Now it is much clearer!