isaac-sim/IsaacLab #3275: [Bug Report] Hyperparameter tuning LogExtractionErrors

SkoveBC, 2025-08-26:
### Problem

I tried testing hyperparameter tuning on my local machine, but unfortunately, I’m getting LogExtractionErrors even though I followed the steps of the Docker-based Local Quickstart section as described. What might be reasons for these errors and how to fix them?

#### Starting the Ray server within the tuning docker image

```
user@machine:~/IsaacLab$ docker run -v $(pwd)/source:/workspace/isaaclab/source -it --gpus all --net=host --entrypoint /bin/bash isaacray
root@machine:/workspace/isaaclab# echo "import ray; ray.init(); import time; [time.sleep(10) for _ in iter(int, 1)]" | ./isaaclab.sh -p
[INFO] Using python from: /workspace/isaaclab/_isaac_sim/python.sh
2025-08-25 20:01:07,664 WARNING services.py:2009 -- WARNING: The object store is using /tmp instead of /dev/shm because /dev/shm has only 67108864 bytes available. This will harm performance! You may be able to free up space by deleting files in /dev/shm. If you are inside a Docker container, you can increase /dev/shm size by passing '--shm-size=10.24gb' to 'docker run' (or add it to the run_options list in a Ray cluster config). Make sure to set this to more than 30% of available RAM.
2025-08-25 20:01:08,844 INFO worker.py:1762 -- Started a local Ray instance. View the dashboard at http://127.0.0.1:8265

```

#### Starting a tuning run in a new terminal inside the same container
```
root@machine:/workspace/isaaclab# nvidia-smi
Mon Aug 25 20:02:29 2025
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 575.64.03              Driver Version: 575.64.03      CUDA Version: 12.9     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  Quadro RTX 6000                Off |   00000000:17:00.0 Off |                  Off |
| 33%   34C    P8             22W /  260W |      11MiB /  24576MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
|   1  Quadro RTX 6000                Off |   00000000:B3:00.0 Off |                  Off |
| 33%   36C    P8             33W /  260W |     102MiB /  24576MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
﻿
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI              PID   Type   Process name                        GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
+-----------------------------------------------------------------------------------------+

root@machine:/workspace/isaaclab# ./isaaclab.sh -p scripts/reinforcement_learning/ray/tuner.py \
  --cfg_file scripts/reinforcement_learning/ray/hyperparameter_tuning/vision_cartpole_cfg.py \
  --cfg_class CartpoleTheiaJobCfg \
  --run_mode local \
  --workflow scripts/reinforcement_learning/rl_games/train.py \
  --num_workers_per_node 2

[INFO] Using python from: /workspace/isaaclab/_isaac_sim/python.sh
[INFO]: The max number of lines to search for experiment logs before (early) terminating the training workflow process is set to 1000.
[INFO]: The process response timeout, used while updating tensorboard scalars and searching for experiment logs, is set to 200.0 seconds.
[INFO]: Max number of LogExtractionError failures before we abort the whole tuning run is set to 2.
﻿
[INFO]: Using 2 workers per node.
[INFO]: Using local mode PYTHON_EXEC='/workspace/isaaclab/isaaclab.sh -p' WORKFLOW='scripts/reinforcement_learning/rl_games/train.py'
[INFO]: Attempting to use sweep config from file_path='scripts/reinforcement_learning/ray/hyperparameter_tuning/vision_cartpole_cfg.py' class_name='CartpoleTheiaJobCfg'
[INFO]: Successfully imported vision_cartpole_cfg from scripts/reinforcement_learning/ray/hyperparameter_tuning/vision_cartpole_cfg.py
[INFO]: Found correct class <class 'vision_cartpole_cfg.CartpoleTheiaJobCfg'>
[INFO]: Successfully instantiated class 'CartpoleTheiaJobCfg' from scripts/reinforcement_learning/ray/hyperparameter_tuning/vision_cartpole_cfg.py
[INFO]: Grabbed the following hyperparameter sweep config:
 {'runner_args': {'--task': <ray.tune.search.sample.Categorical object at 0x7f5b140d5990>, 'headless_singleton': '--headless', 'enable_cameras_singleton': '--enable_cameras', '--num_envs': <ray.tune.search.sample.Categorical object at 0x7f5b140d5d90>}, 'hydra_args': {'env.observations.policy.image.params.model_name': <ray.tune.search.sample.Categorical object at 0x7f5b1670c390>, 'agent.params.config.max_epochs': 200, 'agent.params.config.horizon_length': <ray.tune.search.sample.Categorical object at 0x7f5b140d7290>, 'agent.params.config.minibatch_size': <ray.tune.search.sample.Function object at 0x7f5b140d60d0>, 'agent.params.network.mlp.units': <ray.tune.search.sample.Function object at 0x7f5b140d7390>, 'agent.params.network.mlp.initializer.name': 'default', 'agent.params.network.mlp.activation': 'elu'}}
[WARNING]: Not saving checkpoints, just running experiment...
[INFO]: Model parameters and metrics will be preserved.
[WARNING]: For homogeneous cluster resources only...
2025-08-25 20:02:40,309 INFO worker.py:1586 -- Connecting to existing Ray cluster at address: 10.60.141.29:57901...
2025-08-25 20:02:40,320 INFO worker.py:1762 -- Connected to Ray cluster. View the dashboard at http://127.0.0.1:8265
[INFO]: Available resources [{'CPU': 24.0, 'GPU': 2.0, 'memory': 50069467751.0}]
[INFO]: Using config {'runner_args': {'--task': <ray.tune.search.sample.Categorical object at 0x7f5b140d5990>, 'headless_singleton': '--headless', 'enable_cameras_singleton': '--enable_cameras', '--num_envs': <ray.tune.search.sample.Categorical object at 0x7f5b140d5d90>}, 'hydra_args': {'env.observations.policy.image.params.model_name': <ray.tune.search.sample.Categorical object at 0x7f5b1670c390>, 'agent.params.config.max_epochs': 200, 'agent.params.config.horizon_length': <ray.tune.search.sample.Categorical object at 0x7f5b140d7290>, 'agent.params.config.minibatch_size': <ray.tune.search.sample.Function object at 0x7f5b140d60d0>, 'agent.params.network.mlp.units': <ray.tune.search.sample.Function object at 0x7f5b140d7390>, 'agent.params.network.mlp.initializer.name': 'default', 'agent.params.network.mlp.activation': 'elu'}}
[I 2025-08-25 20:02:40,364] A new study created in memory with name: optuna
[... 327 lines cut ...]
(IsaacLabTuneTrainable pid=1935)
2025-08-25 20:09:33,048 WARNING experiment_analysis.py:180 -- Failed to fetch metrics for 1 trial(s):
- IsaacLabTuneTrainable_1214cfca: FileNotFoundError('Could not fetch metrics for IsaacLabTuneTrainable_1214cfca: both result.json and progress.csv were not found at /tmp/ray/IsaacRay-CartpoleTheiaJobCfg-tune/IsaacLabTuneTrainable_1214cfca_5_trial_index=1,agent_params_config_horizon_length=64,agent_params_config_max_epochs=200,agent_para_2025-08-25_20-06-12')
[DONE!]: Check results with tensorboard dashboard
(IsaacLabTuneTrainable pid=1935) [ERROR]: Could not find experiment logs within 200.0 seconds.
(IsaacLabTuneTrainable pid=1935) [ERROR]: Could not extract experiment_name/logdir from trainer output (experiment_name=None, logdir=None).
(IsaacLabTuneTrainable pid=1935)    Make sure your training script prints the following correctly:
(IsaacLabTuneTrainable pid=1935)        Exact experiment name requested from command line: <name>
(IsaacLabTuneTrainable pid=1935)        [INFO] Logging experiment in directory: <logdir>

```

### System Info

- Commit: 3a1a65bd94
- Isaac Sim Version: 5.0.0-rc.45+release.23960.184afb15.gl
- OS: Ubuntu 22.04.5 LTS
- GPU: Quadro RTX 6000
- CUDA: 12.9
- GPU Driver: 575.64.03

### Checklist

- [x] I have checked that there is no similar issue in the repo
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

SkoveBC, 2025-08-28:
Thank you for helping out @ozhanozen. A short update of the current state:

Unfortunately, setting the limit higher does not solve the problem. After adding the `log_all_output = True` line, I do get this output:

```
root@machine:/workspace/isaaclab# ./isaaclab.sh -p scripts/reinforcement_learning/ray/tuner.py \
  --cfg_file scripts/reinforcement_learning/ray/hyperparameter_tuning/vision_cartpole_cfg.py \
  --cfg_class CartpoleTheiaJobCfg \
  --run_mode local \
  --workflow scripts/reinforcement_learning/rl_games/train.py \
  --num_workers_per_node 2 --process_response_timeout 500

[INFO] Using python from: /workspace/isaaclab/_isaac_sim/python.sh
2025-08-26 14:45:47,628 INFO worker.py:1586 -- Connecting to existing Ray cluster at address: 10.60.141.29:46442...
2025-08-26 14:45:47,638 INFO worker.py:1762 -- Connected to Ray cluster. View the dashboard at http://127.0.0.1:8265
[I 2025-08-26 14:45:47,704] A new study created in memory with name: optuna
2025-08-26 14:45:52,331 WARNING repeater.py:19 -- `num_samples` is now expected to be the total number of trials, including the repeat trials. For example, set num_samples=15 if you intend to obtain 3 search algorithm suggestions and repeat each suggestion 5 times. Any leftover trials (num_samples mod repeat) will be ignored.
[INFO]: The max number of lines to search for experiment logs before (early) terminating the training workflow process is set to 1000.
[INFO]: The process response timeout, used while updating tensorboard scalars and searching for experiment logs, is set to 500.0 seconds.
[INFO]: Max number of LogExtractionError failures before we abort the whole tuning run is set to 2.

[INFO]: Using 2 workers per node.
[INFO]: Using local mode PYTHON_EXEC='/workspace/isaaclab/isaaclab.sh -p' WORKFLOW='scripts/reinforcement_learning/rl_games/train.py'
[INFO]: Attempting to use sweep config from file_path='scripts/reinforcement_learning/ray/hyperparameter_tuning/vision_cartpole_cfg.py' class_name='CartpoleTheiaJobCfg'
[INFO]: Successfully imported vision_cartpole_cfg from scripts/reinforcement_learning/ray/hyperparameter_tuning/vision_cartpole_cfg.py
[INFO]: Found correct class <class 'vision_cartpole_cfg.CartpoleTheiaJobCfg'>
[INFO]: Successfully instantiated class 'CartpoleTheiaJobCfg' from scripts/reinforcement_learning/ray/hyperparameter_tuning/vision_cartpole_cfg.py
[INFO]: Grabbed the following hyperparameter sweep config:
 {'runner_args': {'--task': <ray.tune.search.sample.Categorical object at 0x7f1d820e20d0>, 'headless_singleton': '--headless', 'enable_cameras_singleton': '--enable_cameras', '--num_envs': <ray.tune.search.sample.Categorical object at 0x7f1d820e2850>}, 'hydra_args': {'env.observations.policy.image.params.model_name': <ray.tune.search.sample.Categorical object at 0x7f1d820e23d0>, 'agent.params.config.max_epochs': 200, 'agent.params.config.horizon_length': <ray.tune.search.sample.Categorical object at 0x7f1d820e39d0>, 'agent.params.config.minibatch_size': <ray.tune.search.sample.Function object at 0x7f1d820e2510>, 'agent.params.network.mlp.units': <ray.tune.search.sample.Function object at 0x7f1d820e3ad0>, 'agent.params.network.mlp.initializer.name': 'default', 'agent.params.network.mlp.activation': 'relu'}}
[WARNING]: Not saving checkpoints, just running experiment...
[INFO]: Model parameters and metrics will be preserved.
[WARNING]: For homogeneous cluster resources only...
[INFO]: Available resources [{'CPU': 24.0, 'GPU': 2.0, 'memory': 49697827021.0}]
[INFO]: Using config {'runner_args': {'--task': <ray.tune.search.sample.Categorical object at 0x7f1d820e20d0>, 'headless_singleton': '--headless', 'enable_cameras_singleton': '--enable_cameras', '--num_envs': <ray.tune.search.sample.Categorical object at 0x7f1d820e2850>}, 'hydra_args': {'env.observations.policy.image.params.model_name': <ray.tune.search.sample.Categorical object at 0x7f1d820e23d0>, 'agent.params.config.max_epochs': 200, 'agent.params.config.horizon_length': <ray.tune.search.sample.Categorical object at 0x7f1d820e39d0>, 'agent.params.config.minibatch_size': <ray.tune.search.sample.Function object at 0x7f1d820e2510>, 'agent.params.network.mlp.units': <ray.tune.search.sample.Function object at 0x7f1d820e3ad0>, 'agent.params.network.mlp.initializer.name': 'default', 'agent.params.network.mlp.activation': 'relu'}}
╭──────────────────────────────────────────────────────────────────────╮
│ Configuration for experiment     IsaacRay-CartpoleTheiaJobCfg-tune   │
├──────────────────────────────────────────────────────────────────────┤
│ Search algorithm                 SearchGenerator                     │
│ Scheduler                        FIFOScheduler                       │
│ Number of trials                 100                                 │
╰──────────────────────────────────────────────────────────────────────╯

View detailed results here: /tmp/ray/IsaacRay-CartpoleTheiaJobCfg-tune
To visualize your results with TensorBoard, run: `tensorboard --logdir /tmp/ray/session_2025-08-26_14-45-33_856089_3519/artifacts/2025-08-26_14-45-47/IsaacRay-CartpoleTheiaJobCfg-tune/driver_artifacts`
[WARNING]: Splitting node into more than one worker

Trial status: 1 PENDING
Current time: 2025-08-26 14:45:52. Total running time: 0s
Logical resource usage: 0/24 CPUs, 0/2 GPUs (0.0/1.0 accelerator_type:RTX)
╭─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────╮
[... cut ...] rparameter tuning do work since the above logs do show CUDA and GPU related warnings. The training run like `./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py --task Isaac-Reach-Franka-v0 --headless
` does work without any problems.

Has anyone else encountered similar errors?

My next step will be to contact my system administrator to check whether reinstalling the NVIDIA drivers and CUDA is possible and if that might resolve the issue. I’ll share any updates once I make progress.

ozhanozen, 2025-08-28:
Hi @SkoveBC, I think the issue is not related to Ray but rather that you are unable to run that particular training with those settings.

Can you try to do a standalone training run without Ray, but with the same settings Ray tries, e.g. run this:

`/workspace/isaaclab/isaaclab.sh -p scripts/reinforcement_learning/rl_games/train.py --task Isaac-Cartpole-RGB-TheiaTiny-v0 --headless --enable_cameras --num_envs 2048 env.observations.policy.image.params.model_name=theia-small-patch16-224-cddsv agent.params.config.max_epochs=200 agent.params.config.horizon_length=64 agent.params.config.minibatch_size=128 'agent.params.network.mlp.units=[21,31]' agent.params.network.mlp.initializer.name=default agent.params.network.mlp.activation=relu`

This has `--enable_cameras` in contrast to Franka task you have run, and I am wondering if the issue is related to rendering.

RandomOakForest, 2025-09-02:
Thank you for posting this and following up. We'll aim to repro this.

closed 2025-09-22 by SkoveBC