isaac-sim/IsaacLab #3281: [Bug Report] ModuleNotFoundError: No module named 'pinocchio'

lukehugh, 2025-08-27:
### Describe the bug

The latest commit (b3f6b31674b1e3d7958d9c1161773cc863c88dce) on the main branch caused an error on Windows.

```
2025-08-27T06:52:40Z [6,197ms] [Error] [omni.ext._impl.custom_importer] Failed to import python module isaaclab_tasks. Error: No module named 'pinocchio'. Traceback:
Traceback (most recent call last):
  File "D:\IsaacLab\_isaac_sim/kit/kernel/py\omni\ext\_impl\custom_importer.py", line 85, in import_module
    return importlib.import_module(name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\heyu\miniconda3\envs\env_isaaclab\Lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "d:/isaaclab/source/isaaclab_tasks/isaaclab_tasks/__init__.py", line 25, in <module>
    from .utils import import_packages
  File "d:/isaaclab/source/isaaclab_tasks/isaaclab_tasks/utils/__init__.py", line 9, in <module>
    from .parse_cfg import get_checkpoint_path, load_cfg_from_registry, parse_env_cfg
  File "d:/isaaclab/source/isaaclab_tasks/isaaclab_tasks/utils/parse_cfg.py", line 16, in <module>
    from isaaclab.envs import DirectRLEnvCfg, ManagerBasedRLEnvCfg
  File "d:/isaaclab/source/isaaclab/isaaclab/envs/__init__.py", line 45, in <module>
    from . import mdp, ui
  File "d:/isaaclab/source/isaaclab/isaaclab/envs/mdp/__init__.py", line 18, in <module>
    from .actions import *  # noqa: F401, F403
    ^^^^^^^^^^^^^^^^^^^^^^
  File "d:/isaaclab/source/isaaclab/isaaclab/envs/mdp/actions/__init__.py", line 8, in <module>
    from .actions_cfg import *
  File "d:/isaaclab/source/isaaclab/isaaclab/envs/mdp/actions/actions_cfg.py", line 8, in <module>
    from isaaclab.controllers import DifferentialIKControllerCfg, OperationalSpaceControllerCfg
  File "d:/isaaclab/source/isaaclab/isaaclab/controllers/__init__.py", line 18, in <module>
    from .pink_ik import NullSpacePostureTask, PinkIKController, PinkIKControllerCfg
  File "d:/isaaclab/source/isaaclab/isaaclab/controllers/pink_ik/__init__.py", line 11, in <module>
    from .null_space_posture_task import NullSpacePostureTask
  File "d:/isaaclab/source/isaaclab/isaaclab/controllers/pink_ik/null_space_posture_task.py", line 8, in <module>
    import pinocchio as pin
ModuleNotFoundError: No module named 'pinocchio'

2025-08-27T06:52:40Z [6,197ms] [Error] [carb.scripting-python.plugin] Exception: Extension python module: 'isaaclab_tasks' in 'd:\isaaclab\source\isaaclab_tasks' failed to load.

At:
  D:\IsaacLab\_isaac_sim/kit/kernel/py\omni\ext\_impl\_internal.py(222): startup
  D:\IsaacLab\_isaac_sim/kit/kernel/py\omni\ext\_impl\_internal.py(337): startup_extension
  PythonExtension.cpp::startup()(2): <module>
  D:\IsaacLab\\_isaac_sim\exts\isaacsim.simulation_app\isaacsim\simulation_app\simulation_app.py(466): _start_app
  D:\IsaacLab\\_isaac_sim\exts\isaacsim.simulation_app\isaacsim\simulation_app\simulation_app.py(230): __init__
  D:\IsaacLab\source\isaaclab\isaaclab\app\app_launcher.py(811): _create_app
  D:\IsaacLab\source\isaaclab\isaaclab\app\app_launcher.py(127): __init__
  D:\IsaacLab\scripts\tutorials\00_sim\create_empty.py(29): <module>

2025-08-27T06:52:40Z [6,197ms] [Error] [omni.ext.plugin] [ext: isaaclab_tasks-0.10.47] Failed to startup python extension.
```

### Steps to reproduce

After a fresh installation, run:

```
python scripts\tutorials\00_sim\create_empty.py
```

### System Info

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: 5.0
- OS: Windows 11 24H2
- GPU: RTX 4090
- CUDA: 12.9
- GPU Driver: 576.88

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

---

Here is the complete log file: [log.txt](https://github.com/user-attachments/files/22002508/log.txt)

ZainZh, 2025-08-27:
same here, feel confused

ZainZh, 2025-08-27:
Hi I just fixed it 1 minute ago, you just need to reinstall the `pinocchio` by
```
conda install pinocchio -c conda-forge

```
Good Luck

lukehugh, 2025-08-27:
According to #2808, pinocchio should be installed alongside pin-pink, but this library does not provide Windows wheels.

https://github.com/isaac-sim/IsaacLab/blob/b3f6b31674b1e3d7958d9c1161773cc863c88dce/source/isaaclab/setup.py#L50-L55

To avoid manual installation, perhaps pinocchio from conda-forge should be included in [environment.yml](https://github.com/isaac-sim/IsaacLab/blob/main/environment.yml) or installed via [isaaclab.bat](https://github.com/isaac-sim/IsaacLab/blob/main/isaaclab.bat) for Windows?

louislelay, 2025-08-27:
Just want to add that this commit has also broke Isaac Lab for me (ubuntu 22.04/ RTX 3090), the fixes mentionned didnt work so i just reverted back to the commit before and everything works well.