isaac-sim/IsaacLab #3305: [Bug Report] VSCode Debugger does not work in IsaacLab 2.2.0

vassil-atn, 2025-08-29:
### Describe the bug

When I try to run a script using the VSCode Python Debugger with IsaacSim 5.0.0 (and IsaacLab 2.2.0) I get an ```OSError: libstdc++.so.6: version `GLIBCXX_3.4.30' not found```. If I run the same script directly from the terminal, it works fine.  This same script also worked fine before updating to IsaacSim 5.0.0.

### System Info

- Commit: a80fa
- Isaac Sim Version: 5.0.0
- OS: Ubuntu 24.04
- GPU: RTX 4060
- CUDA: 12.9
- GPU Driver: 575.64.03

### Additional context

Add any other context about the problem here.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] A workaround to getting the Debugger to work.
- [ ] A permanent solution.

hz3014, 2025-09-02:
I am faceing same bug when installing IsaacSim 5.0.0, do we have an solution yet?

vassil-atn, 2025-09-02:
> you can try: conda install -c conda-forge gcc=12 -y

Thanks a lot, this solved it for me! Perhaps it can be added to the install script?

tohsin, 2025-09-02:
uwc

closed 2025-09-03 by RandomOakForest