isaac-sim/IsaacLab #3320: Using Ray Caster Camera in muti env - Invalid Prim Paths

tohsin, 2025-09-01:
### Ray Caster Camera doesn't support multi-env Setups for Prims

I use the Ray Caster Camera to detect faces of Meshes for inspection tasks. Unfortunately, trying to run training on more than a single environment breaks my code because the Ray Caster Camera won't take in wildcard prims and listing the prim paths of the meshes breaks the code, as only one prim path is allowed.

Note that when I use terrain and a single environment, and use the format below, it works fine; however, training is incredibly slow with a single environment, as you can imagine.
```
"/World/ground/terrain/forklift"
```

### Steps to reproduce

I attached some code below

```
@configclass
class WarehouseSceneCfg(InteractiveSceneCfg):
    # scene
    warehouse: AssetBaseCfg = AssetBaseCfg(
        prim_path= env_parameters["env_prim_path"],
        spawn=sim_utils.UsdFileCfg(
            usd_path=env_parameters["env_file_path"],
            # scale=(1.0, 1.0, 1.0),
        ),
        collision_group=-1, # Keep collision settings
        debug_vis=True
    )
 raycaster_camera_cfg = RayCasterCameraCfg(
        prim_path="/World/envs/env_.*/Robot/base_link",
        update_period=0.1,
        data_types=["face_ids"],
        offset=RayCasterCameraCfg.OffsetCfg(
            pos=(0.3, 0.0, 0.15),
            rot = (0,  0, -0.7071068, 0.7071068),
            convention="ros"
        ),
        pattern_cfg= patterns.PinholeCameraPatternCfg(
            height=_height,
            width=_width,
            focal_length=24.0,
            horizontal_aperture=20.955,
        ),
 
        mesh_prim_paths = "/World/envs/env_.*/Warehouse/forklift"]
    )
```
Causes the error
```
  File "/home/tosin/IsaacLab_inspection/source/isaaclab/isaaclab/sim/utils.py", line 610, in get_first_matching_child_prim
    raise ValueError(f"Prim at path '{prim_path}' is not valid.")
ValueError: Prim at path '/World/envs/env_.*/Warehouse/forklift' is not valid.
```
I can also confirm that the path actually exists
[image]

However, I can confirm that passing `"/World/envs/env_0/Warehouse/forklift"` Does work fine. But passing more than one Prim path also breaks.
```
  File "/home/tosin/IsaacLab_inspection/source/isaaclab/isaaclab/sensors/ray_caster/ray_caster.py", line 165, in _initialize_warp_meshes
    raise NotImplementedError(
NotImplementedError: RayCaster currently only supports one mesh prim. Received: 2
```

It's important that I have this running on multiple environments to allow me to experiment quickly. Is there any fix for this?

Thanks in advance!

### System Info

Describe the characteristic of your environment:

- Isaac Sim Version: 5.0
- OS: Ubuntu 22.04
- GPU: RTX 5090]
- CUDA: 12.8
- GPU Driver: 570.86.10 

### Additional context

Add any other context about the problem here.

### Checklist

- [X] I have checked that there is no similar issue in the repo (**required**)
- [X] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

RandomOakForest, 2025-09-02:
Thanks for posting this. The team is aware of this limitation and we are currently working to resolve it. See comments on #3154 for reference.

tohsin, 2025-09-02:
Hey @RandomOakForest,

Part of the solution you suggested was

`Could you verify you have added all occluding meshes (e.g., the wall) to mesh_prim_paths, and ensure they are static USD triangle meshes`

But the Ray Caster Camera doesn’t take multiple meshes in the place hence why I submitted a different issues. My bad

Thanks looks forward to potential fixes. Any suggestions on what needs to be done maybe I can submit a PR.

closed 2025-09-26 by RandomOakForest