isaac-sim/IsaacLab #3324: [Bug Report] `Cloning joints FixedJoint without a body rel may cause issues, since the localPose wont be updated.` when using `replicate_physics=True`

MinchoU, 2025-09-02:
### Describe the bug

I can't use `replicate_physics=True` with my usd file with fixed joints.

### Steps to reproduce

I'm trying to input my scene which comprises of rigid body objects with fixed joint to the default prim along with the floating rigid body objects.

My usd file looks like this:

- world (defaultPrim) : Xform
    - Geometry : Scope
        -  truck : Xform
            - truck_base : Cube
            - FixedJoint : PhysicsFixedJoint
        - other assets ...

and FixedJoint connects 'truck' to 'world'.  But when I include this usd as     
```python3
class TruckSceneCfg(InteractiveSceneCfg):
    truck = AssetBaseCfg(
            prim_path="/World/envs/env_.*/truck_scene",
            init_state=AssetBaseCfg.InitialStateCfg(pos=[0., 0, 0], rot=[1.0, 0, 0, 0]),
            spawn=sim_utils.UsdFileCfg(usd_path=truck_scene.usd",
                                       activate_contact_sensors=False,)
        )

self.cfg.scene = TruckSceneCfg(num_envs=4096, env_spacing=20.0, replicate_physics=True)
```
and spawned via DirectRL environment, I got the following warning
```bash
[Warning] [omni.physx.plugin] Cloning joints /World/envs/env_3/truck_scene/Geometry/truck/FixedJoint without a body rel may cause issues, since the localPose wont be updated.
```

and the links except the env_0 were flew away from the scene.

If I deactivate the `replicate_physics`, it works but simulation start takes more time. 

This is the [usd](https://github.com/user-attachments/files/22095166/truck_real.usd.zip) file I used, and the screenshot when I used `replicate_physics=True`

[image]

The table, truck are gone which are connected with the fixed joint.

### System Info

Describe the characteristic of your environment:

- Commit: 157a19e
- Isaac Sim Version: 4.5.0
- OS: 22.04
- GPU: L40S
- CUDA: 12.8
- GPU Driver: 535.129.03 

### Additional context

Add any other context about the problem here.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related 

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

RandomOakForest, 2025-09-02:
Thank you for posting this. Do you get any errors? Could you post a larger error log? Note that [in the docs is mentioned that you will incur slowdowns if you set replicate physics to false](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.scene.html#isaaclab.scene.InteractiveSceneCfg.replicate_physics). It would be great if you could post a video of what you are seeing to understand your issue.

MinchoU, 2025-09-03:
Thanks for the quick response! @RandomOakForest 
I don't get errors, but I guess the warning `without a body rel may cause issues, since the localPose wont be updated.` is closely related the issue I'm facing.
This is the [video](https://github.com/user-attachments/assets/760a7e19-b219-476e-9994-2f1d4e4f8246) that I recorded, and the [ouput.log](https://github.com/user-attachments/files/22107120/ouput.log) from the video.

Thank you in advance!