isaac-sim/IsaacLab #3326: [Question] Question title

1izeyi, 2025-09-02:
Can I use the action graph extension in isaac sim 4.5 or 5.0? I have customized some nodes and I am using the direct workflow so will that make it easier?

RandomOakForest, 2025-09-03:
Yes, the action graph extension is supported in both Isaac Sim 4.5 and 5.0, allowing workflows that include custom nodes and the direct workflow approach.

1izeyi, 2025-09-03:
@RandomOakForest Thank you very very much！Sir！But I'm still confused now. I'm using the direct workflow in isaaclab. How do I enable my action graph extension? I'm currently using an omni.physx.graph.OnTriggerCollider, an official node, and I customized a Python API node. Now I have two questions. 
1、Firstly, how do I enable both extensions and enable the action graph in the isaaclab direct workflow? 
2、The second question is, can I detect collisions directly in the simulation loop without using an extension? I did do that, but the mesh collision detection relies on the official node On Trigger, so can our official extension be independent of the extension import module?
 I'm a newbie, but I have this difficult and urgent task now. Could I ask your team for some code demos? I wonder if you have any suggestions. Finally, I'm very sorry to bother you, but I'm really anxious. Crying...