isaac-sim/IsaacLab #3332: [Bug Report] GPU resource have not free correctly after stopping the bad trials using ASHAScheduler

PierrePeng, 2025-09-03:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

I'm using ray feature to hpo with ASHAScheduler instead of default scheduler(FIFO). After stoping some bad trials and starting new trials, I found the old trials process doesn't exit correctly. So the gpu resource have not free.

### Steps to reproduce

```
./isaaclab.sh -p scripts/reinforcement_learning/ray/tuner.py \
  --cfg_file scripts/reinforcement_learning/ray/hyperparameter_tuning/vision_cartpole_cfg.py \
  --cfg_class CartpoleTheiaJobCfg \
  --run_mode local \
  --workflow scripts/reinforcement_learning/rl_games/train.py \
  --num_workers_per_node <NUMBER_OF_GPUS_IN_COMPUTER>
```

```
  tuner = tune.Tuner(
        IsaacLabTuneTrainable,
        param_space=cfg,
        tune_config=tune.TuneConfig(
            metric=args.metric,
            mode=args.mode,
            scheduler=ASHAScheduler(time_attr="time_this_iter_s", grace_period=3,
                                    max_t=5, reduction_factor=2),
            search_alg=repeat_search,
            num_samples=args.num_samples,
            reuse_actors=True,
        ),
        run_config=run_config,
    )```

For more information on this, check: https://www.markdownguide.org/extended-syntax/#fenced-code-blocks

-->

### System Info

Describe the characteristic of your environment:

<!-- Please complete the following description. -->
- Commit: [e.g. 2f9298d78c11cf5803bb909cd4420e64a4e50aec]
- Isaac Sim Version: [5.0]
- OS: [Ubuntu 22.04]
- GPU: [RTX 4090]
- CUDA: [12.8]
- GPU Driver: [553.05]

### Additional context

Add any other context about the problem here.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

celestialdr4g0n, 2025-09-03:
you can see the process in nvidia-smi and kill it with 
```bash
kill -9 process_id
```

PierrePeng, 2025-09-04:
Thanks garylvov, I will keep tracking this issue

closed 2025-09-04 by PierrePeng