isaac-sim/IsaacLab #3343: [Question] How to get the inertia tensor of a RigidObject

BaoyiCui, 2025-09-04:
### Question

I imported an .obj file using Mesh Importer with `--mass 1.0`, which automatically assigned rigid body properties to my robot.

[image]

In the example code, i can retrieve the rigid body's mass like this:

```python
self._robot_mass = self._robot.root_physx_view.get_masses()[0].sum()  # mass of the drone
```

Is there a similar API to obtain the rigid body's inertia tensor? I would like to use it for quadrotors attitude controller design.

RandomOakForest, 2025-09-08:
Thank you for posting this. The inertia tensor of a RigidObject imported into Isaac Sim (such as using the Mesh Importer) can be accessed using a documented API. Unlike mass retrieval (using get_masses()), the inertia tensor is typically exposed as a property of the rigid asset or through methods that provide rigid body parameters directly. You may try
```
inertia_tensor = robot.root_physx_view.get_inertias()  # shape (9,)
```
or
```
inertia_tensor = robot.default_inertia
```
usually accessed via the root PhysX view.

BaoyiCui, 2025-09-09:
> Thank you for posting this. The inertia tensor of a RigidObject imported into Isaac Sim (such as using the Mesh Importer) can be accessed using a documented API. Unlike mass retrieval (using [...]

Thank you for you help!!! I have tried it and can successfully get the inertia of the robot.