isaac-sim/IsaacLab #3349: [Bug Report] Missing Ray initialization while running`tuner.py`

ozhanozen, 2025-09-04:
Ray is not initialized while running `tuner.py`, throwing an exception.

### Steps to reproduce

Try to run the `Docker-based Local Quickstart` part of the [docs](https://isaac-sim.github.io/IsaacLab/main/source/features/ray.html). You should get the following exception:
"Ray is not initialized. Please initialize Ray before getting node resources."

### Additional context

PR #2847, commit [286eb39](https://github.com/isaac-sim/IsaacLab/pull/2847/commits/286eb3970774fe739e01b82e19c3e69b289e4340), modified `get_gpu_node_resources()` to be:
https://github.com/isaac-sim/IsaacLab/blob/5dae83eb921dad478d87f21d4db154d8e0cd68fe/scripts/reinforcement_learning/ray/util.py#L345-L346
which was originally:
```
if not ray.is_initialized():
        ray.init(address=ray_address)
```
Now, ray is not initialized when `tuner.py` is directly used. 

Is there a reason to change the previous implementation? If not, we could simply revert this line or use the new `ray_init()`.

### System Info

- Commit: [5dae83e](https://github.com/isaac-sim/IsaacLab/commit/5dae83eb921dad478d87f21d4db154d8e0cd68fe)
- Isaac Sim Version: 5.0
- OS: Ubuntu 22.04
- GPU: RTX 4090
- CUDA: 12.2
- GPU Driver: 535.129.03

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

- [x] `tuner.py` runs without errors

binw666, 2025-09-04:
> Hi [@garylvov](https://github.com/garylvov) and [@binw666](https://github.com/binw666), I'm tagging you because I thought you might know best if reverting the aforementioned line is safe for the new functionalities, or alternatively, if doing `ray_init()` somewhere else is better.

I'm very sorry for introducing the bug. I think reverting this line or using the new `ray_init()` is safe for the new feature.

ozhanozen, 2025-09-04:
No problem, I will create a PR for the fix then.

PR #3350 (2025-09-05): Fixes the missing Ray initialization
# Description

Recent changes introduced a minor bug: now, Ray is not initialized when `tuner.py` is called. This PR fixes this by adding back the removed initialization. 

Fixes #3349 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Screenshots
[image]

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with `./isaaclab.sh --format`
- [ ] I have made corresponding changes to the documentation
- [x] My changes generate no new warnings
- [ ] I have added tests that prove my fix is effective or that  [... cut]
commit 1a71e24a (2025-09-05): Fixes the missing Ray initialization (#3350)

# Description

Recent changes introduced a minor bug: now, Ray is not initialized when
`tuner.py` is called. This PR fixes this by adding back the removed
initialization.

Fixes #3349 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Screenshots
<img width="1637" height="34" alt="pr_changes"
src="https://github.com/user-attachments/assets/820c55ce-9951-405a-ac7b-005dfeab877c"
/>

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./isaaclab.sh --format`
- [ ] I have made corresponding [... cut]
closed 2025-09-05 by kellyguo11