isaac-sim/IsaacLab #3365: [Question] Randomizing visual texture of table / ground in franka lift cube task?

jiwonyou0420, 2025-09-05:
I'm trying to implement domain randomization on visuals in franka lift cube task.

I was able to randomize visual texture of cube and franka robot using `mdp.randomize_visual_texture_material` event.

However, I cannot apply the same logic to table or ground. Appears that the above event can be applied to `RigidObjectCfg` or `ArticulationCfg` scene entities but doesn't work on table and ground because they're declared via `AssetBaseCfg`? 

What would be good practice to randomize visual texture of table or ground? 

I'm using IsaacSim 5.0.

Thanks in advance!

RandomOakForest, 2025-09-08:
Thank you for posting this. To randomize the **visual texture of the table or ground** in the Franka lift cube task avoid the error-prone approach of using `RigidObjectCfg`/`ArticulationCfg` logic for entities declared via `AssetBaseCfg`. In your case,

- The `mdp.randomize_visual_texture_material` function expects entities that have bodies (i.e., `RigidObjectCfg` or `ArticulationCfg`) but `AssetBaseCfg` scene primitives like the table/ground are typically `XFormPrim` objects, which do **not** have `find_bodies()`.
- Your error is due to passing `SceneEntityCfg("table", body_names=".*")` for an entity (`table`) that is an `XFormPrim` without physics bodies. The method calls fail on this mismatch.
- Additionally, the `randomize_visual_texture_material` event handler may not support some expected arguments (like `asset_cfg`) for statically added assets.

## Recommended Best Practice

### 1. Use `randomize_visual_material_prim` for XFormPrim entities

For entities that do **not** have rigid bodies or articulations (like tables or ground planes created via `AssetBaseCfg`), utilize the event function designed for general prims: `mdp.randomize_visual_material_prim`. This works directly with the prim path.

#### Example EventCfg usage:

```python
@configclass
class EventCfg:
    rand_table_texture = EventTerm(
        func=mdp.randomize_visual_material_prim,
        mode="reset",
        params={
            "prim_paths": ["/World/envs/env_0/table"],  # Use the actual prim path of your table
            "texture_paths": [
                f"{NVIDIA_NUCLEUS_DIR}/Materials/Base/Wood/Bamboo_Planks/Bamboo_Planks_BaseColor.png",
                f"{NVIDIA_NUCLEUS_DIR}/Materials/Base/Wood/Cherry/Cherry_BaseColor.png",
                f"{NVIDIA_NUCLEUS_DIR}/Materials/Base/Wood/Oak/Oak_BaseColor.png",
                # etc.
            ],
            "event_name": "rand_table_texture",
            "texture_rotation": (0.0, 6.283185307179586),
        }
    )
```

- Replace prim paths with those in your actual scene hierarchy (check via Isaac Sim’s stage inspector).

### 2. Correct Arguments for Domain Randomization Functions

- When randomizing visuals for non-body scene entities, do **not** use the `asset_cfg` or `body_names` arguments; instead, use `prim_paths`.

#### Supported domain randomization functions for visuals:

- `mdp.randomize_visual_texture_material` – for assets defined by RigidObjectCfg/ArticulationCfg
- `mdp.randomize_visual_material_prim` – for assets defined by AssetBaseCfg (e.g., tables, ground, custom meshes)

### 3. How to get prim paths

- In Isaac Sim, right-click the asset in the stage panel to copy its **prim path**. This is necessary for randomization events targeting those assets.

## Additional Notes

- Keep `self.scene.replicate_physics = False` as recommended when randomizing visuals; otherwise, asset replications could interfere with texture resets.
- If randomizing the ground plane, use its prim path (e.g., `/World/envs/env_0/ground`), and utilize the same event construction and function as above.

jiwonyou0420, 2025-09-08:
Thank you but I cannot find `randomize_visual_material_prim` from main branch. From events.py, I only see `randomize_visual_texture_material` and `randomize_visual_color`. Is it on another branch?

jiwonyou0420, 2025-09-09:
chatgpt made this and seems to work for me.

```
def _as_list(x: Union[str, Iterable[str]]) -> List[str]:
    if x is None:
        return []
    if isinstance(x, str):
        return [x]
    return list(x)

# gpt made this one
class randomize_visual_material_prim(ManagerTermBase):
    """
    Randomize the *visual* material (diffuse texture + rotation) on arbitrary prim paths
    (e.g., tables, ground planes, static meshes) using the Replicator API.

    This is meant for assets added via AssetBaseCfg (without rigid bodies/articulations).
    It binds OmniPBR materials to the *visual* prims under the given prim paths and
    randomly assigns textures each time the event is triggered.

    Params (in EventTerm.params):
        prim_paths: str | list[str]
            One or more prim roots to target. Their descendants' visual prims are included.
            Tip: pass exact prim (e.g., "/World/envs/env_.*/table") – wildcards allowed.
        texture_paths: list[str]
            List of image files (color/base/diffuse maps). Use your wrapper to expand folders first.
        event_name: str
            Custom event name (used only for Replicator < 1.12.4).
        texture_rotation: tuple[float, float] (radians)
            Range for random rotation (min, max) in radians.
    """

    def __init__(self, cfg, env: ManagerBasedEnv):
        super().__init__(cfg, env)

        # Safety: replicator requires scene replication off for stable USD randomization
        scene_cfg: InteractiveSceneCfg = env.cfg.scene
        if scene_cfg.replicate_physics:
            raise RuntimeError(
                "Unable to randomize visual material with scene replication enabled. "
                "Set scene.replicate_physics = False in InteractiveSceneCfg."
            )

        enable_extension("omni.replicator.core")
        import omni.replicator.core as rep

        # Read base params from cfg (can be overridden at __call__)
        params = self.cfg.params
        self._prim_paths = _as_list(params.get("prim_paths", []))
        self._texture_paths = _as_list(params.get("texture_paths", []))
        self._event_name = params.get("event_name", "rand_visual_material_prim")
        self._rot_range = params.get("texture_rotation", (0.0, 0.0))  # radians
        self._use_project_uvw = bool(params.get("use_project_uvw", True))
        self._meters_per_repeat = params.get("meters_per_repeat", None)   # float or (u,v) in meters
        self._uv_tiling = params.get("uv_tiling", None)    
        # Extract replicator version
        version = re.match(r"^(\d+\.\d+\.\d+)", rep.__file__.split("/")[-5][21:]).group(1)

        self._new_api = compare_versions(version, "1.12.4") >= 0
        self._material_prims = None
        self._rng = None

        if self._new_api:
            # Pre-create materials and bind (once) for all matching prims
            stage = get_current_stage()

            # Expand each provided prim path to include descendants (visuals often live below)
            # Patterns like "/World/envs/env_0/table/.*" work well; wildcards in roots also work.
            expanded_patterns = []
            for p in self._prim_paths:
[... 128 lines cut ...]
                        usd_prim, mpr_val,
                        ["inputs:project_uvw_scale", "inputs:texture_scale", "inputs:uv_scale"],
                        "projected-UV scale (meters_per_repeat)"
                    )
                if (not (use_project_uvw or self._use_project_uvw)) and uv_tiling is not None:
                    til = uv_tiling
                    til_val = (float(til[0]), float(til[1])) if isinstance(til, (list, tuple)) else (float(til), float(til))
                    _set_first_available_input(
                        usd_prim, til_val,
                        ["inputs:uv_tiling", "inputs:texture_tiling", "inputs:uv_scale", "inputs:texture_scale"],
                        "UV tiling"
                    )
            rot_min_deg, rot_max_deg = [math.degrees(a) for a in rot_range]
            num = len(self._material_prims)
            # Choose with replacement to cover arbitrary counts
            random_textures = self._rng.generator.choice(texture_paths, size=num)
            random_rotations = self._rng.generator.uniform(rot_min_deg, rot_max_deg, size=num)

            rep.functional.modify.attribute(self._material_prims, "diffuse_texture", random_textures)
            rep.functional.modify.attribute(self._material_prims, "texture_rotate", random_rotations)
        else:
            # Old API: fire the custom event graph
            rep.utils.send_og_event(event_name)

```