isaac-sim/IsaacLab #3414: [Question] Question title

zsy9512, 2025-09-10:
### Question

I encountered an issue while installing with pip. When I ran `isaaclab.bat --install`, it output: 
``` ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts. isaacsim-kernel 5.0.0.0 requires sentry-sdk==1.43.0, but you have sentry-sdk 2.37.1 which is incompatible. Successfully installed isaaclab_rl-0.2.3 rsl-rl-lib-2.3.3 sentry-sdk-2.37.1 ``` 
This package conflict may caused by the previous step: ``` pip install "isaacsim[all,extscache]==5.0.0" --extra-index-url https://pypi.nvidia.com ``` which requires `sentry-sdk>2.0.0`.
 System environment: -
 Isaac Sim Version: 5.0.0 
- Isaac Lab Version: v2.2.0 
- OS: Windows 11 
- GPU: RTX 5070 
- CUDA: 12.8 
- GPU Driver: 580.97
 Additionally, there is a bug in `isaaclab.bat` on the main branch of Isaac Lab: the installer cannot locate the already-installed PyTorch package, causing it to reinstall repeatedly.

RandomOakForest, 2025-09-15:
Thank you for posting this. Have you tried
```
pip uninstall sentry-sdk
pip install sentry-sdk==1.43.0
```
before rerunning the install?

AntoineRichard, 2026-08-27:
@zsy9512, thanks for reporting this with the exact version pins — dependency conflicts like this are easy to lose track of otherwise.

The clash came from `isaacsim-kernel` 5.0.0 pinning `sentry-sdk==1.43.0` while `wandb` needs `>=2.0`. Isaac Lab now targets Isaac Sim 6.0.1 and ships a uv-managed lockfile that resolves a single `sentry-sdk` 2.42.1 for both, so a fresh install doesn't hit this. Your secondary complaint is moot too: `isaaclab.bat` no longer touches PyTorch at all. Closing — reopen if the current uv install path still trips over it.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard